aymanbagabas/Huawei-WMI #2: No support for Matebook X

aymanbagabas, 2018-10-31:
This driver is meant to work for all Huawei laptops. Currently, MBXP is the only laptop that is supported. Supporting its successor, Matebook X, is the next step. I will probably need some testers and more information about the MBX such as DSDT and SSDT tables.

cytrinox, 2018-11-01:
The DSDT from my MBX is located at https://github.com/cytrinox/matebook-hotkeys/tree/master/acpi

aymanbagabas, 2018-11-05:
@cytrinox I updated the `dev` branch to support the MBX. Could you please verify that all of the keys work as expected?

Also, we need to figure out what function triggers the micmute led if there is any, so try to test if any of the following does trigger that led. Use `acpi_call`
```
echo "\WMI0.WM05 0 0 0x01" | sudo tee /proc/acpi/call
echo "\WMI0.WM05 0 0 0x02" | sudo tee /proc/acpi/call
echo "\WMI0.WM05 0 0 0x03" | sudo tee /proc/acpi/call
echo "\WMI0.WM05 0 0 0x04" | sudo tee /proc/acpi/call
echo "\WMI0.WM05 0 0 0x05" | sudo tee /proc/acpi/call
echo "\WMI0.WM05 0 0 0x06" | sudo tee /proc/acpi/call
echo "\WMI0.WM05 0 0 0x07" | sudo tee /proc/acpi/call
echo "\WMI0.WM05 0 0 0x08" | sudo tee /proc/acpi/call
```
If any of these worked, what is the behavor? Does it toggle the led? Or does one turns it on and another turns it off?

aymanbagabas, 2018-11-05:
@cytrinox Can you provide me a dump your acpi tables? Since you updated the BIOS there gotta be some changes.
```
# acpidump > acpidump.out
```

cytrinox, 2018-11-05:
Attached is the acpidump.

[acpidump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/2549892/acpidump.txt)

cytrinox, 2018-11-05:
None of the __\WMI0.WM05__ calls results in a MICMUTE-LED change.

cytrinox, 2018-11-05:
Just tested the dev-branch. Each hotkey works as expected. Micmute works well after adding the udev hwdb and is correctly reported by __xev__ as __XF86AudioMicMute__.

aymanbagabas, 2018-11-05:
Looking at the DSDT table, **WMI0.M242** seems to be the one responsible with changing micmute LED.
```
            If ((STAT == One))
            {
                \_SB.PCI0.LPCB.EC0.SPIN (Zero, 0x04, Zero)
            }
            Else
            {
                \_SB.PCI0.LPCB.EC0.SPIN (Zero, 0x04, One)
            }
```

Try if these change the LED. On the MBXP, pin 0x04 is the micmute LED.
```
echo "\_SB.PCI0.LPCB.EC0.SPIN 0 0x04 0" | sudo tee /proc/acpi/call
echo "\_SB.PCI0.LPCB.EC0.SPIN 0 0x04 1" | sudo tee /proc/acpi/call
```

Method **WMI0.WM05** is the one associated with the GUID. It calls **WMI0.WMIG** then this calls **WGLN** which calls multiple methods including **WMI0.MF24** which then calls **WMI0.M242**. So if the above acpi_call calls trigger the LED, there should be a way to control the LED through a WMI call. Maybe the WMI0.INIT has something to do with it? Could you try the calls from the previous comment after calling **WMI0.INIT** with 0 argument. You could see the output of each acpi_call using `cat /proc/acpi/call`.

cytrinox, 2018-11-05:
__\_SB.PCI0.LPCB.EC0.SPIN__ is correct, this call (0|1) disable/enable the LED.
What is the correct line for WMI0.INIT? Do you mean zero arguments or one argument with 0 value?

aymanbagabas, 2018-11-05:
One with a 0 value argument.

cytrinox, 2018-11-05:
No change, each call is responded with "0x0called"

aymanbagabas, 2018-11-05:
@cytrinox try fiddling around with **WMI0.M242** and figure out what are the values that turn the led on/off. This way we could trace back the call and find out what are the correct inputs for **WMI0.WM05** that triggers the led. I'm using `acpiexec` to emulate the ACPI tables in user-space which has its limitations.

cytrinox, 2018-11-05:
M242 called with 0 0 0 enables the LED, but I've not found a combination which disable it.
I don't know how to read the DSL, but the definition of M242 looks like it checks a memory region or a internal field? If this is true, M242 may just toogle the LED depending on data not provided by arguments, but by a internal value.

aymanbagabas, 2018-11-05:
**M242** only looks for the first argument. Try using 0xFFFF, 0x1 ,0x10000. It depends on the variable 'STAT' in that method which is only mentioned 4 times in the DSDT file. The argument has a size of 32 bit and it will look like 0xXXXXXXXX in hex. 

While the LED is on, try if the previous **WM05** calls change anything. Maybe do the same while it's off. 

What is the value of
```
echo "\_SB.PMIM" | sudo tee /proc/acpi/call`
sudo cat /proc/acpi/call
```

@cytrinox we could try a brute-force approach by feeding arbitrary arguments to the method associated with the GUID and we know it has to be **WMI0.WM05**. Looking at your `bmf.mof` file, it seems that the method takes one uint32 argument.
`void MyFn1([in] uint32 Indx)` We could try to execute the method with arguments from 0x0 to 0xFFFFFFFF. 4294967295 calls to the method. The number is huge and it will definitely take time. Let's make that our last option. Now, let's try to narrow our search down and see if we get anything.

cytrinox, 2018-11-06:
````
root@antares:~/Huawei-WMI# echo "\_SB.PMIM" | sudo tee /proc/acpi/call
\_SB.PMIM
root@antares:~/Huawei-WMI# cat /proc/acpi/call
0x7ff6a000root@antares:~/Huawei-WMI#
````

No luck with 0xFFFF, 0x1, 0x10000 or the WM05 calls.

aymanbagabas, 2018-11-07:
@cytrinox I think I'm overthinking this problem. I went with using ACPI calls to toggle the micmute LED.
I updated the dev branch with these changes. We will probably need to apply these changes to the kernel and test it out. I'll let you know once I finish with the kernel patches.

aymanbagabas, 2018-11-07:
@cytrinox Could you please provide the output of this:
```
cat /sys/class/sound/hwC0D0/subsystem_id
```
I just want to know the subsystem id of the sound card so that we can hook the micmute led with ALSA.

janbaer, 2018-11-08:
```
$ cat /sys/class/sound/hwC0D0/subsystem_id
0x19e53201
 
$ aplay --list-devices
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC298 Analog [ALC298 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
 
$ lspci -nnv
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers [8086:5904] (rev 02)
        Subsystem: Huawei Technologies Co., Ltd. Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers [19e5:3e02]
        Flags: bus master, fast devsel, latency 0
        Capabilities: <access denied>
lspci: Unable to load libkmod resources: error -12

00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 620 [8086:5916] (rev 02) (prog-if 00 [VGA controller])
        Subsystem: Huawei Technologies Co., Ltd. HD Graphics 620 [19e5:3e02]
        Flags: bus master, fast devsel, latency 0, IRQ 132
        Memory at a0000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 90000000 (64-bit, prefetchable) [size=256M]
        I/O ports at 4000 [size=64]
        [virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
        Capabilities: <access denied>
        Kernel driver in use: i915

00:04.0 Signal processing controller [1180]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem [8086:1903] (rev 02)
        Subsystem: Huawei Technologies Co., Ltd. Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem [19e5:3e02]
        Flags: fast devsel, IRQ 16
        Memory at a1220000 (64-bit, non-prefetchable) [size=32K]
        Capabilities: <access denied>
        Kernel driver in use: proc_thermal

00:14.0 USB controller [0c03]: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller [8086:9d2f] (rev 21) (prog-if 30 [XHCI])
        Subsystem: Huawei Technologies Co., Ltd. Sunrise Point-LP USB 3.0 xHCI Controller [19e5:3e02]
        Flags: bus master, medium devsel, latency 0, IRQ 129
        Memory at a1200000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: <access denied>
        Kernel driver in use: xhci_hcd

00:14.2 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Thermal subsystem [8086:9d31] (rev 21)
        Subsystem: Huawei Technologies Co., Ltd. Sunrise Point-LP Thermal subsystem [19e5:3e02]
        Flags: bus master, fast devsel, latency 0, IRQ 18
        Memory at a1230000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: <access denied>
        Kernel driver in use: intel_pch_thermal

00:15.0 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 [8086:9d60] (rev 21)
        Subsystem: Huawei Technologies Co., Ltd. Sunrise Point-LP Serial IO I2C Controller [19e5:3e02]
        Flags: bus master, fast devsel, latency 0, IRQ 16
        Memory at a1231000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: <access denied>
        Kernel driver in use: intel-lpss

[... 60 lines cut ...]
        Capabilities: <access denied>
        Kernel driver in use: snd_hda_intel

00:1f.4 SMBus [0c05]: Intel Corporation Sunrise Point-LP SMBus [8086:9d23] (rev 21)
        Subsystem: Huawei Technologies Co., Ltd. Sunrise Point-LP SMBus [19e5:3e02]
        Flags: medium devsel, IRQ 16
        Memory at a1236000 (64-bit, non-prefetchable) [size=256]
        I/O ports at 4040 [size=32]
        Kernel driver in use: i801_smbus

01:00.0 Non-Volatile memory controller [0108]: Lite-On Technology Corporation Device [14a4:5100] (rev 01) (prog-if 02 [NVM Express])
        Subsystem: Lite-On Technology Corporation Device [14a4:5100]
        Flags: bus master, fast devsel, latency 0, IRQ 18, NUMA node 0
        Memory at a1100000 (64-bit, non-prefetchable) [size=32K]
        I/O ports at 3000 [size=8]
        Capabilities: <access denied>
        Kernel driver in use: nvme

02:00.0 Network controller [0280]: Intel Corporation Wireless 8265 / 8275 [8086:24fd] (rev 78)
        Subsystem: Huawei Technologies Co., Ltd. Wireless 8265 / 8275 [19e5:3e02]
        Flags: bus master, fast devsel, latency 0, IRQ 131
        Memory at a1000000 (64-bit, non-prefetchable) [size=8K]
        Capabilities: <access denied>
        Kernel driver in use: iwlwifi
```

janbaer, 2018-11-08:
What's the current state. Is it working now also for the MatebookX? Should I test something?

Thanks for your effort!

aymanbagabas, 2018-11-09:
@janbaer Thanks for the response.

Yes, it should work for the Matebook X, it's just that I haven't updated the master branch and the documentation. Use dev branch to try it out. It would be really nice if you could try building a custom kernel with this module and patching the sound driver for the micmute led to work. You could use the patches in the dev branch and apply them using `git am <patch file>` to a kernel git clone.

[This](https://bugzilla.kernel.org/attachment.cgi?id=260539) shows a different subsystem id (19e5:3200) from yours (19e5:3201) which is kinda interesting. I wanted these outputs so I can proceed with pushing the driver to kernel upstream. This is needed for the led to work.

Again thank you for reaching out.

cytrinox, 2018-12-06:
AFAIK everything works on MBX, what is in the testing pipeline?

aymanbagabas, 2018-12-07:
> AFAIK everything works on MBX, what is in the testing pipeline?

I am almost done with submitting this to upstream. However, the only thing left is testing. Now I don't have a MBX which makes it harder for me to test. The problem is that the code has to be very well written and clean as possible. Now the method used to read input from MBX is bad and that is due to how Huawei implemented their firmware.
This would make things much cleaner by using the GUID associated with fetching the input code from keyboard. I'm interested in dmesg output on a hotkey press.

[huawei-wmi-r1.zip](https://github.com/aymanbagabas/Huawei-WMI/files/2655976/huawei-wmi-r1.zip)

janbaer, 2018-12-07:
@aymanbagabas I have an MBX and I would like to help you with testing, but I'm sorry my knowledge about Linux is not so deep to compile the kernel on my own. But of course, I'm very interested to use also the fn-keys on my MBX.

aymanbagabas, 2018-12-07:
@janbaer Soon this patch should be included in the linux kernel. @cytrinox is doing a great job in testing but it wouldn't hurt to test it on another machine.
You don't need to recompile the kernel you could try the patch just by itself. Download it, compile it and insmod. 
[huawei-wmi-r1.zip](https://github.com/aymanbagabas/Huawei-WMI/files/2655976/huawei-wmi-r1.zip)
```
$ unzip huawei-wmi-r1.zip -d huawei-wmi
$ cd huawei-wmi
$ make
$ sudo modprobe sparse-keymap
$ sudo insmod huawei-wmi.ko
```
and in a separate window run `dmesg -w` You should see `Huawei WMI hotkeys`. Test the keys and submit dmesg log.

aymanbagabas, 2018-12-09:
@janbaer I don't see any errors. These are warnings that you can ignore.

aymanbagabas, 2018-12-10:
@janbaer Did you get it to work?

janbaer, 2018-12-11:
Yes, it worked, thanks! So this is the output:

```
[    0.000000] microcode: microcode updated early to revision 0x8e, date = 2018-03-24
[    0.000000] Linux version 4.19.6-1-MANJARO (builduser@development) (gcc version 8.2.1 20180831 (GCC)) #1 SMP PREEMPT Sat Dec 1 12:21:26 UTC 2018
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.19-x86_64 root=UUID=bcd63e36-7ab8-42de-b554-f9e7d10a22cd rw quiet resume=UUID=e7ed8062-f33c-4173-89f0-363eea8751c5
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000066876fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000066877000-0x0000000066877fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000066878000-0x00000000668abfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000668ac000-0x00000000668acfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000668ad000-0x0000000067b87fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067b88000-0x0000000068c87fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000068c88000-0x000000007ee9dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ee9e000-0x000000007f09dfff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007f09e000-0x000000007f88dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007f88e000-0x000000007ff7dfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007ff7e000-0x000000007fffdfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007fffe000-0x000000007fffefff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffff000-0x000000008cffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fe7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed84000-0x00000000fed84fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000271ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.50 by HUAWEI
[    0.000000] efi:  ACPI 2.0=0x7fffd014  SMBIOS=0x7f114000  SMBIOS 3.0=0x7f112000  ESRT=0x7f110dd8  PROP=0x66873098 
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: HUAWEI HUAWEI MateBook X/HUAWEI MateBook X, BIOS 1.03 05/19/2017
[    0.000000] tsc: Detected 2900.000 MHz processor
[    0.001282] tsc: Detected 2904.000 MHz TSC
[    0.001282] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001284] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001293] last_pfn = 0x272000 max_arch_pfn = 0x400000000
[    0.001297] MTRR default type: write-back
[    0.001297] MTRR fixed ranges enabled:
[    0.001298]   00000-9FFFF write-back
[    0.001299]   A0000-BFFFF uncachable
[    0.001300]   C0000-FFFFF write-protect
[    0.001301] MTRR variable ranges enabled:
[    0.001302]   0 base 00C0000000 mask 7FC0000000 uncachable
[    0.001303]   1 base 00A0000000 mask 7FE0000000 uncachable
[    0.001304]   2 base 0090000000 mask 7FF0000000 uncachable
[    0.001304]   3 base 008C000000 mask 7FFC000000 uncachable
[    0.001305]   4 base 008A000000 mask 7FFE000000 uncachable
[    0.001306]   5 base 0089000000 mask 7FFF000000 uncachable
[    0.001307]   6 base 0088800000 mask 7FFF800000 uncachable
[    0.001307]   7 disabled
[... 1048 lines cut ...]
[  903.269779] audit: type=1130 audit(1544504310.292:183): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-tmpfiles-clean comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  903.269788] audit: type=1131 audit(1544504310.292:184): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-tmpfiles-clean comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  903.775551] Returned obj: 642
[  904.452140] Returned obj: 660
[  904.868866] Returned obj: 644
[  905.378859] Returned obj: 645
[  905.875520] Returned obj: 646
[  906.825508] Returned obj: 647
[  907.842149] Returned obj: 649
[... cut ...]
[  912.723424] audit: type=1130 audit(1544504319.745:186): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  912.777502] audit: type=1130 audit(1544504319.799:187): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=ntpd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  912.859489] audit: type=1131 audit(1544504319.882:188): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
````

janbaer, 2018-12-11:
What I see is, that the keys for mute, volume and wlan are working now, but the keys for the light (F1, F2, F3) have no reaction.

aymanbagabas, 2018-12-11:
Thank you for your work @janbaer. The driver seems to successfully report those keys.
```
[  817.719774] Returned obj: 641
[  820.952951] Returned obj: 642
[  822.362798] Returned obj: 659
...
[  903.255595] Returned obj: 641
[  903.775551] Returned obj: 642
[  904.452140] Returned obj: 660
```
641 -> brightness down
642 -> brightness up
659, 660, 661 -> keyboard backlight which is ignored because it should work natively using BIOS

The display key is mapped to `super+p` which is what Windows and most Linux DEs use for display options.

Maybe it has to do with the DE you're using. What DE you running? Try using `xev`, `acpi_listen` or `evtest` to detect if a key is pressed.

janbaer, 2018-12-11:
What so you mean with DE?

I just testet the output with  **evtest** and got the following output

```
sudo evtest /dev/input/event15

Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "Huawei WMI hotkeys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 171 (KEY_CONFIG)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
    Event code 238 (KEY_WLAN)
    Event code 240 (KEY_UNKNOWN)
    Event code 248 (KEY_MICMUTE)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1544558454.954004, type 4 (EV_MSC), code 4 (MSC_SCAN), value 281
Event: time 1544558454.954004, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1544558454.954004, -------------- SYN_REPORT ------------
Event: time 1544558454.954033, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 0
Event: time 1544558454.954033, -------------- SYN_REPORT ------------
Event: time 1544558470.664122, type 4 (EV_MSC), code 4 (MSC_SCAN), value 282
Event: time 1544558470.664122, type 1 (EV_KEY), code 225 (KEY_BRIGHTNESSUP), value 1
Event: time 1544558470.664122, -------------- SYN_REPORT ------------
Event: time 1544558470.664166, type 1 (EV_KEY), code 225 (KEY_BRIGHTNESSUP), value 0
Event: time 1544558470.664166, -------------- SYN_REPORT ------------
Event: time 1544558477.227367, type 4 (EV_MSC), code 4 (MSC_SCAN), value 284
Event: time 1544558477.227367, type 1 (EV_KEY), code 113 (KEY_MUTE), value 1
Event: time 1544558477.227367, -------------- SYN_REPORT ------------
Event: time 1544558477.227390, type 1 (EV_KEY), code 113 (KEY_MUTE), value 0
Event: time 1544558477.227390, -------------- SYN_REPORT ------------
Event: time 1544558483.620747, type 4 (EV_MSC), code 4 (MSC_SCAN), value 285
Event: time 1544558483.620747, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1544558483.620747, -------------- SYN_REPORT ------------
Event: time 1544558483.620783, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1544558483.620783, -------------- SYN_REPORT ------------
Event: time 1544558484.864115, type 4 (EV_MSC), code 4 (MSC_SCAN), value 286
Event: time 1544558484.864115, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1544558484.864115, -------------- SYN_REPORT ------------
Event: time 1544558484.864158, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1544558484.864158, -------------- SYN_REPORT ------------
Event: time 1544558486.630782, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1544558486.630782, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
Event: time 1544558486.630782, -------------- SYN_REPORT ------------
Event: time 1544558486.630825, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
Event: time 1544558486.630825, -------------- SYN_REPORT ------------
Event: time 1544558494.090748, type 4 (EV_MSC), code 4 (MSC_SCAN), value 289
Event: time 1544558494.090748, type 1 (EV_KEY), code 238 (KEY_WLAN), value 1
Event: time 1544558494.090748, -------------- SYN_REPORT ------------
Event: time 1544558494.090770, type 1 (EV_KEY), code 238 (KEY_WLAN), value 0
Event: time 1544558494.090770, -------------- SYN_REPORT ------------
Event: time 1544558496.450568, type 4 (EV_MSC), code 4 (MSC_SCAN), value 28a
Event: time 1544558496.450568, type 1 (EV_KEY), code 171 (KEY_CONFIG), value 1
Event: time 1544558496.450568, -------------- SYN_REPORT ------------
Event: time 1544558496.450590, type 1 (EV_KEY), code 171 (KEY_CONFIG), value 0
Event: time 1544558496.450590, -------------- SYN_REPORT ------------
```

The key that is behind F3 gave no output and the same with F8.

aymanbagabas, 2018-12-11:
@janbaer F3 (keyboard back light) should work without any problem because it is controlled through BIOS. F8 (display key) is a different issue, on my system when I use xev, F8 reports super+p. I use Gnome and I had to remap super+p to get it to work which is an issue with Gnome I guess. But in XFCE for instance, F8 worked fine.

Designator-Ol, 2018-12-28:
Hi there, thanks for your work with the function keys. they work perfectly now on my MBX (w/o pro). 
Do you plan on fixing the sound issue on the MBX too ? As all the others I only have sound on the front left speaker but not on the right one. If I should test something for you let me know ! Your help is higly appreciated ! 

Thanks an advance.

janbaer, 2019-01-01:
I just tested it today with a Gnome installation instead of i3 and it turned out that all keys are working now like it was expected. Means, that also the keys for the display light and the display key (F8) are working. It was an issue with i3 I think. Thanks for the great job! What about the file you attached here to this issue. Will you merge it  to the original source and update the Readme?

aymanbagabas, 2019-01-03:
> I just tested it today with a Gnome installation instead of i3 and it turned out that all keys are working now like it was expected. Means, that also the keys for the display light and the display [...]

I'm glad that worked for you. I already merged the changes to the master branch. Also I have created dkms packages for Fedora, Debian based, and Arch pkgbuild that you can use instead of compiling the module yourself. This module along with the micmute led have been merged to the next linux kernel release.

janbaer, 2019-01-04:
Great, thank you for your effort! It's very helpful for a lot people. Personally I like the concept to have a notebook without any active cooling and the MBX works perfectly for me since a year.

closed 2019-01-10 by aymanbagabas