aymanbagabas/Huawei-WMI #5: Can't run make command

lenngro, 2018-12-19:
Hi,
I have the following problem when running the `make` command:

```
arch/x86/Makefile:150: CONFIG_X86_X32 enabled but no binutils support
arch/x86/Makefile:227: *** You are building kernel with non-retpoline compiler, please update your compiler..  
```
I'm running Manjaro with the 4.19 Linux Kernel.
Is there any requirement to the linux kernel used?

aymanbagabas, 2018-12-19:
> Hi,
> I have the following problem when running the `make` command:
>
> ```
> arch/x86/Makefile:150: CONFIG_X86_X32 enabled but no binutils support
> arch/x86/Makefile:227: *** You are building kernel with [...]

Hmm.. It's complaining about `binutils`. Try to install it and see what happens. `pacman -S binutils`
There is no requirements, this should work with any recent version.

lenngro, 2018-12-19:
I was missing the `sudo` before `make` . Sometimes it's the little things.
After running `sudo make` and `sudo make install` afterwards + reboot the hotkeys are working.
Thank you so much for this awesome driver!

closed 2018-12-19 by lenngro