aymanbagabas/Huawei-WMI #6: Make install failed

janbaer, 2019-02-02:
After reinstalling my Matebook I also had to apply your patch again. But it failed with the following error

```
╰─➤ sudo make install
make -C /lib/modules/4.19.16-1-MANJARO/build/ M=/home/jan/Projects/github/Huawei-WMI modules_install
make[1]: Entering directory '/usr/lib/modules/4.19.16-1-MANJARO/build'
  INSTALL /home/jan/Projects/github/Huawei-WMI/huawei-wmi.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.19.16-1-MANJARO
make[1]: Leaving directory '/usr/lib/modules/4.19.16-1-MANJARO/build'
depmod -a
```
Do you have any idea what's missing and how I can solve the problem? I use Manjaro 18.1 with the 4.19.16.1 kernel. Thanks!

aymanbagabas, 2019-02-02:
I don’t see any errors with the build process, it’s complaining about missing keys. Make sure you’ve disabled secured boot in your BIOS and have ca-certificates installed. Try `lsmod | grep wmi` and see if the module is loaded.
You could also try using PKGBUILD from [here](
https://github.com/aymanbagabas/Huawei-WMI/releases).

janbaer, 2019-02-03:
I had installed the **ca-certificate**. But probably the kernel requires to sign drivers before install.
As alternative I tried now to install it with using dkms and this was running without any problems and after a reboot all fn keys worked perfectly. Thanks!

So with the help of [here](https://github.com/lidel/linux-on-huawei-matebook-x-2017/issues/8) I have now also sound from both speakers.

There's only one last thing, which we can't use on Linux. It's the finger print sensor for authentification and I'm very curious if this will also work some day.

closed 2019-02-03 by janbaer