aymanbagabas/Huawei-WMI #10: Multiple keys not working on Huawei Matebook D 14" AMD

owenthewizard, 2019-04-17:
Brightness keys and volume keys including micmute don't work. Keyboard backlight key works and Wi-Fi key works. Not sure what `F8` and `F10` are supposed to do so not sure if they're working or not.

Included is all the info from `acpidump`. Let me know what else you need and I'll provide it.

`uname` string: `Linux Quayle 5.0.7-arch1-1-ARCH #1 SMP PREEMPT Mon Apr 8 10:37:08 UTC 2019 x86_64 GNU/Linux`
[Huawei-Matebook-D-14-AMD.tar.gz](https://github.com/aymanbagabas/Huawei-WMI/files/3091924/Huawei-Matebook-D-14-AMD.tar.gz)

aymanbagabas, 2019-04-17:
Can you please provide the output for `acpi_listen`, `xev`, `lsmod | grep wmi`, and `sudo evtest`. When using `evtest`, choose event number corresponding to 'Huawei WMI hotkeys'.

owenthewizard, 2019-04-18:
`acpi_listen`, `F8` didn't display anything, micmute is `button/f20`, `F10` is `ABBC0F5C`
```
video/brightnessdown BRTDN 00000087 00000000
 ABBC0F5C-8EA1-11 000000a0 00000000
video/brightnessdown BRTDN 00000087 00000000 K
video/brightnessup BRTUP 00000086 00000000
 ABBC0F5C-8EA1-11 000000a0 00000000
video/brightnessup BRTUP 00000086 00000000 K
button/mute MUTE 00000080 00000000 K
button/volumedown VOLDN 00000080 00000000 K
button/volumeup VOLUP 00000080 00000000 K
button/f20 F20 00000080 00000000 K
 ABBC0F5C-8EA1-11 000000a0 00000000
```

[`xev.txt.gz`](https://github.com/aymanbagabas/Huawei-WMI/files/3092289/xev.txt.gz)

`lsmod | grep wmi`:
```
huawei_wmi             16384  0
sparse_keymap          16384  1 huawei_wmi
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek
wmi_bmof               16384  0
wmi                    32768  2 huawei_wmi,wmi_bmof
```

[`evtest.txt.gz`](https://github.com/aymanbagabas/Huawei-WMI/files/3092298/evtest.txt.gz)
`evtest` didn't show any output for volume up/down/mute.

aymanbagabas, 2019-04-18:
Seem working to me!

Brightness keys and multimedia keys seem to work as expected as `acpi_listen` shows. `xev` shows `XF86AudioMicMute` which means you're using systemd >= 241 that has the micmute mapping, F20 -> `XF86AudioMicMute`. `xev` also shows `Super+p` which is how F8 is being handled in this model and others. `Super+p` in Windows, and most Linux DEs work the same way, it cycles through available display options. F10 is mapped to `XF86Tools` which works with most DEs also in `xev` log.

owenthewizard, 2019-04-18:
I'm using `i3`, I guess that'd be it. Sorry to trouble you.

closed 2019-04-18 by owenthewizard

aymanbagabas, 2019-04-18:
> I'm using `i3`, I guess that'd be it. Sorry to trouble you.

No trouble at all.

You could try to remap those keys so that `i3` recognizes them. For the brightness keys to work, try using some sort of power manager like `xfce4-power-manager`, it will detect those key events accordingly. 

Idk what you're using for key remapping but you could use `xmodmap` or `acpid` to get the multimedia keys to work. Gnome detects ACPI events which is how I got the multimedia keys to work on my machine. These keys, according to `evtest` then 'AT keyboard, have keycodes from 113-115. You could do the same to the rest.

codeitlikemiley, 2019-11-14:
is there a way we can bind other keys to f10 , its being registered as Super+p
I have other key bindings for Super P and i wanna use this f10 key by huawei for activating and deactivating my keyboard since only f1-12 works when disabling my keyboard that goes with all the huawei fb keys...
please let me know if i can bind it to other keys thanks

aymanbagabas, 2019-11-14:
@codeitlikemiley F10 should map to `XF86Tools` and yes you can remap it using anything i.e. `xmodmap`. Unfortunately, the F8 key (display switch) cannot be remapped.

codeitlikemiley, 2019-11-14:
Yeah I just resorted to remapping f10 instead of f8...  And remap the old
keys to another one.
I really need this f8 or f10 since that's the only keys not being disabled
when I disabled keyboard except of course the key brightness and screen...
Anyway thanks for reply