aymanbagabas/Huawei-WMI #9: Use WMI calls

aymanbagabas, 2019-04-17:
Most of Huawei devices have the AMW0 WMI device which has a GUID of `ABBC0F5C-8EA1-11D1-A000-C90629100000`. This WMI device has a method AMW0.WMAA with GUID `ABBC0F5B-8EA1-11D1-A000-C90629100000` which seems to control most of the features.

Right now, one of controlling the mic led only works with devices that implement method `\_SB.PCI0.LPCB.EC0.WPIN` and `\_SB.PCI0.LPCB.EC0.SPIN`. Models that don't have one these methods simply don't work! Like #7 

Using WMI calls, specifically AMW0.WMAA, #7 could be solved and #8 could be integrated. The only model that might not work with this update is the Matebook X (2017).

aymanbagabas, 2019-04-17:
I might need updated data from Matebook X (2017). @janbaer could you please provide me with DSDT/SSDT tables? Just make sure you're on the latest BIOS update. Also, do you know if this model supports battery protection and fn-lock?
Thank you!

aymanbagabas, 2019-04-23:
Battery protection and fn-lock were added and can be controlled through the driver. You can access these features under `/sys/devices/platform/huawei-wmi`. There you'd find three newly attributes, `charge_start_threshold`, `charge_stop_threshold`, and `fn_lock_state`.

Charge thresholds accept values from 0-100 and fn-lock takes either 0 or 1. These attributes directly access AML code using WMI commands, specifically method `ABBC0F5B-8EA1-11D1-A000-C90629100000`.

This method is found in most Huawei models. However, some models have other methods as their WMI BIOS controller such as Matebook X (2017). From what I have, this model supports fn-lock control using this method but not battery or micmute led.

There are many other methods that this WMI device can control but I haven't found what they're for yet.

janbaer, 2019-04-28:
@aymanbagabas I'm sorry but I just read this issue today. Do you still need the DSDT/SSDT tables.
Could you please give me a hint how to list it? How about the BIOS update? To be honest, I have never checked if there's an update for the BIOS at all and I'm also not sure how to update it from a Linux system. I assume, if there's an update, that there's only a Windows supported updating app provided by Huawei.

nekr0z, 2019-04-28:
> To be honest, I have never checked if there's an update for the BIOS at all and I'm also not sure how to update it from a Linux system. I assume, if there's an update, that there's only a Windows supported updating app provided by Huawei.

I con't know which model you're running, but for MateBook 13 it [has been figured out](https://github.com/nekr0z/linux-on-huawei-matebook-13-2019#bios-updates). Probably works on other MateBooks, too (if you use correct device GUID for your model), but I don't think anyone tested it.

aymanbagabas, 2019-04-29:
> To be honest, I have never checked if there's an update for the BIOS at all and I'm also not sure how to update it from a Linux system. I assume, if there's an update, that there's only a Windows supported updating app provided by Huawei.

All huawei laptops that I have seen have a common BIOS WMI device controller. Except the old matebook, it has this common device but it also has another one that controls most of the features. I wanna see if the latest BIOS adds support for this common device so that we don’t need to implement the “legacy” controller.
If you have any questions I’ll be happy to help.

> I don't know which model you're running, but for MateBook 13 it [has been figured out](https://github.com/nekr0z/linux-on-huawei-matebook-13-2019#bios-updates). Probably works on other MateBooks, too (if you use correct device GUID for your model), but I don't think anyone tested it.

You’re awesome. I tested it and it worked on my x pro. You know that when you edit the .fd file in a hex editor there is a config section that you can change allowing you to do things like downgrading the bios. I’m wondering if anyone tried that

aymanbagabas, 2019-05-02:
@janbaer if you're still interested in helping to implement the "legacy" BIOS interface that's found on laptops such as Matebook X (2017) I will be glad to work with you. Right now, with tag `v3`, micmute LED wouldn't work on Matebook X (2017).

aymanbagabas, 2019-05-02:
Right now, I'm trying to figure out a way to detect laptop available features. I created a simple bash script that uses `acpi_call` to execute native AML methods. I have noticed lagging in my system after the script finished executing. This is the script if anyone is interested. [testAML.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3140136/testAML.txt)

From the output of this script, I've come to this AML method `\GTSI` I don't know what it stands for but it might be the one we're looking for since it holds static fixed values. From the data I have (DSDTs) for the XPro, 13 (2019), and D-AMD, I've noticed that each one of these laptops has similar return values from this method.

| Laptop | hex | bin |
| ---         | ---        | ---       |
| XPro     | 0xe3, 0x41, 0xe0, 0x07 | 0b**1**1**1**0001**1**, 0b0**1**00000**1**, 0b11100000, 0b00000111 |
| 13         | 0xe3, 0x49 | 0b**1**1**1**0001**1**, 0b0**1**00100**1** |
| D AMD | 0xa1, 0x49 | 0b**1**0**1**0000**1**, 0b0**1**00100**1** |

Maybe these values mean something maybe not. Any ideas?

janbaer, 2019-05-04:
@aymanbagabas I'm sorry for the late answer. Yes the Matebook X 2017 has also fn-lock and it's also working for me (if me mean the same). The mic-mute fn  key is actually not working, but till today I didn't miss it, because I never used it. About battery protection I have no idea, what feature it's.

Unfortunately I still don't know how to generate the DSDT/SSDT tables you asked for. But how about this one from @cytrinox https://github.com/cytrinox/matebook-hotkeys/tree/master/acpi

aymanbagabas, 2019-05-04:
> @aymanbagabas I'm sorry for the late answer. Yes the Matebook X 2017 has also fn-lock and it's also working for me (if me mean the same). The mic-mute fn key is actually not working, but till today [...]

Did you remap the micmute key? Fn-lock works by inverting the behavior of the function keys. It's not when you have the Fn led on. With Fn-lock, and Fn led is on, the function keys would be the multimedia keys **NOT** F1-F12. For battery protection, it sets charging limits so it protects the battery from wearing out.

> Unfortunately I still don't know how to generate the DSDT/SSDT tables you asked for. But how about this one from @cytrinox https://github.com/cytrinox/matebook-hotkeys/tree/master/acpi

It is unlikely that the vendor updated the BIOS with the new management interface that's equipped with their newer devices. I just want to make sure that that's the case so we don't have to implement the "legacy" interface. You could get these tables using `sudo acpidump > acpidump.out` acpidump.out should contain all the needed tables just make sure you're on the **latest** BIOS update. Thank you!

aymanbagabas, 2019-05-30:
Hi guys! @nekr0z @wasakakero @janbaer
I've updated the driver to include a debugfs interface for future debugging purposes and made tons of refactors. Right now, I'm preparing to submit this version to mainline kernel so I would need some help verifying all the functions work as expected.

Please test the following while watching dmesg for any errors, you could use `dmesg -w`. If you could also provide the output of dmesg and the following.
- Hotkeys (special keys, fn keys, etc)
- Set micmute LED through the driver (workaround using ACPI EC on MateBook X 2017)
`echo 0 | sudo tee /sys/devices/platform/huawei-wmi/leds/platform::micmute/brightness`
`echo 1 | sudo tee /sys/devices/platform/huawei-wmi/leds/platform::micmute/brightness`
- Set/get battery charging threshold values (doesn't exist in MateBook X 2017)
`echo "40 70" | sudo tee /sys/devices/platform/huawei-wmi/charge_thresholds`
`cat /sys/devices/platform/huawei-wmi/charge_thresholds`
`echo "0 100" | sudo tee /sys/devices/platform/huawei-wmi/charge_thresholds`
`cat /sys/devices/platform/huawei-wmi/charge_thresholds`
- Set/get fn-lock (Should now work on MateBook X 2017 NEED TESTING!!)
`echo 0 | sudo tee /sys/devices/platform/huawei-wmi/fn_lock_state`
`cat /sys/devices/platform/huawei-wmi/fn_lock_state`
`echo 1 | sudo tee /sys/devices/platform/huawei-wmi/fn_lock_state`
`cat /sys/devices/platform/huawei-wmi/fn_lock_state`
- Debugfs interface
`echo 0xffff | sudo tee /sys/kernel/debug/huawei-wmi/arg`
`sudo cat /sys/kernel/debug/huawei-wmi/arg`
`sudo cat /sys/kernel/debug/huawei-wmi/call`
`echo 0x010b04 | sudo tee /sys/kernel/debug/huawei-wmi/arg` (micmute LED on)
`sudo cat /sys/kernel/debug/huawei-wmi/arg`
`sudo cat /sys/kernel/debug/huawei-wmi/call`
`echo 0x0b04 | sudo tee /sys/kernel/debug/huawei-wmi/arg` (micmute LED off)
`sudo cat /sys/kernel/debug/huawei-wmi/arg`
`sudo cat /sys/kernel/debug/huawei-wmi/call`

I would also need someone with a MateBook X (2017) to test this and/or to help with implementing the "legacy" interface, @nekr0z @wasakakero if you guys know someone it would be really really appreciated! This is one of those times where your code isn't complete but works. As a dev, I'm really really annoyed that this part isn't complete because I don't have the hardware and can't use debugging tools to figure it out since this old freaking interface has parts tied to the hardware level that won't be emulated with software such as LED stuff.

Finally, thank you guys for your contributions toward this project, without you guys, and the latest news regarding Huawei and the US :-1:, we wouldn't get this far with the driver.

nekr0z, 2019-05-30:
OK, here's for MateBook 13:
* Driver compiles and installs as expected.

> * Hotkeys (special keys, fn keys, etc)

Works as expected, no issues, all correct.

> * Set micmute LED through the driver (workaround using ACPI EC on MateBook X 2017)

Works as expected.

> * Set/get battery charging threshold values (doesn't exist in MateBook X 2017)

Again, works as expected.

> * Set/get fn-lock (Should now work on MateBook X 2017 NEED TESTING!!)

What do you think, works as expected!

> * Debugfs interface

Well, here we go...
```
$ echo 0xffff | sudo tee /sys/kernel/debug/huawei-wmi/arg
0xffff
$ sudo cat /sys/kernel/debug/huawei-wmi/arg
0x000000000000ffff
$ sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]
```
Not sure whether this is expected or not, but let's go on...
```
$ echo 0x010b04 | sudo tee /sys/kernel/debug/huawei-wmi/argnel/debug/huawei-wmi/arg
0x010b04
$ sudo cat /sys/kernel/debug/huawei-wmi/arg
0x0000000000010b04
$ sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]
```
The last command here brings micmute led on (I kinda expected the first command would, but what do I know). From this moment on, micmute led doesn't react to actual ALSA micmute settings and can only be switched on or off manually through the driver, not sure if this is expected either.
```
$ echo 0x0b04 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x0b04
$ sudo cat /sys/kernel/debug/huawei-wmi/arg
0x0000000000000b04
$ sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]
```
Again, the last command (not the first one) kills micmute led.

After `sudo rmmod huawei_wmi && sudo modprobe huawei-wmi` micmute led returns to expected functionality (i.e. reacting not only to manual setting via driver but to ALSA micmute state as well.

All along, nothing in dmesg other than initial registering Huawei WMI hotkeys as input.

nekr0z, 2019-05-30:
> I would also need someone with a MateBook X (2017) to test this and/or to help with implementing the "legacy" interface, @nekr0z @wasakakero if you guys know someone it would be really really appreciated!

@lidel, care to join in?

aymanbagabas, 2019-06-24:
@cytrinox @thisisshi care to join?

jannis3005, 2020-01-21:
@aymanbagabas sorry, i didnt see that you also added the matebook-e label here. Can i still help with this issue?

aymanbagabas, 2020-01-23:
> @aymanbagabas sorry, i didnt see that you also added the matebook-e label here. Can i still help with this issue?

Yes, the Matebook E has the same interface. However, unlike the Matebook X, keyboard keys are not handled through this WMI interface, it uses USBHID instead.

You can see from the files attached that the WMI event handler has the same guid that the Matebook X has `39142400-C6A3-40fa-BADB-8A2652834100`.

<details>
<summary>MBE WMI mof</summary>

```
[WMI, Dynamic, Provider("WmiProv"), Locale("MS\\0x409"), Description("MyWMIEventCodeClass used for Query EventCode"), guid("{39142400-C6A3-40fa-BADB-8A2652834100}")]
class MyWMIEventCodeClass {
  [key, read] String InstanceName;
  [read] Boolean Active;
  [WmiDataId(1), read, Description("Get EventCode")] uint32 EVCD;
};

[WMI, Dynamic, Provider("WmiProv"), Locale("MS\\0x409"), Description("MyWMIDataClass1 used for WMI Argument1"), guid("{49142400-C6A3-40fa-BADB-8A2652834100}")]
class MyWMIARGClass1 {
  [key, read] String InstanceName;
  [read] Boolean Active;
  [WmiDataId(1), read, write, Description("used for passing WMI Argumnts")] uint8 WAG1[256];
};

[WMI, Dynamic, Provider("WmiProv"), Locale("MS\\0x409"), Description("MyWMIPayloadClass1 used for WMI Payload1"), guid("{4A142400-C6A3-40fa-BADB-8A2652834100}")]
class MyWMIPayloadClass1 {
  [key, read] String InstanceName;
  [read] Boolean Active;
  [WmiDataId(1), read, write, Description("used for passing WMI Payloads")] uint8 WPD1[4096];
};

[WMI, Dynamic, Provider("WmiProv"), Locale("MS\\0x409"), Description("MyWMIDataClass2 used for WMI Argument2"), guid("{49142401-C6A3-40fa-BADB-8A2652834100}")]
class MyWMIARGClass2 {
  [key, read] String InstanceName;
  [read] Boolean Active;
  [WmiDataId(1), read, write, Description("used for passing WMI Argumnts")] uint8 WAG2[256];
};

[WMI, Dynamic, Provider("WmiProv"), Locale("MS\\0x409"), Description("MyWMIPayloadClass2 used for WMI Payload2"), guid("{4A142401-C6A3-40fa-BADB-8A2652834100}")]
class MyWMIPayloadClass2 {
  [key, read] String InstanceName;
  [read] Boolean Active;
  [WmiDataId(1), read, write, Description("used for passing WMI Payloads")] uint8 WPD2[4096];
};

[WMI, Dynamic, Provider("WmiProv"), Locale("MS\\0x409"), Description("MyWMIDataClass3 used for WMI Argument3"), guid("{49142402-C6A3-40fa-BADB-8A2652834100}")]
class MyWMIARGClass3 {
  [key, read] String InstanceName;
  [read] Boolean Active;
  [WmiDataId(1), read, write, Description("used for passing WMI Argumnts")] uint8 WAG3[256];
};

[WMI, Dynamic, Provider("WmiProv"), Locale("MS\\0x409"), Description("MyWMIPayloadClass3 used for WMI Payload3"), guid("{4A142402-C6A3-40fa-BADB-8A2652834100}")]
class MyWMIPayloadClass3 {
  [key, read] String InstanceName;
  [read] Boolean Active;
  [WmiDataId(1), read, write, Description("used for passing WMI Payloads")] uint8 WPD3[4096];
};

[WMI, Dynamic, Provider("WmiProv"), Locale("MS\\0x409"), Description("MyWMIDataClass4 used for WMI Argument4"), guid("{49142403-C6A3-40fa-BADB-8A2652834100}")]
class MyWMIARGClass4 {
  [key, read] String InstanceName;
  [read] Boolean Active;
  [WmiDataId(1), read, write, Description("used for passing WMI Argumnts")] uint8 WAG4[256];
};

[WMI, Dynamic, Provider("WmiProv"), Locale("MS\\0x409"), Description("MyWMIPayloadClass4 used for WMI Payload4"), guid("{4A142403-C6A3-40fa-BADB-8A2652834100}")]
class MyWMIPayloadClass4 {
  [key, read] String InstanceName;
  [read] Boolean Active;
[... 1766 lines cut ...]
      Name=write
      Tosubclass=FALSE
      Type=Boolean
      Value=TRUE
    Qualifier 4:
      Name=Description
      Tosubclass=FALSE
      Type=String
      Value=Execute the WDOG
    Return value:
      Type=Void
    Parameter 0:
      Direction=in
      Name=Arg0
      Type=Basic:uint32
    Parameter 1:
      Direction=out
      Name=REST
      Type=Basic:uint32
```

</details>

[mbemof.txt](https://github.com/aymanbagabas/Huawei-WMI/files/4103989/mbemof.txt)
[mbeparse.txt](https://github.com/aymanbagabas/Huawei-WMI/files/4103990/mbeparse.txt)