aymanbagabas/Huawei-WMI #12: Im using latest kernel and archlinux

codeitlikemiley, 2019-05-18:
Ived read that this was included in the latest kernel update
if you using 5.0, 
Im using huawei matebook 13 , i try to check using lsmod | grep wmi

i got this
```
mxm_wmi                16384  1 nouveau
huawei_wmi             16384  0
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek
sparse_keymap          16384  1 huawei_wmi
intel_wmi_thunderbolt    20480  0
wmi_bmof               16384  0
wmi                    32768  5 intel_wmi_thunderbolt,huawei_wmi,wmi_bmof,mxm_wmi,nouveau

```
Some problems im facing is that i cannot detect a headset when plugged in.
also i wanna know i can use the battery protection 
Also ive read there is a matebook applet but its not running i dont see it on my status bar
May i know the command to run it on start , so i can include it on my i3 config

Things that are not working also is the screen brightness i cannot adjust it,
xbacklight command wont work

```
bindsym XF86MonBrightnessDown	exec --no-startup-id xbacklight -dec 15 ;; exec --no-startup-id notify-send "Decrease Backlight." #f1
bindsym XF86MonBrightnessUp	exec --no-startup-id xbacklight -inc 15 ;; exec --no-startup-id notify-send "Increase Backlight."  #f2

```
Note im hitting the correct keys if i do try to echo some text on the notification it is showing it when i press the brightness keys

Wireless button on off works ok

dont know if the mic mute is working on my end though
but i can confirm i can use the command on my i3 conf 
```
bindsym XF86AudioMicMute		exec $micmute ;; exec --no-startup-id notify-send "Mic Toggled." #f7
```
the f8 +fn works odd cause it makes my music player pause and play , but i can live without the keybindings on that if that is for display i do have it bind on other keys as such

```
bindsym $mod+F3		exec --no-startup-id displayselect
```

But the fn+f10 which is the Huawei manager button which i guess is for the huawei matebook applet is not working...

Hope you can help me with this thanks

codeitlikemiley, 2019-05-18:
i manage to control the mic on off
by adding this to my i3 conf
```
bindsym XF86AudioMicMute		exec --no-startup-id amixer set Capture toggle && amixer get Capture | grep '\[off\]' && notify-send "MIC switched OFF" || notify-send "MIC switched ON" #f7
```

hope you can help me with screen brightness problem
cause if i use the xbacklight command 
it just returns me this message 
```
No outputs have backlight property
```

aymanbagabas, 2019-05-19:
To use battery protection, you have to compile the latest driver from master and set charging thresholds from /sys/devices/platform/huawei-wmi/charge_thresholds as described in the readme.

[matebook-applet](https://github.com/nekr0z/matebook-applet) is another project by @nekr0z. This is not the same project!

Brightness is handled through acpi_video. AMOF this driver would cause reporting brightness twice to OS, I'm working on a fix for this. You can try another DE like KDE or Gnome to get a fully working environment.

What is the output of `acpi_listen` and `sudo evtest` of 'Huawei WMI hotkeys' and 'Video Bus'? Try using something like acpilight or light instead of xbacklight.

F8 (DisplaySwitch) is actually not a hotkey, Huawei oddly mapped it to `super+p` which in Windows pops up the display switch sidebar and works the same in most Linux DEs at least on Gnome.

The driver has to be active to use F10 (Huawei management), it maps it to XF86Tools so if i3 picks that, you could bind it to anything else. By default, in Gnome and I believe KDE, XF86Tools opens up settings.

codeitlikemiley, 2019-05-20:
i have no luck compiling your repo
```

make -C /lib/modules/5.1.2-arch1-1-ARCH/build/ M=/home/uriah/Packages/Huawei-WMI modules
make[1]: Entering directory '/usr/lib/modules/5.1.2-arch1-1-ARCH/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.1.2-arch1-1-ARCH/build'
make: *** [Makefile:7: modules] Error 2
```

but i do manage to make a work around on the screen brightness
ive installed brightnessctl 
also the  audio jack is detected but needs to be switch manually thru pavucontrol 
btw heres my i3 conf
i did install some scripts from matebook 13 applet project like the batpro and fnlock
```
# #-- FN KEYS FOR MY KEYBOARD --#
# # check this for guide : http://wiki.linuxquestions.org/wiki/XF86_keyboard_symbols --#
bindsym fn exec --no-startup-id sudo fnlock toggle
bindsym XF86MonBrightnessDown	exec --no-startup-id sudo brightnessctl set 5%-
bindsym XF86MonBrightnessUp	exec --no-startup-id sudo brightnessctl set +5%
# Huawei WMI specific Keyboard Brightness  # f3
bindsym XF86AudioMute		exec --no-startup-id lmc mute #f4
bindsym XF86AudioLowerVolume	exec --no-startup-id lmc down 1 #f5
bindsym XF86AudioRaiseVolume	exec --no-startup-id lmc up 1 #f6
bindsym XF86AudioMicMute		exec --no-startup-id amixer set Capture toggle && amixer get Capture | grep '\[off\]' && notify-send "MIC switched OFF" || notify-send "MIC switched ON" #f7
# Same key bindings to Super +P # f8
# works out of the box on/off WIFI #f9
# bindsym XF86Tools exec notify-send "power" # testing 
# PrintScreen # f11
# Ins # f12
```

aymanbagabas, 2019-05-20:
> i have no luck compiling your repo
>
> ```
>
> make -C /lib/modules/5.1.2-arch1-1-ARCH/build/ M=/home/uriah/Packages/Huawei-WMI modules
> make[1]: Entering directory [...]

Please follow the readme for compiling the driver.

```
$ git clone https://github.com/aymanbagabas/Huawei-WMI
$ cd Huawei-WMI
$ make
$ sudo cp huawei-wmi.ko /lib/modules/$(uname -r)/updates/
$ sudo depmod
$ reboot
```

> but i do manage to make a work around on the screen brightness
> ive installed brightnessctl
> also the audio jack is detected but needs to be switch manually thru pavucontrol
> btw heres my i3 conf
> i did [...]

fn lock is different from fn toggled (led on) be aware of that!

codeitlikemiley, 2019-05-20:
I believe im missing some dependencies which is not explicitly define on the readme.
i tried on my other arch laptop , not huawei but ...
i tried to install archlinux lts kernel
```
sudo pacman -S linux-lts
sudo grub-mkconfig -o /boot/grub/grub.cfg
sudo pacman -S linux-lts-headers
reboot
```
now my kernel is an lts kernel

I did try to cd to Huawei-WMI but it failed again, maybe because the laptop i used was not huawei matebook 13.

but it produces a different error now

```
make -C /lib/modules/4.19.44-1-lts/build/ M=/home/yuri/Programs/Huawei-WMI modules
make[1]: Entering directory '/usr/lib/modules/4.19.44-1-lts/build'
  CC [M]  /home/yuri/Programs/Huawei-WMI/huawei-wmi.o
In file included from ./include/linux/list.h:9,
                 from ./include/linux/resource_ext.h:17,
                 from ./include/linux/acpi.h:26,
                 from /home/yuri/Programs/Huawei-WMI/huawei-wmi.c:8:
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c: In function 'huawei_wmi_eval':
./include/linux/kernel.h:845:29: warning: comparison of distinct pointer types lacks a cast
   (!!(sizeof((typeof(x) *)1 == (typeof(y) *)1)))
                             ^~
./include/linux/kernel.h:859:4: note: in expansion of macro '__typecheck'
   (__typecheck(x, y) && __no_side_effects(x, y))
    ^~~~~~~~~~~
./include/linux/kernel.h:869:24: note: in expansion of macro '__safe_cmp'
  __builtin_choose_expr(__safe_cmp(x, y), \
                        ^~~~~~~~~~
./include/linux/kernel.h:878:19: note: in expansion of macro '__careful_cmp'
 #define min(x, y) __careful_cmp(x, y, <)
                   ^~~~~~~~~~~~~
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c:175:9: note: in expansion of macro 'min'
   len = min(buflen, obj->buffer.length);
         ^~~
In file included from ./include/uapi/linux/posix_types.h:5,
                 from ./include/uapi/linux/types.h:14,
                 from ./include/linux/compiler.h:174,
                 from ./include/linux/ioport.h:13,
                 from ./include/linux/acpi.h:25,
                 from /home/yuri/Programs/Huawei-WMI/huawei-wmi.c:8:
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c: In function 'huawei_wmi_micmute_led_set':
./include/linux/stddef.h:8:14: warning: passing argument 5 of 'huawei_wmi_cmd' makes integer from pointer without a cast [-Wint-conversion]
 #define NULL ((void *)0)
              ^~~~~~~~~~~
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c:377:66: note: in expansion of macro 'NULL'
   huawei_wmi_cmd(led_cdev->dev->parent, MICMUTE_LED, &arg, NULL, NULL);
                                                                  ^~~~
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c:185:106: note: expected 'size_t' {aka 'long unsigned int'} but argument is of type 'void *'
 static int huawei_wmi_cmd(struct device *dev, enum wmaa_cmd cmd, struct amw0_arg *arg, void *out, size_t outlen)
                                                                                                   ~~~~~~~^~~~~~
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c: In function 'huawei_wmi_leds_setup':
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c:389:28: error: implicit declaration of function 'ledtrig_audio_get'; did you mean 'led_trigger_set'? [-Werror=implicit-function-declaration]
  huawei->cdev.brightness = ledtrig_audio_get(LED_AUDIO_MICMUTE);
                            ^~~~~~~~~~~~~~~~~
                            led_trigger_set
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c:389:46: error: 'LED_AUDIO_MICMUTE' undeclared (first use in this function); did you mean 'KEY_MICMUTE'?
  huawei->cdev.brightness = ledtrig_audio_get(LED_AUDIO_MICMUTE);
                                              ^~~~~~~~~~~~~~~~~
                                              KEY_MICMUTE
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c:389:46: note: each undeclared identifier is reported only once for each function it appears in
In file included from ./include/uapi/linux/posix_types.h:5,
                 from ./include/uapi/linux/types.h:14,
                 from ./include/linux/compiler.h:174,
                 from ./include/linux/ioport.h:13,
                 from ./include/linux/acpi.h:25,
[... 9 lines cut ...]
 static int huawei_wmi_cmd(struct device *dev, enum wmaa_cmd cmd, struct amw0_arg *arg, void *out, size_t outlen)
                                                                                                   ~~~~~~~^~~~~~
In file included from ./include/uapi/linux/posix_types.h:5,
                 from ./include/uapi/linux/types.h:14,
                 from ./include/linux/compiler.h:174,
                 from ./include/linux/ioport.h:13,
                 from ./include/linux/acpi.h:25,
                 from /home/yuri/Programs/Huawei-WMI/huawei-wmi.c:8:
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c: In function 'huawei_wmi_fn_lock_set':
./include/linux/stddef.h:8:14: warning: passing argument 5 of 'huawei_wmi_cmd' makes integer from pointer without a cast [-Wint-conversion]
 #define NULL ((void *)0)
              ^~~~~~~~~~~
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c:473:54: note: in expansion of macro 'NULL'
  return huawei_wmi_cmd(dev, FN_LOCK_SET, &arg, NULL, NULL);
                                                      ^~~~
/home/yuri/Programs/Huawei-WMI/huawei-wmi.c:185:106: note: expected 'size_t' {aka 'long unsigned int'} but argument is of type 'void *'
 static int huawei_wmi_cmd(struct device *dev, enum wmaa_cmd cmd, struct amw0_arg *arg, void *out, size_t outlen)
                                                                                                   ~~~~~~~^~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:310: /home/yuri/Programs/Huawei-WMI/huawei-wmi.o] Error 1
make[1]: *** [Makefile:1524: _module_/home/yuri/Programs/Huawei-WMI] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.19.44-1-lts/build'
make: *** [Makefile:7: modules] Error 2

```

codeitlikemiley, 2019-05-20:
I fucked up, but i manage to make it work the make command.
I did remove the linux-lts kernel and the linus-lts-headers

reinstall linux and linux-headers

sudo grub-mkconfig -o /boot/grub/grub.cfg
reboot
then go to Huawei WMI folder
no errors

aymanbagabas, 2019-05-20:
> I fucked up, but i manage to make it work the make command.
> I did remove the linux-lts kernel and the linus-lts-headers
> 
> reinstall linux and linux-headers
> 
> sudo grub-mkconfig -o /boot/grub/grub.cfg
> reboot
> then go to Huawei WMI folder and make
> no errors

Glad everything worked out. The errors you were getting were because of kernel < 5.0.
Closing this issue for now.

closed 2019-05-20 by aymanbagabas

rien333, 2019-06-06:
@aymanbagabas my Arch install doesn't have `/lib/modules/$(uname -r)/updates/` (or rather, there is no "updates" folder). Should i create this updates folder?  Or should I maybe replace the old kernel module in `/usr/lib/modules/5.1.7-arch1-1-ARCH/kernel/drivers/platform/x86/`?

aymanbagabas, 2019-06-07:
@rien333 just create one or use dkms

rien333, 2019-11-05:
Note [this AUR package](https://aur.archlinux.org/packages/huawei-wmi-dkms/) seems to work fine. (maybe mention it in the readme?)