aymanbagabas/Huawei-WMI #13: /sys/devices/platform/huawei-wmi : No such fie or directory

codeitlikemiley, 2019-05-20:
I did manage to on your steps i did a reboot
but i dont know why i still dont have it , 
Dont know where i fucked up again

codeitlikemiley, 2019-05-20:
i tried to remove all my bindings in i3 to see if any keys would work after the patch
but majority of the fn keys wont work, i think the only thing that is working like before is the keyboard brightness and the wifi, the super +P .

the rest , the screen brightness cannot be control the mic led wount light up
the volumn wont work also.

so i have to put back all those keybindings which uses third party softwares to progmatically acrhive the desired behaviour when that key is press

aymanbagabas, 2019-05-20:
> i tried to remove all my bindings in i3 to see if any keys would work after the patch
> but majority of the fn keys wont work, i think the only thing that is working like before is the keyboard [...]

To be clear, this driver makes these function keys report events to the system where you MUST have a running daemon that triggers each event. On Gnome, this would be `gnome-settings-daemon`, KDE, Xfce, etc would have something similar. This usually listens to brightness key events and other events and calls the appropriate command. To get audio keys you have to use something like [volumeicon](https://github.com/Maato/volumeicon).

In your case, you have to either use these keybindings in i3 or use some sort of daemon that triggers appropriate event command. Try using Xfce power manager to get the brightness keys working without bindings.

i3 is NOT a DE (desktop environment), it's a window manager meaning the only thing it does is managing windows. A desktop environment consists of lots of components including a window manager and a settings daemon that listens to such key events.

I've uploaded a dkms source zip under releases [here](https://github.com/aymanbagabas/Huawei-WMI/releases/download/v3.1/huawei-wmi-3.1.zip). Please use this because it is more convenient to use and autoupdate.
```
$ sudo unzip huawei-wmi-3.1.zip -d /usr/src/
$ sudo dkms add huawei-wmi/3.1
$ sudo dkms install huawei-wmi/3.1
$ sudo dkms autoinstall --all
$ sudo reboot
```

codeitlikemiley, 2019-05-20:
i tried to use follow your instruction...
`$ sudo unzip huawei-wmi-3.1.zip -d /usr/src/
$ sudo dkms add huawei-wmi/3.1
$ sudo dkms install huawei-wmi/3.1
$ sudo dkms autoinstall --all
$ sudo reboot``

```
im stocked on `sudo dkms autoinstall --all`
i got this error
```
Error! echo
Your kernel headers for kernel  cannot be found at
/usr/lib/modules//build or /usr/lib/modules//source.

```

aymanbagabas, 2019-05-20:
> i tried to use follow your instruction...
> 
> ```
> $ sudo dkms add huawei-wmi/3.1
> $ sudo dkms install huawei-wmi/3.1
> $ sudo dkms autoinstall --all
> $ sudo reboot``
> ```
> 
> im stocked on `sudo dkms autoinstall --all`
> i got this error
> 
> ```
> Error! echo
> Your kernel headers for kernel  cannot be found at
> /usr/lib/modules//build or /usr/lib/modules//source.
> ```

That's fine you can skip these reboot and you should have everything working.

aymanbagabas, 2019-05-23:
Please close the issue if everything works as expected. Thank you

darthf1, 2019-05-26:
Came here having the same issue. Running Ubuntu 19.04 on a Matebook X Pro. Kernel 5 should be shipped with this driver but the folder was empty. So i manually installed the driver via dkms.
```
$ dkms status
huawei-wmi, 3.2, 5.0.0-15-generic, x86_64: installed
```
My folder `/sys/devices/platform/huawei-wmi` is still empty and i cannot run the applet because of it.
```
uname -a
Linux laptop 5.0.0-15-generic #16-Ubuntu SMP Mon May 6 17:41:33 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

aymanbagabas, 2019-05-26:
@darthf1 version v2.0 is the one shipped with kernel 5.0 but the newer one v3.0 that has battery protection is not, I'm working on submitting it to mainline hopefully to kernel 5.2. I should perhaps note that in the readme.
Please use the version from the repository. Thanks

aymanbagabas, 2019-06-06:
Closed due to inactivity.

closed 2019-06-06 by aymanbagabas

goldcoders, 2019-12-13:
HI im having issue also 
uname -r = 5.3.0-22-generic
```
╰ lsb_release -a 
No LSB modules are available. 
Distributor ID: Ubuntu 
Description:    Pop!_OS 19.10 
Release:        19.10 
Codename:       eoan 
   uriah@dev  ~/A/matebook-applet   master ●                                   ✔   02:26  14/12/19 

```
 im positive ive installed huawei-wmi
```
╰ sudo apt install huawei-wmi 
Reading package lists... Done 
Building dependency tree 
Reading state information... Done 
huawei-wmi is already the newest version (1.1.0). 
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded. 
   uriah@dev  ~/A/matebook-applet   master ●                                   ✔   02:19  14/12/19 

```

when i try to run matebook-applet it gives me this error

```
╰ matebook-applet -vv 
INFO: 2019/12/14 02:18:20 main.go:61: matebook-applet version 2.4.8 
TRACE: 2019/12/14 02:18:20 driver.go:123: Couldn't access "/sys/devices/platform/huawei-wmi/fn_lock_state".
TRACE: 2019/12/14 02:18:20 driver.go:229: Couldn't access "/sys/devices/platform/huawei-wmi/charge_control_thresholds". 
TRACE: 2019/12/14 02:18:20 driver.go:229: Couldn't access "/sys/devices/platform/huawei-wmi/charge_thresholds". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT0/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT1/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT2/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT3/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT4/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT5/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT6/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT7/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT8/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 driver.go:292: Couldn't access "/sys/class/power_supply/BAT9/charge_control_start_threshold". 
TRACE: 2019/12/14 02:18:20 main.go:71: looking for endpoint to save thresholds to... 
TRACE: 2019/12/14 02:18:20 driver.go:229: Couldn't access "/etc/default/huawei-wmi/charge_control_thresholds". 
TRACE: 2019/12/14 02:18:20 driver.go:229: Couldn't access "/etc/default/huawei-wmi/charge_thresholds". 
ERROR: 2019/12/14 02:18:20 main.go:91: Neither a supported version of Huawei-WMI driver, nor any of the required scripts are properly installed, see README.md#installation-and-setup for instructions 
   uriah@dev  ~/A/matebook-applet   master ●                                   ✔   02:18  14/12/19 

```

aymanbagabas, 2019-12-13:
@goldcoders What's your `lsmod|grep wmi` output? If you have problems with `matebook-applet` please head to that repo to submit tickets. The deb package doesn't seem to be the same one from this repo, please use the correct deb package.

goldcoders, 2019-12-13:
```
╰ lsmod | grep wmi 
snd_rawmidi            36864  1 snd_seq_midi 
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi 
snd                    86016  25 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi 
huawei_wmi             16384  0 
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek 
sparse_keymap          16384  1 huawei_wmi 
wmi_bmof               16384  0 
intel_wmi_thunderbolt    20480  0 
wmi                    32768  3 intel_wmi_thunderbolt,huawei_wmi,wmi_bmof 
   uriah@dev  ~                                      ✔   03:06  14/12/19 

```