aymanbagabas/Huawei-WMI #14: Keys and micmute led not working on MateBook 13 with v 3.1

nekr0z, 2019-05-20:
After update to driver v 3.1 additional keys (WiFi switch, micmute, PCmanager) and micmute LED stopped working on MateBook 13. Batpro and fnlock functionality still available.

aymanbagabas, 2019-05-20:
Hmm.. that's strange. Does it register the input driver? What's the output of `lsmod|grep wmi`, `dmesg|grep wmi`, and `sudo evtest`. Do you have `leds` under `/sys/devices/platform/huawei-wmi/`? Is there a `platform::micmute`?

aymanbagabas, 2019-05-20:
Could you try using DKMS from [here](https://github.com/aymanbagabas/Huawei-WMI/releases).
```
sudo unzip huawei-wmi-3.1.zip -d /usr/src/
sudo dkms add huawei-wmi/3.1
sudo dkms install huawei-wmi/3.1
```

nekr0z, 2019-05-21:
> Does it register the input driver?

Ermmm... Maybe. Not sure. How do I check? ;-)

> What's the output of `lsmod|grep wmi`

```
huawei_wmi             16384  1
wmi_bmof               16384  0
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek
sparse_keymap          16384  1 huawei_wmi
wmi                    32768  2 huawei_wmi,wmi_bmof
```
> dmesg|grep wmi
```
[   11.501949] huawei_wmi: loading out-of-tree module taints kernel.
[   11.502347] Modules linked in: huawei_wmi(OE+) wmi_bmof(E) intel_cstate(E) ledtrig_audio(E) intel_uncore(E) snd_timer(E) sparse_keymap(E) cfg80211(E) mei_me(E) snd(E) processor_thermal_device(E) iTCO_wdt(E) intel_rapl_perf(E) efivars(E+) pcspkr(E) iTCO_vendor_support(E) serio_raw(E) soundcore(E) spi_pxa2xx_platform(E+) idma64(E+) rfkill(E) mei(E) intel_soc_dts_iosf(E) intel_pch_thermal(E) ac(E) acpi_als(E) kfifo_buf(E) industrialio(E) battery(E) int3403_thermal(E) int340x_thermal_zone(E) tpm_crb(E) tpm_tis(E) tpm_tis_core(E) tpm(E) int3400_thermal(E) rng_core(E) int3406_thermal(E) acpi_thermal_rel(E) evdev(E) acpi_pad(E) pcc_cpufreq(E) parport_pc(E) ppdev(E) lp(E) parport(E) efivarfs(E) ip_tables(E) x_tables(E) autofs4(E) ext4(E) crc16(E) mbcache(E) jbd2(E) crc32c_generic(E) fscrypto(E) ecb(E) algif_skcipher(E) af_alg(E) dm_crypt(E) dm_mod(E) hid_generic(E) crct10dif_pclmul(E) crc32_pclmul(E) crc32c_intel(E) ghash_clmulni_intel(E) i915(E) aesni_intel(E) i2c_algo_bit(E) xhci_pci(E)
[   11.502368]  aes_x86_64(E) nvme(E) drm_kms_helper(E) crypto_simd(E) cryptd(E) glue_helper(E) xhci_hcd(E) i2c_i801(E) nvme_core(E) usbcore(E) intel_lpss_pci(E) drm(E) intel_lpss(E) usb_common(E) i2c_hid(E) thermal(E) wmi(E) hid(E) fan(E) video(E) button(E)
[   11.502419]  ? dmi_matched+0x19/0x19 [huawei_wmi]
[   11.502421]  ? huawei_wmi_init+0x31/0xfe7 [huawei_wmi]
[   11.502454] Modules linked in: huawei_wmi(OE+) wmi_bmof(E) intel_cstate(E) ledtrig_audio(E) intel_uncore(E) snd_timer(E) sparse_keymap(E) cfg80211(E) mei_me(E) snd(E) processor_thermal_device(E) iTCO_wdt(E) intel_rapl_perf(E) efivars(E+) pcspkr(E) iTCO_vendor_support(E) serio_raw(E) soundcore(E) spi_pxa2xx_platform(E+) idma64(E+) rfkill(E) mei(E) intel_soc_dts_iosf(E) intel_pch_thermal(E) ac(E) acpi_als(E) kfifo_buf(E) industrialio(E) battery(E) int3403_thermal(E) int340x_thermal_zone(E) tpm_crb(E) tpm_tis(E) tpm_tis_core(E) tpm(E) int3400_thermal(E) rng_core(E) int3406_thermal(E) acpi_thermal_rel(E) evdev(E) acpi_pad(E) pcc_cpufreq(E) parport_pc(E) ppdev(E) lp(E) parport(E) efivarfs(E) ip_tables(E) x_tables(E) autofs4(E) ext4(E) crc16(E) mbcache(E) jbd2(E) crc32c_generic(E) fscrypto(E) ecb(E) algif_skcipher(E) af_alg(E) dm_crypt(E) dm_mod(E) hid_generic(E) crct10dif_pclmul(E) crc32_pclmul(E) crc32c_intel(E) ghash_clmulni_intel(E) i915(E) aesni_intel(E) i2c_algo_bit(E) xhci_pci(E)
[   11.502473]  aes_x86_64(E) nvme(E) drm_kms_helper(E) crypto_simd(E) cryptd(E) glue_helper(E) xhci_hcd(E) i2c_i801(E) nvme_core(E) usbcore(E) intel_lpss_pci(E) drm(E) intel_lpss(E) usb_common(E) i2c_hid(E) thermal(E) wmi(E) hid(E) fan(E) video(E) button(E)
[   11.502517]  ? dmi_matched+0x19/0x19 [huawei_wmi]
[   11.502519]  ? huawei_wmi_init+0x31/0xfe7 [huawei_wmi]
```

> sudo evtest
```
sudo: evtest: command not found
```
:-D

Well, let's try again...
```
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Lid Switch
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      AT Translated Set 2 keyboard
/dev/input/event4:      Video Bus
/dev/input/event5:      HDA Digital PCBeep
/dev/input/event6:      HDA Intel PCH Headphone Mic
/dev/input/event7:      PC Speaker
/dev/input/event8:      HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event9:      HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event10:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event11:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event12:     HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event13:     ELAN962C:00 04F3:30D0 Touchpad
/dev/input/event14:     HD Camera: HD Camera
Select the device event number [0-14]:  
```
That's better...

> Do you have `leds` under `/sys/devices/platform/huawei-wmi/`?

And here's where the weird part starts. There is no `huawei-wmi` under `/sys/devices/platform/`. At all. 
```
$ ls /sys/devices/platform/
ACPI000C:00  coretemp.0         efivars.0   INT33A1:00  MSFT0101:00  PNP0C0B:01  PNP0C0B:04  PNP0C14:01  power         rtc_cmos       uevent
alarmtimer   dock.0             i8042       INT3400:00  pcspkr       PNP0C0B:02  PNP0C0C:00  PNP0C14:02  reg-dummy     serial8250     WDT0001:00
BTF0001:00   efi-framebuffer.0  INT0E0C:00  microcode   PNP0C0B:00   PNP0C0B:03  PNP0C14:00  PNP0C14:03  regulatory.0  snd-soc-dummy
```
However, matebook-applet successfully reads and writes... Oh! Oh! Disregard my previous comment on how batpro levels and fnlock work, they don't! The applet falls back to using scripts, that's what's happening.

> Could you try using DKMS from [here](https://github.com/aymanbagabas/Huawei-WMI/releases).

Hangs forever on `depmod...`

aymanbagabas, 2019-05-21:
Ahhh you got a kernel panic! Can I see you full `dmesg`? The module didn't get probed due to error.

nekr0z, 2019-05-21:
> Ahhh you got a kernel panic!

Yep, I looked into it more closely, and it's all shambles with 3.1: sometimes it wouldn't even boot, attempting to kill init, other times it wouldn't reboot correctly, requiring the SysRq REISUB dance (thank heavens for FnLock ;))) No such behaviour with 3.0.

> Can I see you full `dmesg`?

Why not, [here](https://github.com/aymanbagabas/Huawei-WMI/files/3201371/dmesg.txt) you go.

aymanbagabas, 2019-05-21:
Fixed in v3.2
Thank you!

closed 2019-05-21 by aymanbagabas

nekr0z, 2019-05-21:
Yep, works. Thank _you_! Keep up the awesome work!

ivan-infro, 2019-08-28:
@aymanbagabas I tried to install 'huawei-wmi/3.2' in three ways (rpm, tar.gz and build from source), but "/sys/devices/platform/huawei-wmi" doesn’t appear :(

`lsmod |grep wmi`
`intel_wmi_thunderbolt    20480  0`
`wmi_bmof               16384  0`
`wmi                    36864  2 intel_wmi_thunderbolt,wmi_bmof`

`dmesg|grep wmi`
nothing...

Huawei Matebook 13