aymanbagabas/Huawei-WMI #15: Won't compile on Huawei Matebook 13 following your installation guide

mariusrodi, 2019-05-22:
Hey,

I won't get it to work following your installation guide.
The errors are: 
```make
error: implicit declaration of function ‘ledtrig_audio_get’; did you mean ‘led_trigger_set’? [-Werror=implicit-function-declaration]
  huawei->cdev.brightness = ledtrig_audio_get(LED_AUDIO_MICMUTE);
                            ^~~~~~~~~~~~~~~~~
                            led_trigger_set
```
and:
```make
error: ‘LED_AUDIO_MICMUTE’ undeclared (first use in this function); did you mean ‘KEY_MICMUTE’?
  huawei->cdev.brightness = ledtrig_audio_get(LED_AUDIO_MICMUTE);
                                              ^~~~~~~~~~~~~~~~~
                                              KEY_MICMUTE
```
I am running Kernel Version 4.19.42-1-MANJARO on Manjaro Awesome.
My steps were:
- installing kernel headers
- cloning repo
- make --> Errors

Would appreciate it if you have the time to come up with a fix.
Thanks

aymanbagabas, 2019-05-22:
You're getting this error because you're running 4.19. `LED_AUDIO_MICMUTE` has been implemented in 5.0 so you either upgrade to at least 5.0 or run the old version of this driver version v1.0.

I recommend upgrading so you can try out the new features in v3.2

mariusrodi, 2019-05-23:
I ran the old version of the driver following the guide for version 1.0.

I will try it again after updating the kernel.

closed 2019-06-04 by aymanbagabas