aymanbagabas/Huawei-WMI #16: fn / caps lock led stays on

bernimoses, 2019-06-04:
If the matebook display turns off (e.g. screen lock) the fn and caps lock leds turn on. I think this should indicate that it is still running. This is normally not a problem since the leds would turn off after the display turns on again. But if you use a external monitor as your only monitor the leds stay on until the external monitor is disconnected. If you try to toggle the keys the first click activates the toggle (fn / caps lock) and the led stays on (like it should since the key is toggled). The second click would turn the led finally off.

Is this something that can be fixed / should be fixed here?

Thanks.

aymanbagabas, 2019-06-05:
Does this behavior exists in Windows? If it does, then it might be a firmware/bios issue. If it's Linux specific, then further debugging is required.

bernimoses, 2019-06-05:
Same behavior in Windows. :disappointed: 
I suppose this isn't something to implement regardless?

aymanbagabas, 2019-06-05:
The leds control are tied to laptop panel that's why you're getting this. A workaround could be to modify ec memory and turn off those leds once external screen is on again. You can use `ec_sys` module to modify these values.

Let me know if you have any other questions. I'll be closing this for now but feel free to reply to this thread any time.

closed 2019-06-05 by aymanbagabas

dinimar, 2023-06-20:
@aymanbagabas Hi
I was searching topics about this problem. But found only this issue on GitHub.
So, my platform is Windows 11. And I also has the same problem with fn and caps lock leds. As I understand, `es_sys` is Linux-related stuff. Do you know how to implement this workaround on Windows? Or maybe you have some links with instructions?

Any help would be appreciated. Thanks in advance