aymanbagabas/Huawei-WMI #17: Bad return status for dkms module (huawei-wmi-3.2) build on kernel: 5.1.8-1

blackboxlabs, 2019-06-14:
I'm not sure what how exactly I should go about debugging this:

cd huawei-wmi-3.2
make
ls
dkms.conf  huawei-wmi.c  huawei-wmi.ko  huawei-wmi.mod.c  huawei-wmi.mod.o  huawei-wmi.o  Makefile  modules.order  Module.symvers

sudo dkms install -m huawei-wmi -v 3.2

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.1.8-1-MANJARO -C /usr/lib/modules/5.1.8-1-MANJARO/build M=/var/lib/dkms/huawei-wmi/3.2/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.1.8-1-MANJARO (x86_64)
Consult /var/lib/dkms/huawei-wmi/3.2/build/make.log for more information.

 cat /var/lib/dkms/huawei-wmi/3.2/build/make.log

DKMS make.log for huawei-wmi-3.2 for kernel 5.1.8-1-MANJARO (x86_64)
Fri 14 Jun 09:45:25 BST 2019
make: Entering directory '/usr/lib/modules/5.1.8-1-MANJARO/build'
make[1]: *** No rule to make target '/var/lib/dkms/huawei-wmi/3.2/build/huawei-wmi.c', needed by '/var/lib/dkms/huawei-wmi/3.2/build/huawei-wmi.o'.  Stop.
make: *** [Makefile:1571: _module_/var/lib/dkms/huawei-wmi/3.2/build] Error 2
make: Leaving directory '/usr/lib/modules/5.1.8-1-MANJARO/build'

dkms status
acpi_call, 1.1.0, 4.19.49-1-MANJARO, x86_64: installed
acpi_call, 1.1.0, 5.0.21-1-MANJARO, x86_64: installed
acpi_call, 1.1.0, 5.1.8-1-MANJARO, x86_64: installed
huawei-wmi, 3.2: added

aymanbagabas, 2019-06-15:
Hey @blackboxlabs please try [this](https://github.com/aymanbagabas/Huawei-WMI/releases/download/v3.2/PKGBUILD) PKGBUILD file. I've also added further instructions in the readme.

blackboxlabs, 2019-06-15:
----------------------------------------------------------------------------------------------------------------------
Thanks @aymanbagabas I just tried the PKGBUILD file. I got the following error message:
----------------------------------------------------------------------------------------------------------------------

makepkg -si
==> Making package: huawei-wmi-dkms 3.2-1 (Sat 15 Jun 2019 19:55:20 BST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found huawei-wmi-dkms-3.2.zip
==> Validating source files with sha256sums...
    huawei-wmi-dkms-3.2.zip ... Passed
==> Extracting sources...
  -> Extracting huawei-wmi-dkms-3.2.zip with bsdtar
==> Removing existing $pkgdir/ directory...
==> Entering fakeroot environment...
==> Starting package()...
Archive:  huawei-wmi-dkms-3.2.zip
replace huawei-wmi-3.2/Makefile? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: huawei-wmi-3.2/Makefile  
replace huawei-wmi-3.2/huawei-wmi.c? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: huawei-wmi-3.2/huawei-wmi.c  
replace huawei-wmi-3.2/dkms.conf? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: huawei-wmi-3.2/dkms.conf  
install: target '/home/chukwuka/Downloads/pkg/huawei-wmi-dkms/usr/src/huawei-wmi-3.2' is not a directory
==> ERROR: A failure occurred in package().
    Aborting...
----------------------------------------------------------------------------------------------
I then checked the dkms status
----------------------------------------------------------------------------------------------
dkms status
acpi_call, 1.1.0, 4.19.49-1-MANJARO, x86_64: installed
acpi_call, 1.1.0, 5.0.21-1-MANJARO, x86_64: installed
acpi_call, 1.1.0, 5.1.8-1-MANJARO, x86_64: installed
huawei-wmi, 3.2: added

----------------------------------------------------------------------------------------------
So then I thought I would just uninstall the huawei-wmi module:
----------------------------------------------------------------------------------------------

sudo dkms uninstall -m huawei-wmi -v 3.2

Error! The module huawei-wmi 3.2 is not currently installed.
This module is not currently ACTIVE for kernel 5.1.8-1-MANJARO (x86_64).

aymanbagabas, 2019-06-15:
@blackboxlabs could you please try this one
```sh
_pkgbase="huawei-wmi"
pkgname="${_pkgbase}-dkms"
pkgver=3.2
pkgrel=1
pkgdesc="Huawei WMI laptop extras driver"
url="https://github.com/aymanbagabas/Huawei-WMI"
arch=("x86_64")
license=("GPLv2")
depends=("dkms")
makedepends=("linux-headers")
zipname="huawei-wmi-dkms-${pkgver}"
source=("https://github.com/aymanbagabas/Huawei-WMI/releases/download/v${pkgver}/${zipname}.zip")
sha256sums=("80dcb00e978d851336fcd2acae294e1da92eb6342b98d75d6017b826a50647eb")

package() {
	cd "${_pkgbase}-${pkgver}"
	mkdir -p "${pkgdir}/usr/src/${_pkgbase}-${pkgver}"
	install -Dm644 Makefile huawei-wmi.c dkms.conf "${pkgdir}/usr/src/${_pkgbase}-${pkgver}"
}
```

blackboxlabs, 2019-06-16:
Amazing @aymanbagabas it worked! Thank you so much!!

The new PKGBUILD file also suggested some troubleshoot pointers:

I had to remove previously installed files:
"/usr/src/huawei-wmi-3.2/huawei-wmi.c", 
"/usr/src/huawei-wmi-3.2/Makefile" and
 "/usr/src/huawei-wmi-3.2/dkms.conf" 
And then also delete the old linux kernel image 4.19.59-1

closed 2019-06-16 by blackboxlabs