aymanbagabas/Huawei-WMI #19: HUAWEI Matebook E

jannis3005, 2019-08-27:
Hey,

I've tried getting it to work on the Matebook E, but I couldn't get it to work. Since I don't have any Experience in developing in C nor in developing device drivers, it would be cool if you could help me out. 

Here's the output of the commands from your email. 

lsmod|grep wmi
`huawei_wmi             16384  0
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek
wmi_bmof               16384  0
intel_wmi_thunderbolt    20480  0
wmi                    36864  3 intel_wmi_thunderbolt,huawei_wmi,wmi_bmof
sparse_keymap          16384  3 intel_hid,huawei_wmi,intel_vbtn
`

dmesg
`[    0.000000] microcode: microcode updated early to revision 0xb4, date = 2019-04-01
[    0.000000] Linux version 5.2.9-arch1-1-ARCH (builduser@heftig-119803) (gcc version 9.1.0 (GCC)) #1 SMP PREEMPT Fri Aug 16 11:29:43 UTC 2019
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-linux root=UUID=ed16dad4-073a-47d3-aa81-d3b6c1441d82 rw audit=0 loglevel=3 quiet
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000066825fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000066826000-0x0000000066826fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000066827000-0x000000006685afff] usable
[    0.000000] BIOS-e820: [mem 0x000000006685b000-0x000000006685bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000006685c000-0x0000000067b87fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067b88000-0x0000000068c87fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000068c88000-0x000000007ee9dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ee9e000-0x000000007f09dfff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007f09e000-0x000000007f88dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007f88e000-0x000000007ff7dfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007ff7e000-0x000000007fffdfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007fffe000-0x000000007fffefff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffff000-0x000000008cffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fe7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed84000-0x00000000fed84fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000171ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.50 by INSYDE Corp.
[    0.000000] efi:  ACPI 2.0=0x7fffd014  SMBIOS=0x7f114000  SMBIOS 3.0=0x7f112000  ESRT=0x7f110dd8  PROP=0x66822f98 
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: HUAWEI HUAWEI MateBook E/HUAWEI MateBook E, BIOS 2.07 10/31/2018
[    0.000000] tsc: Detected 1600.000 MHz processor
[    0.001565] tsc: Detected 1608.000 MHz TSC
[    0.001565] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001569] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001587] last_pfn = 0x172000 max_arch_pfn = 0x400000000
[... 874 lines cut ...]
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
`
`Select the device event number [0-23]: 17
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "Huawei WMI hotkeys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 171 (KEY_CONFIG)
    Event code 190 (KEY_F20)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
    Event code 238 (KEY_WLAN)
    Event code 240 (KEY_UNKNOWN)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
`

aymanbagabas, 2019-08-29:
This might be out of the driver's scope. Could you provide the ACPI dumps using `acpidump` specifically I need DSDT and SSDT tables.

For more info refer to https://01.org/linux-acpi/utilities

jannis3005, 2019-09-01:
[dump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3563570/dump.txt)

i hope everything is in the attached file

aymanbagabas, 2019-09-01:
Your Matebook E doesn't use the same mechanism as other models. I'd like to investigate on this more if you're still interested.

> `[    3.397014] acpi PNP0C14:02: duplicate WMI GUID 05901221-D566-11D1-B2F0-00A0C9062910 (first instance was on PNP0C14:01)` 

1. This line from `dmesg` is interesting because when I was looking at the DSDT table from your `dump.txt` I couldn't find a device that has the aforementioned GUID. This means that the `dump.txt` I'm afraid is not sufficient and this may be involved with more firmware data.

Could you provide the rest of the firmware data? You could do the following.
```
cd /tmp
sudo cp -r /sys/firmware/acpi/tables firmware
sudo rm -f firmware/MSDM
sudo chown -R 1000:1000 firmware
zip -r firmware.zip firmware
```
We're deleting MSDM because it includes your Windows license key and I don't think you wanna share that :smile: Now, save `/tmp/firmware.zip` somewhere in your system so then you can upload it here.

2. I'd need more info from `dmesg` but with verbose `wmi`. First, stick `options wmi debug_event=1 debug_dump_wdg=1` in `/etc/modprobe.d/wmi.conf`. You could do
```
sudoedit /etc/modprobe.d/wmi.conf
```
Then copy  `options wmi debug_event=1 debug_dump_wdg=1` and paste it in `/etc/modprobe.d/wmi.conf`. Now, do `sudo mkinitcpio -p linux` and reboot. This would make `dmesg` dumps WMI data.

3. Please try out branch `matebook-e`. I've updated that branch to listen to events coming from your device. You could do the following to branch to `matebook-e`, compile the code, and then test it.
```
git clone https://github.com/aymanbagabas/Huawei-WMI.git
cd Huawei-WMI
git checkout matebook-e
make
sudo rmmod huawei_wmi
sudo insmod huawei-wmi.ko
```
Now, try out all the special keys from F1 to F10. Nothing would happen but I'm interested in the logs that the driver would record on `dmesg`. This should also include the `wmi` data from step 2.

After you do these three steps use `dmesg > dmesg.txt` and upload `dmesg.txt` and `firmware.zip` here.

Please let me know if you have any questions or if you're having any troubles. These data would help debug the issue and hopefully, we can fix it and get it to work.

aymanbagabas, 2019-09-01:
Can you also provide `lsusb -v` and `lspci -nnv`? Since the keyboard is detachable and there is a keyboard driver firmware on https://consumer.huawei.com/en/support/laptops/matebook-e/ , this might need an extra driver :crossed_fingers: hopefully not.

jannis3005, 2019-09-02:
[firmware.zip](https://github.com/aymanbagabas/Huawei-WMI/files/3564517/firmware.zip)
[dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3564518/dmesg.txt)

hope I've got everything correct :)

<details>
<summary>lsusb-v</summary>
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.02
  iManufacturer           3 
  iProduct                2 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0

Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x8087 Intel Corp.
  idProduct          0x0a2b 
  bcdDevice            0.10
[... 598 lines cut ...]
00:1f.3 Audio device [0403]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d71] (rev 21) (prog-if 80)
        Subsystem: Huawei Technologies Co., Ltd. Sunrise Point-LP HD Audio [19e5:3e01]
        Flags: bus master, fast devsel, latency 32, IRQ 136
        Memory at a1538000 (64-bit, non-prefetchable) [size=16K]
        Memory at a1520000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: <access denied>
        Kernel driver in use: snd_hda_intel
        Kernel modules: snd_hda_intel, snd_soc_skl

00:1f.4 SMBus [0c05]: Intel Corporation Sunrise Point-LP SMBus [8086:9d23] (rev 21)
        Subsystem: Huawei Technologies Co., Ltd. Sunrise Point-LP SMBus [19e5:3e01]
        Flags: medium devsel, IRQ 16
        Memory at a154c000 (64-bit, non-prefetchable) [size=256]
        I/O ports at 3040 [size=32]
        Kernel driver in use: i801_smbus
        Kernel modules: i2c_i801

01:00.0 Network controller [0280]: Intel Corporation Wireless 8265 / 8275 [8086:24fd] (rev 78)
        Subsystem: Huawei Technologies Co., Ltd. Wireless 8265 / 8275 [19e5:3e01]
        Flags: bus master, fast devsel, latency 0, IRQ 129
        Memory at a1400000 (64-bit, non-prefetchable) [size=8K]
        Capabilities: <access denied>
        Kernel driver in use: iwlwifi
        Kernel modules: iwlwifi
</details>

aymanbagabas, 2019-09-03:
Hi @jannis3005,

Thank you for taking the time and yes, you got everything correct thank you!

The Matebook E is interesting, looking at the provided firmware files, it appears to follow the same mechanism that the Matebook X has. It uses the same "legacy" interface. Check out #9 for more info. If you'd like to help with that, I'd really appreciate it.

Now, your dmesg log shows `Unknown response type 3` at the bottom. I've updated `matebook-e` branch to print out incoming keycodes. Make sure you have both `debug_event=1` and `debug_dump_wdg=1` in your `/etc/modprobe.d/wmi.conf` then do `sudo mkinitcpio -p linux`. If you have both enabled, you should see a DEBUG message in `dmesg` everytime you press a special key. Please do the following:
```
rm -rf Huawei-WMI
git clone https://github.com/aymanbagabas/Huawei-WMI.git
cd Huawei-WMI
git checkout matebook-e
make
sudo rmmod huawei_wmi
sudo insmod huawei-wmi.ko
```

Now, try all of the Fn row, special keys, and observe changes in `dmesg -w`. After pressing all of F1-F12, please provide `dmesg > dmesg.txt` file.

jannis3005, 2019-09-03:
[dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3568023/dmesg.txt)

When I do <code>dmesg -w</code> and press all the keys from F1 to F12 with the FN key set to media functionality, nothing happens. I've attached dmesg.txt anyways. I think I followed all of your instructions correctly. Shouldn't <code>dmesg -w</code> print out all the pressed keys from the WMI?

<details>
<summary>cat /etc/modprobe.d/wmi.conf</summary>
options wmi debug_event=1 debug_dump_wdg=1
</details>

aymanbagabas, 2019-09-03:
Could you try adding `wmi.debug_event=1 wmi.debug_dump_wdg=1` to your cmdline and then reboot. If you're using GRUB, it would go in `/etc/default/grub`. You have update your GRUB menu using `sudo grub-mkconfig -o /boot/grub/grub.cfg`. Also it looks like the driver didn't get loaded since it's an out of kernel driver, dmesg should give a warning when `insmod` the driver. Please try the following:
```
sudo mkdir /lib/modules/$(uname -r)/updates/
rm -rf Huawei-WMI
git clone https://github.com/aymanbagabas/Huawei-WMI.git
cd Huawei-WMI
git checkout matebook-e
make
sudo cp huawei-wmi.ko /lib/modules/$(uname -r)/updates/
sudo depmod
sudo depmod -a
```
Make sure non of the above commands fail or give you an error message.

Now reboot, press all Fn keys while observing `dmesg -w` and upload `dmesg > dmesg.txt`.

jannis3005, 2019-09-04:
This is weird. I followed all of your instructions, there were no errors. After rebooting, when I run lsmod it shows huawei_wmi as a loaded module. But it still doesn't show anything on <code>dmesg -w</code>. I tried it again with the first method, but with the GRUB cmdline parameters set, and it shows the module with lsmod and it does not give an error on insmod. but still, i do not get any output when pressing the keys.

jannis3005, 2019-09-04:
I just tried the following: I removed the kernel module and rebooted. then i run <code>dmesg -w</code> before running <code>insmod huawei_wmi.ko</code>. The moment i run <code>insmod huawei_wmi.ko</code>, 4 lines pop up in dmesg:
<code>
GUID 59142400-C6A3-40fa-BADB-8A265283
</code>

so it seems to be loading correctly

aymanbagabas, 2019-09-08:
@jannis3005 I sort of see the problem now.

Nothing wrong with the driver, the issue is that the keyboard is connected through USB as you can see in [this comment](https://github.com/aymanbagabas/Huawei-WMI/issues/19#issuecomment-527003094) `lsusb -v` `Bus 001 Device 002: ID 12d1:1097 Huawei Technologies Co., Ltd.` That's the keyboard USB device.

Now, I want you to try `sudo evtest` with the device corresponding to this USB device. It might have the name of `12d1:1097 Huawei Technologies Co., Ltd.`, `HOLTEK`, or `USB-HID KB MS` and get the output after pressing the hotkeys. If that didn't work, try to find the device in `sudo evtest` that detects brightness keys since brightness down works for you. The display key is different, it's actually mapped to `Super-p` which cycles through display options and works in Windows, GNOME, KDE, etc.

The issue is that we're missing the USB driver that reports events to Huawei-WMI. You're getting nothing because events from this USB device doesn't go anywhere and that's why you're not getting anything in `dmesg`. Using `options wmi debug_event=1 debug_dump_wdg=1`, `dmesg` should report ANY WMI event received, something like this
```
[10000.569616] wmi: DEBUG Event INTEGER_TYPE - 641
[10000.569625] wmi: DEBUG Event GUID: ABBC0F5C-8EA1-11D1-A000-C90629100000
```
And that's without the driver being loaded notice 641 is 0x281 which is the brightness down key defined [here](https://github.com/aymanbagabas/Huawei-WMI/blob/master/huawei-wmi.c#L72).

Once this "theory" is confirmed, I'll put some code up to fix this. 

Thank you for your support and contribution to this project.

jannis3005, 2019-09-09:
Hey,

i found the correct device, take a look

<details>
<summary>here</summary>
`No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Lid Switch
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      AT Translated Set 2 keyboard
/dev/input/event4:      HOLTEK USB-HID KB MS
/dev/input/event5:      HOLTEK USB-HID KB MS Consumer Control
/dev/input/event6:      HOLTEK USB-HID KB MS
/dev/input/event7:      HOLTEK USB-HID KB MS Touchpad
/dev/input/event8:      Intel Virtual Button driver
/dev/input/event9:      Intel HID events
/dev/input/event10:     Intel HID 5 button array
/dev/input/event11:     PC Speaker
/dev/input/event12:     Video Bus
/dev/input/event13:     Wacom HID 485B Pen
/dev/input/event14:     Wacom HID 485B Finger
/dev/input/event15:     HDA Intel PCH Mic
/dev/input/event16:     HDA Intel PCH Front Headphone
/dev/input/event17:     Huawei WMI hotkeys
/dev/input/event18:     Huawei WMI hotkeys
/dev/input/event19:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event22:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event23:     HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-23]: 4
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x12d1 product 0x1097 version 0x110
Input device name: "HOLTEK USB-HID KB MS"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
[... 201 lines cut ...]
Event: time 1568009472.639658, -------------- SYN_REPORT ------------
Event: time 1568009472.752188, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e1
Event: time 1568009472.752188, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1568009472.752188, -------------- SYN_REPORT ------------
Event: time 1568009472.860447, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70006
Event: time 1568009472.860447, type 1 (EV_KEY), code 46 (KEY_C), value 1
Event: time 1568009472.860447, -------------- SYN_REPORT ------------
`
</details>

also, here's

<details>
<summary>`/sys/class/leds/`</summary>
`insgesamt 0
lrwxrwxrwx 1 root root 0 2019-09-09 08:13 input3::capslock -> ../../devices/platform/i8042/serio0/input/input3/input3::capslock
lrwxrwxrwx 1 root root 0 2019-09-09 08:13 input3::numlock -> ../../devices/platform/i8042/serio0/input/input3/input3::numlock
lrwxrwxrwx 1 root root 0 2019-09-09 08:13 input3::scrolllock -> ../../devices/platform/i8042/serio0/input/input3/input3::scrolllock
lrwxrwxrwx 1 root root 0 2019-09-09 08:13 input4::capslock -> ../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:12D1:1097.0001/input/input4/input4::capslock
lrwxrwxrwx 1 root root 0 2019-09-09 08:13 input4::numlock -> ../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:12D1:1097.0001/input/input4/input4::numlock
lrwxrwxrwx 1 root root 0 2019-09-09 08:13 input4::scrolllock -> ../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:12D1:1097.0001/input/input4/input4::scrolllock
lrwxrwxrwx 1 root root 0 2019-09-09 08:13 phy0-led -> ../../devices/pci0000:00/0000:00:1d.0/0000:01:00.0/leds/phy0-led
lrwxrwxrwx 1 root root 0 2019-09-09 08:13 platform::micmute -> ../../devices/platform/huawei-wmi/leds/platform::micmute
`
</details>

aymanbagabas, 2019-09-09:
That's great news! Please try the attached file using the method described in the readme. Feel free to adjust it as you like. Refer to `/usr/include/linux/input-event-codes.h` and for more info [here](https://github.com/systemd/systemd/blob/master/hwdb/60-keyboard.hwdb).

Although it should hopefully work, I see from your output that F1-11 keys are being clicked. Are you sure Fn toggle is off? That's why you're getting `^@` and `^[OP` etc. Use something like `xev` to detect that F1-11 keys aren't being triggered and report back.

Also I'd like to see what the other inputs have
```
/dev/input/event5:      HOLTEK USB-HID KB MS Consumer Control
/dev/input/event6:      HOLTEK USB-HID KB MS
/dev/input/event7:      HOLTEK USB-HID KB MS Touchpad
```
One of them could lead us to a functional mic mute LED.

[99-Huawei.zip](https://github.com/aymanbagabas/Huawei-WMI/files/3592923/99-Huawei.zip)

jannis3005, 2019-09-10:
It seems to be a bit more complicated. The device listet as device 4 seems to be responsible for the monitor key, as well as print and paste. device 6 seems to be responsible for brightness up and down.

Though, Device 6 seems to report brightness down for both brightness down and up and the fn key.

as for device 5, it shows events for key_mute key_volumeup and down and a whole bunch of other stuff, however, when i press the buttons nothing outputs

jannis3005, 2019-09-10:
<details>
<summary>Here's the Ouput of event6 pressing all keys, with the ouput caused by brightness_down and brightness_up</summary>

```No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Lid Switch
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      AT Translated Set 2 keyboard
/dev/input/event4:      HOLTEK USB-HID KB MS
/dev/input/event5:      HOLTEK USB-HID KB MS Consumer Control
/dev/input/event6:      HOLTEK USB-HID KB MS
/dev/input/event7:      HOLTEK USB-HID KB MS Touchpad
/dev/input/event8:      PC Speaker
/dev/input/event9:      Intel HID events
/dev/input/event10:     Intel HID 5 button array
/dev/input/event11:     Intel Virtual Button driver
/dev/input/event12:     Video Bus
/dev/input/event13:     Wacom HID 485B Pen
/dev/input/event14:     Wacom HID 485B Finger
/dev/input/event15:     HDA Intel PCH Mic
/dev/input/event16:     HDA Intel PCH Front Headphone
/dev/input/event17:     Huawei WMI hotkeys
/dev/input/event18:     Huawei WMI hotkeys
/dev/input/event19:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event22:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event23:     HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-23]: 6
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x12d1 product 0x1097 version 0x111
Input device name: "HOLTEK USB-HID KB MS"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 224 (KEY_BRIGHTNESSDOWN)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)
Event: time 1568099866.221314, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1568099866.221314, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1568099866.221314, -------------- SYN_REPORT ------------
Event: time 1568099866.301040, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1568099866.301040, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 0
Event: time 1568099866.301040, -------------- SYN_REPORT ------------
Event: time 1568099867.957312, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1568099867.957312, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1568099867.957312, -------------- SYN_REPORT ------------
Event: time 1568099868.009117, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1568099868.009117, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 0
Event: time 1568099868.009117, -------------- SYN_REPORT ------------
```
</details>

jannis3005, 2019-09-10:
Maybe some more details that i found out:

event4 seems to receive events for every key on the keyboard, EXCEPT:
FN-Key
Brightness-Down
Brightness-UP
Keyboard-Brightness
Mute
Volume-DOWN
Volume-UP
Mic-MUTE
WIFI
SETUP

However, it does receive events when pressing f1-f12, as well as screenmirrormodetoggle or hoewver that is called, and also for print and paste. The screenmirrormodetoggle event is received as LEFT_META.

The Volume-Keys on the side of the detachable tablet, located next to the fingerprint, do work. The events show up in event3, the AT Translated Set 2 Keyboard

aymanbagabas, 2019-09-13:
> as for device 5, it shows events for key_mute key_volumeup and down and a whole bunch of other stuff, however, when i press the buttons nothing outputs

I'm interested in this. Please upload `sudo evtest` output after pressing all special keys.

jannis3005, 2019-09-16:
Sorry for the late response, here are the outputs:

<details>
<summary><code>sudo evtest</code></summary>

```
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Lid Switch
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      AT Translated Set 2 keyboard
/dev/input/event4:      HOLTEK USB-HID KB MS
/dev/input/event5:      HOLTEK USB-HID KB MS Consumer Control
/dev/input/event6:      HOLTEK USB-HID KB MS
/dev/input/event7:      HOLTEK USB-HID KB MS Touchpad
/dev/input/event8:      Intel Virtual Button driver
/dev/input/event9:      Intel HID events
/dev/input/event10:     Intel HID 5 button array
/dev/input/event11:     PC Speaker
/dev/input/event12:     HDA Intel PCH Mic
/dev/input/event13:     HDA Intel PCH Front Headphone
/dev/input/event14:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event15:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event16:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event17:     Huawei WMI hotkeys
/dev/input/event18:     Huawei WMI hotkeys
/dev/input/event19:     Video Bus
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event22:     Wacom HID 485B Pen
/dev/input/event23:     Wacom HID 485B Finger
Select the device event number [0-23]: 5
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x12d1 product 0x1097 version 0x111
Input device name: "HOLTEK USB-HID KB MS Consumer Control"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 128 (KEY_STOP)
    Event code 140 (KEY_CALC)
    Event code 144 (KEY_FILE)
    Event code 155 (KEY_MAIL)
    Event code 156 (KEY_BOOKMARKS)
    Event code 158 (KEY_BACK)
    Event code 159 (KEY_FORWARD)
    Event code 163 (KEY_NEXTSONG)
    Event code 164 (KEY_PLAYPAUSE)
    Event code 165 (KEY_PREVIOUSSONG)
    Event code 166 (KEY_STOPCD)
    Event code 171 (KEY_CONFIG)
    Event code 172 (KEY_HOMEPAGE)
    Event code 173 (KEY_REFRESH)
    Event code 217 (KEY_SEARCH)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
```

</details>

<details>
<summary><code>usbhid-dump --entitiy=all model=12d1</code></summary>

[... 284 lines cut ...]
ip_tables              36864  0
x_tables               49152  1 ip_tables
ext4                  770048  1
crc32c_generic         16384  0
crc16                  16384  2 bluetooth,ext4
mbcache                16384  1 ext4
hid_generic            16384  0
jbd2                  135168  1 ext4
usbhid                 65536  1 wacom
hid                   143360  7 i2c_hid,wacom,usbhid,hid_multitouch,hid_sensor_hub,intel_ishtp_hid,hid_generic
sd_mod                 57344  4
serio_raw              20480  0
ahci                   40960  3
atkbd                  36864  0
libahci                40960  1 ahci
libps2                 20480  1 atkbd
libata                282624  2 libahci,ahci
scsi_mod              249856  5 vhba,sd_mod,libata,sg,sr_mod
xhci_pci               20480  0
crc32c_intel           24576  2
xhci_hcd              278528  1 xhci_pci
i8042                  32768  0
serio                  28672  4 serio_raw,atkbd,i8042
```
</details>

jannis3005, 2019-09-16:
For usbhid-dump the key order pressed is: fn-key > brighness-down > brightness-up > Keyboard brightness > ... > WIFI > Setup > print > paste

aymanbagabas, 2019-09-16:
> For usbhid-dump the key order pressed is: fn-key > brighness-down > brightness-up > Keyboard brightness > ... > WIFI > Setup > print > paste

Just to be clear, when you say fn-key, do you mean fn key was toggled on? LED is on? Because that would produce F1-F12 keys to my knowledge.

jannis3005, 2019-09-17:
No for sure it wasn't on

jannis3005, 2020-01-21:
@aymanbagabas are you still interested in trying to solve this?

aymanbagabas, 2020-01-23:
You know what, this seems interesting. Now that I've seen [this](https://eleccelerator.com/usbdescreqparser/) tool, this should be very trivial (ofc after doing some more reading about USB HID). If you'd be willing to help in building this I'd be happy to work on it.

<details>
<summary>001:002:003:DESCRIPTOR</summary>

```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x09, 0x02,        //   Usage (0x02)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x40,        //   Report Count (64)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x03,        //   Usage (0x03)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x40,        //   Report Count (64)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0x22,        //   Usage (0x22)
0x95, 0x08,        //   Report Count (8)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 40 bytes
```

</details>

<details>
<summary>001:002:002:DESCRIPTOR</summary>

```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x02,        //     Usage Maximum (0x02)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x01,        //     Report Size (1)
0x95, 0x02,        //     Report Count (2)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x06,        //     Report Count (6)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x09, 0x38,        //     Usage (Wheel)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x75, 0x08,        //     Report Size (8)
0x95, 0x03,        //     Report Count (3)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x0C,        //     Usage Page (Consumer)
0x0A, 0x38, 0x02,  //     Usage (AC Pan)
0x95, 0x01,        //     Report Count (1)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x08,        //     Report Size (8)
0x95, 0x03,        //     Report Count (3)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x06, 0x00, 0xFF,  //   Usage Page (Vendor Defined 0xFF00)
0x85, 0x0D,        //   Report ID (13)
[... 221 lines cut ...]
0x95, 0x03,        //   Report Count (3)
0x75, 0x01,        //   Report Size (1)
0x05, 0x08,        //   Usage Page (LEDs)
0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x03,        //   Usage Maximum (Scroll Lock)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x01,        //   Report Count (1)
0x75, 0x05,        //   Report Size (5)
0x91, 0x03,        //   Output (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xA4, 0x00,  //   Logical Maximum (164)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x29, 0xA4,        //   Usage Maximum (0xA4)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 64 bytes
```

</details>

These were taken from https://github.com/aymanbagabas/Huawei-WMI/issues/19#issuecomment-531661897 USB HID dump. We have 4 descriptors for 4 devices (shown in the evtest). Now we need to know which one is which and which one needs working on.

aymanbagabas, 2020-01-23:
Can you run `usbhid-dump` again, press all special keys, and attach that here? So basically, a `STREAM` is the data coming from the usb device. We need to translate them to key presses (though I'm not yet sure how).

jannis3005, 2020-01-23:
<details>
<summary><code>usbhid-dump</code></summary>
```
001:002:001:STREAM             1579811704.448003
 06 00 03 00 00 00 00 00       Brightness_DOWN P

001:002:001:STREAM             1579811704.536050
 06 00 00 00 00 00 00 00       Brightness_DOWN R

001:002:001:STREAM             1579811705.724051
 06 00 04 00 00 00 00 00       Brightness_UP P

001:002:001:STREAM             1579811705.796046
 06 00 00 00 00 00 00 00       Brightness_UP R

001:002:001:STREAM             1579811706.676019
 06 00 31 00 00 00 00 00       KB_Brightness P

001:002:001:STREAM             1579811706.752103
 06 00 00 00 00 00 00 00       KB_Brightness R

001:002:001:STREAM             1579811707.584109
 06 00 09 00 00 00 00 00       Sound_Mute P

001:002:001:STREAM             1579811707.652101
 06 00 00 00 00 00 00 00       Sound_Mute R

001:002:001:STREAM             1579811708.524099
 06 00 0A 00 00 00 00 00       Sound_DOWN P

001:002:001:STREAM             1579811708.600008
 06 00 00 00 00 00 00 00       Sound_DOWN R

001:002:001:STREAM             1579811709.488061
 06 00 0B 00 00 00 00 00       Sound_UP P

001:002:001:STREAM             1579811709.576060
 06 00 00 00 00 00 00 00       Sound_UP R

001:002:001:STREAM             1579811710.528087
 06 00 0C 00 00 00 00 00       MIC_MUTE P

001:002:001:STREAM             1579811710.600031
 06 00 00 00 00 00 00 00       MIC_MUTE R

001:002:000:STREAM             1579811711.639087
 08 00 13 00 00 00 00 00       Screenmode P

001:002:000:STREAM             1579811711.640079
 08 00 00 00 00 00 00 00       Screenmode R

001:002:000:STREAM             1579811711.641080
 00 00 00 00 00 00 00 00  Prbly also Screenmode?

001:002:001:STREAM             1579811712.648110
 06 00 0D 00 00 00 00 00       WIFI_Toggle P

001:002:001:STREAM             1579811712.736112
 06 00 00 00 00 00 00 00       WIFI_Toggle R

001:002:001:STREAM             1579811713.700111
 06 00 0E 00 00 00 00 00       PCManager P

001:002:001:STREAM             1579811713.792099
 06 00 00 00 00 00 00 00       PC_Manager R

001:002:001:STREAM             1579811714.868125
 06 00 01 00 00 00 00 00       FN-Key P

001:002:001:STREAM             1579811714.920113
 06 00 00 00 00 00 00 00       FN-Key R```
</details>

aymanbagabas, 2020-01-23:
Cool, now we gotta figure out how to translate these data to the input sub-system. Needs some research :) I still don't know if this necessarily *needs* a kernel driver or if it can be done in user-space.

[hid-holtek-kbd](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hid/hid-holtek-kbd.c?h=v5.5-rc7) this, for example, changes the device descriptor because it's faulty and thus needs a kernel driver. In our case, I'm not sure. Needs more investigation. LMK if you find anything (literally anything can be examples forums wikis) and I'll do the same whenever I have time.

jannis3005, 2020-01-23:
Sure. Ill search whenever i've got some spare time :)

Yours sincerely

Jannis Göing

jannis-goeing.com/ <https://www.jannis-goeing.com/>

P.s. BTW I am using Arch ;)

P.p.s. You can also reach me encryptet with PGP - Public key
<https://www.jannis-goeing.de/files/jannis3005@gmail.asc> or Public Key
<https://flowcrypt.com/pub/jannis3005@gmail.com> or through FlowCrypt
<https://flowcrypt.com/me/jannis3005>

aymanbagabas, 2020-01-24:
Alright, I might've found something. [This](https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/) tutorial explains how these descriptors work. First, we have 4 `usbhid` devices defined by the usb hub, each has its own descriptor. If you look at your `lsusb` dump, you can see that it defines 4 interfaces numbered from 0..3. Now looking at the `dmesg` log you provided, we can see that the system identified all 4 devices and mapped them to input devices which they are. These 4 devices define your touchpad, mouse, touchscreen, and keyboard (as you can see from the descriptors).

Now the one responsible for the special keys is provided by `event6` i.e. `001:002:001` Bus001 Device002 Interface001 (you can see this in `lsusb`).

From that device descriptor, we can see that it has a vendor defined usage `// Usage Page (Vendor Defined 0xFF01)`, which means, it probably needs a driver. You can also see the report ID (6) which corresponds with the stream data we're getting. The descriptor also defines one usage `Usage (0x03)` which is the same data coming from the streams. Looking at your evtest log for `event6`, we can see that it only defines the brightness down key with a value of `ff010003`. I'm very certain that this value corresponds to `Usage Page (Vendor Defined 0xFF01)` and `Usage (0x03)` -> `ff010003`.

TLDR, we might need to write a driver similar to [hid-holtek-kbd](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hid/hid-holtek-kbd.c?h=v5.5-rc7) but with a fixed descriptor that has all missing key usages defined.

jannis3005, 2020-01-24:
Okay, that sounds pretty solid and looking at hid-holtek-kbd, also looks doable, even considering that I know almost nothing about kernel drivers

aymanbagabas, 2020-01-24:
https://github.com/torvalds/linux/blob/master/drivers/hid/usbhid/hid-core.c#L67

This might be easier :)

jannis3005, 2020-01-24:
So this means we just give the USB HID Core Driver some Params?

aymanbagabas, 2020-01-24:
Yes, you are right. You'd pass the fixed descriptors as parameters. My
concern is how LEDs would work in this case.

jannis3005, 2020-01-24:
I'd suggest we look into getting the Buttons working at all first because
that's the most important part and would make the most difference. The
FN-LED works already. It Would just be the Microphone-LED we'd have to
worry about afterwards.
I could try and find out though, just have to find out how to find out
first ;) :D

aymanbagabas, 2020-01-24:
<details>
<summary>Fixed descriptor</summary>

```
```

</details>

Try 
```
quirks=0x12d1:0x1097:0x05,0x0C,0x09,0x01,0xA1,0x01,0x85,0x02,0x15,0x00,0x25,0x01,0x95,0x14,0x75,0x01,0x0A,0x83,0x01,0x0A,0x8A,0x01,0x0A,0x92,0x01,0x0A,0x94,0x01,0x09,0xCD,0x09,0xB7,0x09,0xB6,0x09,0xB5,0x09,0xE2,0x09,0xEA,0x09,0xE9,0x0A,0x21,0x02,0x0A,0x23,0x02,0x0A,0x24,0x02,0x0A,0x25,0x02,0x0A,0x26,0x02,0x0A,0x27,0x02,0x0A,0x2A,0x02,0x09,0x70,0x09,0x6F,0x81,0x02,0x95,0x01,0x75,0x0C,0x81,0x01,0xC0,0x06,0x01,0xFF,0x09,0x01,0xA1,0x01,0x85,0x06,0x15,0x00,0x26,0xFF,0x00,0x09,0x01,0x09,0x03,0x09,0x04,0x09,0x09,0x75,0x08,0x95,0x07,0x81,0x00,0xC0
```

or

```
quirks=0x12d1:0x1097:0x050C0901A101850215002501951475010A83010A8A010A92010A940109CD09B709B609B509E209EA09E90A21020A23020A24020A25020A26020A27020A2A020970096F81029501750C8101C00601FF0901A1018506150026FF000901090309040909750895078100C0
```

not sure how this works.

jannis3005, 2020-01-24:
Do you know where I put it?

aymanbagabas, 2020-01-24:
Put it in `/etc/modprobe.d/usbhid.conf`

```
options usbhid quirks=...
```

jannis3005, 2020-01-24:
i tried in cmdline usbhid.quirks=... with the first one, the entire keyboard stopped working, with the second one it was like before

jannis3005, 2020-01-24:
For the first one:

<details>
<summary>dmesg</summary>

```
[    0.000000] Linux version 5.4.12-arch1-1 (linux@archlinux) (gcc version 9.2.0 (GCC)) #1 SMP PREEMPT Tue, 14 Jan 2020 21:44:31 +0000
[    0.000000] Command line: \\vmlinuz-linux root=UUID=8803bc76-19a0-4ca9-9900-7799fd84bae8 rw add_efi_memmap quiet splash loglevel=3 rd.udev.log_priority=3 vt.global_cursor_default=0 rd.systemd.show_status=auto initrd=\initramfs-linux.img
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000066825fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000066826000-0x0000000066826fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000066827000-0x000000006685afff] usable
[    0.000000] BIOS-e820: [mem 0x000000006685b000-0x000000006685bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000006685c000-0x0000000067b87fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067b88000-0x0000000068c87fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000068c88000-0x000000007ee9dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ee9e000-0x000000007f88dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007f88e000-0x000000007ff7dfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007ff7e000-0x000000007fffdfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007fffe000-0x000000007fffefff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffff000-0x000000008cffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fe7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed84000-0x00000000fed84fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000171ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x665db018-0x665eb057] usable ==> usable
[    0.000000] e820: update [mem 0x665db018-0x665eb057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x00000000665db017] usable
[    0.000000] reserve setup_data: [mem 0x00000000665db018-0x00000000665eb057] usable
[    0.000000] reserve setup_data: [mem 0x00000000665eb058-0x0000000066825fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000066826000-0x0000000066826fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000066827000-0x000000006685afff] usable
[    0.000000] reserve setup_data: [mem 0x000000006685b000-0x000000006685bfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000006685c000-0x0000000067b87fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000067b88000-0x0000000068c87fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000068c88000-0x000000007ee9dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000007ee9e000-0x000000007f88dfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007f88e000-0x000000007ff7dfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000007ff7e000-0x000000007fffdfff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000007fffe000-0x000000007fffefff] usable
[... 955 lines cut ...]
<details>
<summary>evtest event6</summary>

```
Event: time 1579886629.723136, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579886629.723136, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1579886629.723136, -------------- SYN_REPORT ------------
Event: time 1579886629.831128, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
[... cut ...]
Event: time 1579886630.963108, -------------- SYN_REPORT ------------
Event: time 1579886638.971289, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579886638.971289, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1579886638.971289, -------------- SYN_REPORT ------------
Event: time 1579886639.031191, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579886639.031191, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 0
Event: time 1579886639.031191, -------------- SYN_REPORT ------------
```

</details>

jannis3005, 2020-01-24:
For the second one:

<details>
<summary>dmsg</summary>

```
[    0.000000] Linux version 5.4.12-arch1-1 (linux@archlinux) (gcc version 9.2.0 (GCC)) #1 SMP PREEMPT Tue, 14 Jan 2020 21:44:31 +0000
[    0.000000] Command line: \\vmlinuz-linux root=UUID=8803bc76-19a0-4ca9-9900-7799fd84bae8 rw add_efi_memmap quiet splash loglevel=3 rd.udev.log_priority=3 vt.global_cursor_default=0 rd.systemd.show_status=auto initrd=\initramfs-linux.img
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000066825fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000066826000-0x0000000066826fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000066827000-0x000000006685afff] usable
[    0.000000] BIOS-e820: [mem 0x000000006685b000-0x000000006685bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000006685c000-0x0000000067b87fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067b88000-0x0000000068c87fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000068c88000-0x000000007ee9dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ee9e000-0x000000007f88dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007f88e000-0x000000007ff7dfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007ff7e000-0x000000007fffdfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007fffe000-0x000000007fffefff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffff000-0x000000008cffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fe7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed84000-0x00000000fed84fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000171ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x665db018-0x665eb057] usable ==> usable
[    0.000000] e820: update [mem 0x665db018-0x665eb057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x00000000665db017] usable
[    0.000000] reserve setup_data: [mem 0x00000000665db018-0x00000000665eb057] usable
[    0.000000] reserve setup_data: [mem 0x00000000665eb058-0x0000000066825fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000066826000-0x0000000066826fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000066827000-0x000000006685afff] usable
[    0.000000] reserve setup_data: [mem 0x000000006685b000-0x000000006685bfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000006685c000-0x0000000067b87fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000067b88000-0x0000000068c87fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000068c88000-0x000000007ee9dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000007ee9e000-0x000000007f88dfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007f88e000-0x000000007ff7dfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000007ff7e000-0x000000007fffdfff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000007fffe000-0x000000007fffefff] usable
[... 969 lines cut ...]
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)
Event: time 1579886884.063138, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579886884.063138, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1579886884.063138, -------------- SYN_REPORT ------------
[... cut ...]
Event: time 1579886884.871150, -------------- SYN_REPORT ------------
Event: time 1579886889.443020, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579886889.443020, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1579886889.443020, -------------- SYN_REPORT ------------
Event: time 1579886889.498923, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579886889.498923, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 0
Event: time 1579886889.498923, -------------- SYN_REPORT ------------
```

jannis3005, 2020-01-24:
I'll try and add that file to initframs

jannis3005, 2020-01-24:
The first one breaks the entire keyboard and touchpad except for the FN-Toggle and Backlight & Backlight Toggle. Ill have to boot from live-usb and remove the quirks to be able to continue.

aymanbagabas, 2020-01-24:
> The first one breaks the entire keyboard and touchpad except for the FN-Toggle and Backlight & Backlight Toggle. Ill have to boot from live-usb and remove the quirks to be able to continue.

No need just override the modprobe during boot with GRUB https://wiki.archlinux.org/index.php/Kernel_module#Using_kernel_command_line set it back to `usbhid.quirks=`

aymanbagabas, 2020-01-24:
Try this 
[hid-huawei-1.zip](https://github.com/aymanbagabas/Huawei-WMI/files/4109969/hid-huawei-1.zip)
```
# extract
make
sudo insmod hid-huawei.ko
```

jannis3005, 2020-01-25:
<details>
<summary>dmesg</sumarry>

```
[    0.000000] Linux version 5.4.12-arch1-1 (linux@archlinux) (gcc version 9.2.0 (GCC)) #1 SMP PREEMPT Tue, 14 Jan 2020 21:44:31 +0000
[    0.000000] Command line: \\vmlinuz-linux root=UUID=8803bc76-19a0-4ca9-9900-7799fd84bae8 rw add_efi_memmap quiet splash loglevel=3 rd.udev.log_priority=3 vt.global_cursor_default=0 rd.systemd.show_status=auto initrd=\initramfs-linux.img
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000066825fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000066826000-0x0000000066826fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000066827000-0x000000006685afff] usable
[    0.000000] BIOS-e820: [mem 0x000000006685b000-0x000000006685bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000006685c000-0x0000000067b87fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067b88000-0x0000000068c87fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000068c88000-0x000000007ee9dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ee9e000-0x000000007f88dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007f88e000-0x000000007ff7dfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007ff7e000-0x000000007fffdfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007fffe000-0x000000007fffefff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffff000-0x000000008cffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fe7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed84000-0x00000000fed84fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000171ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x665db018-0x665eb057] usable ==> usable
[    0.000000] e820: update [mem 0x665db018-0x665eb057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x00000000665db017] usable
[    0.000000] reserve setup_data: [mem 0x00000000665db018-0x00000000665eb057] usable
[    0.000000] reserve setup_data: [mem 0x00000000665eb058-0x0000000066825fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000066826000-0x0000000066826fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000066827000-0x000000006685afff] usable
[    0.000000] reserve setup_data: [mem 0x000000006685b000-0x000000006685bfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000006685c000-0x0000000067b87fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000067b88000-0x0000000068c87fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000068c88000-0x000000007ee9dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000007ee9e000-0x000000007f88dfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007f88e000-0x000000007ff7dfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000007ff7e000-0x000000007fffdfff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000007fffe000-0x000000007fffefff] usable
[    0.000000] reserve setup_data: [mem 0x000000007ffff000-0x000000008cffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[... 970 lines cut ...]
/dev/input/event0:      Lid Switch
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      AT Translated Set 2 keyboard
/dev/input/event4:      HOLTEK USB-HID KB MS
[... cut ...]
/dev/input/event14:     Wacom HID 485B Finger
/dev/input/event15:     HDA Intel PCH Mic
/dev/input/event16:     HDA Intel PCH Front Headphone
/dev/input/event17:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event18:     Huawei WMI hotkeys
/dev/input/event19:     Huawei WMI hotkeys
/dev/input/event20:     Video Bus
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event22:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event23:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event24:     HDA Intel PCH HDMI/DP,pcm=10
```

aymanbagabas, 2020-01-25:
I just wanna see if it changes anything for `event4` and `event5` (`HOLTEK USB-HID KB MS`) as what dmesg states after insmod.

jannis3005, 2020-01-25:
Okay, here you go:

<details>
<summary>'sudo evtest event4'</summary>

```
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Lid Switch
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      AT Translated Set 2 keyboard
/dev/input/event4:      HOLTEK USB-HID KB MS
/dev/input/event5:      HOLTEK USB-HID KB MS
/dev/input/event7:      HOLTEK USB-HID KB MS Mouse
/dev/input/event8:      HOLTEK USB-HID KB MS Touchpad
/dev/input/event9:      Intel HID events
/dev/input/event10:     Intel HID 5 button array
/dev/input/event11:     Intel Virtual Button driver
/dev/input/event12:     HDA Intel PCH Mic
/dev/input/event13:     PC Speaker
/dev/input/event14:     HDA Intel PCH Front Headphone
/dev/input/event15:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event16:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event17:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event18:     Huawei WMI hotkeys
/dev/input/event19:     Huawei WMI hotkeys
/dev/input/event20:     Video Bus
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event22:     HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event23:     Wacom HID 485B Pen
/dev/input/event24:     Wacom HID 485B Finger
Select the device event number [0-24]: 4
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x12d1 product 0x1097 version 0x110
Input device name: "HOLTEK USB-HID KB MS"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
[... 223 lines cut ...]
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)
Event: time 1579956007.042326, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579956007.042326, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1579956007.042326, -------------- SYN_REPORT ------------
Event: time 1579956007.126408, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579956007.126408, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 0
Event: time 1579956007.126408, -------------- SYN_REPORT ------------
Event: time 1579956007.738325, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579956007.738325, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1579956007.738325, -------------- SYN_REPORT ------------
Event: time 1579956007.806235, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff010003
Event: time 1579956007.806235, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 0
Event: time 1579956007.806235, -------------- SYN_REPORT ------------
```

</details>

The Brightness-Down Event in Event5 comes from the FN-Key which still acts as Brightness down, however the Brightness-Keys that both workes as Brightness down before now do nothing