aymanbagabas/Huawei-WMI #20: micmute LED doesn't activate automatically

wrtiap, 2019-08-31:
Everything else works absolutely perfectly for my Matebook 14D AMD. But the micmute LED does not know to turn itself on when mic is muted. If I go and directly change the value of the LED file from 0 to 1, that will turn on the micmute LED though.
Using Ubuntu 19.04

aymanbagabas, 2019-08-31:
What kernel you running? Are you running the driver from this repo or the one that comes with kernel 5.0? This might be related to this [commit](0fbf21c3b36a9921467aa7525d2768b07f9f8fbb) which got merged in stable not sure about Ubuntu 19.04.

wrtiap, 2019-09-01:
Kernel: 5.0.0-25-generic
I installed everything following "Use dkms tarball installation" in the readme.
My systemd version is 240 but my micmute fn-hotkey works so I didn't do the "sudo udevadm --debug hwdb --update; sudo udevadm trigger" step.
Thanks a lot for all this already, and sorry for being an absolute noob

wrtiap, 2019-09-01:
I've just looked at the commit thread you linked, but not sure if it's fully related (don't really understand enough of it). But I know the fn-keys, special keys, charge threshold, micmute LED (when setting manually) all work perfectly for me. But if you think it will be fixed in the next kernel of course I'd be happy to wait and report back!
By the way, if this driver comes with the kernel, does it mean it should work the same without me installing from this repo? Since I'm pretty sure I never had the directory /sys/devices/platform/huawei-wmi before I installed from here.

aymanbagabas, 2019-09-01:
> By the way, if this driver comes with the kernel, does it mean it should work the same without me installing from this repo? Since I'm pretty sure I never had the directory /sys/devices/platform/huawei-wmi before I installed from here.

Yes, but the one comes with the kernel is old that's why. I'm working on submitting the new updates.

If the LED works by setting it manually, it means the kernel didn't link the led with the sound subsystem. Kernel 5.2 fixes that using the mentioned patch which should detect all Huawei laptops instead of only MACH-WX9, which is what I have, and MatebookX the original one from 2017. Try installing kernel 5.2 and let me know what happens. The easiest way to install kernel 5.2 is through Ukuu in Ubuntu though I'd use that for testing only and then remove it.

wrtiap, 2019-09-01:
I've spent a while trying to install Ukuu without success, so I think I will probably leave it and just wait until the next kernel is officially upgraded, since I feel like it's reaching much beyond my current knowledge. Sorry that I wasn't able to help test it out more!
Do I close this issue or leave it open until then? Also if I am feeling brave/impatient and update the kernel manually later on I'll be sure to test it and report back.

aymanbagabas, 2019-09-01:
> I've spent a while trying to install Ukuu without success, so I think I will probably leave it and just wait until the next kernel is officially upgraded, since I feel like it's reaching much beyond [...]

I'm sorry I should've given you instructions on what to do my apologies. Follow [this](https://github.com/teejee2008/ukuu#installation) to install Ukuu. You would do 
```
sudo apt-add-repository -y ppa:teejee2008/ppa
sudo apt-get update
sudo apt-get install ukuu
```
to install add Ukuu repository, refresh apt, and install Ukuu.
* Then once you install it go to the menu and search for Ukuu. Open it up and look for the latest kernel version 5.2 and click install.
* After the installation is finished, reboot your laptop and in GRUB choose the version you just installed. Once you boot into your system, check if this driver got installed correctly (it should if you used DKMS autoinstall). You could check if you have the `/sys/devices/platform/huawei-wmi` directory.
* Now you can test the functionality of the LED. It should turn on/off if you mute or change mic sound settings.
* If it's working correctly, then it means it's just a kernel update. Go ahead and remove the newly installed kernel and reboot (since it can reduce performance because the kernel is not optimized).

If you don't want to do any of that you can close the issue and ignore all of the above lol.

wrtiap, 2019-09-01:
Hi, many many thanks again for your time! And no it's definitely my own responsibility to not require you to guide me step-by-step!
But I had already tried to do so earlier, and reason I couldn't install via "apt-get" was apparently because it's been completely removed from the (repository/ppa?) due to it now being a paid software. I then tried to install manually following the "build instructions" github (https://github.com/teejee2008/ukuu), but there were errors at "make all" which I think lead to some files not being made. I googled around for a bit longer and it *seemed* to be a much more complicated issue.
If you think it's worth it I'm 100% willing to try more things to get that newer kernel, but will probably need one or two more pointers.

aymanbagabas, 2019-09-01:
> Hi, many many thanks again for your time! And no it's definitely my own responsibility to not require you to guide me step-by-step!
> But I had already tried to do so earlier, and reason I couldn't [...]

Hmm, I didn't know that see I use Fedora :smile: which is what I call it semi-cutting-edge when it comes to updates. As of today, Fedora 30 comes with kernel 5.2.9. The current stable version is 5.2.11.

I did some googling and I found this [article](http://ubuntuhandbook.org/index.php/2019/07/install-linux-kernel-5-2-ubuntu-linux-mint/). It turns out Ubuntu provides latest testing mainline kernel builds hosted on https://kernel.ubuntu.com/~kernel-ppa/mainline/ . You could follow that article but use a more recent version maybe [5.2.11](https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.2.11/) the current stable one.

wrtiap, 2019-09-01:
Hi again, I'm now on "5.2.11-050211-generic" and everything is the same, i.e. huawei-wmi directory existed and all the functions seemed to be behaving the same way. So the micmute LED is still not triggering automatically (I can still change it directly by editing the value of /sys/devices/platform/huawei-wmi/leds/platform::micmute/brightness).

aymanbagabas, 2019-09-02:
What's your model codename/product name? Is it `KPL-W0X`? This might be related to #7 . You can check your device product's name using `sudo dmidecode -t1` or `dmesg`. Could you provide a `dmesg` log please.

wrtiap, 2019-09-02:
yes it is KPL-W0X
`sudo dmidecode dmseg
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.10.1 present.
# SMBIOS implementations newer than version 3.2.0 are not
# fully supported by this version of dmidecode.
Table at 0x8C4EA000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: HUAWEI
	Version: 1.19
	Release Date: 01/11/2019
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 8192 kB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		Smart battery is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.19
	Firmware Revision: 1.19

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: HUAWEI
	Product Name: KPL-W0X
	Version: M1F
	Serial Number: EVVPM18513000721
	UUID: 94eb498a-0aa2-47ac-ab13-3d8009f7f5b9
	Wake-up Type: Power Switch
	SKU Number: C128
	Family: MateBook D

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: HUAWEI
	Product Name: KPL-W0X
	Version: M1F
	Serial Number: PM00F01856001667
	Asset Tag: NULL
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: NULL
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 23 bytes
Chassis Information
	Manufacturer: HUAWEI
	Type: Notebook
	Lock: Not Present
	Version: M1F
	Serial Number: EVVPM18513000721
	Asset Tag: NULL
	Boot-up State: Safe
	Power Supply State: Safe
[... 254 lines cut ...]
	Interleaved Data Depth: Unknown

Handle 0x0014, DMI type 21, 7 bytes
Built-in Pointing Device
	Type: Touch Pad
	Interface: Other
	Buttons: 2

Handle 0x0015, DMI type 22, 26 bytes
Portable Battery
	Location: Rear cover inside
	Manufacturer: DYNAPACK
	Manufacture Date: 4/19/2018
	Serial Number: AF3
	Name: HB4593R1ECW
	Chemistry: Lithium Ion
	Design Capacity: 0 mWh
	Design Voltage: 7600 mV
	SBDS Version: Not Specified
	Maximum Error: 1%
	OEM-specific Information: 0x00000002

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table
`

wrtiap, 2019-09-02:
And yes I remember reading through that thread a while back, it seemed quite relevant but I wasn't really able to follow at all (disabling SElinux? etc.).

aymanbagabas, 2019-09-02:
You could try disabling SElinux but I don't see that it would affect this issue. What DE are you using? GNOME, KDE, etc. Does muting the mic from has any effect? Could you provide `dmesg > dmesg.txt`, `cat /sys/bus/hdaudio/devices/hdaudio*/*`, and `cat /sys/devices/platform/huawei-wmi/leds/platform::micmute/trigger`. The first would give you a file named `dmesg.txt`.

Since the led works manually, then the issue wouldn't be related to the driver. I'm trying to think of anything that would affect this other than the driver and the sound driver issue (that should be included in 5.2) is the only thing comes to my mind.

wrtiap, 2019-09-02:
Hi I'm just using everything default from Ubuntu at the moment thankfully (so I think it's just Gnome?). Muting the mic via the function keys I see that it mutes the mic from settings->sound, and it also shows the 'mic muted/mic activaed' icon in the center of the screen (like when u change volume/screen brightness) when you use the hotkey.

[dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3565620/dmesg.txt)

hdaudio*/* (I had 2 folders, hdaudioC0D0 and hdaudioC1D0)
`0x1
R6xx HDMI
cat: /sys/bus/hdaudio/devices/hdaudioC0D0/driver: Is a directory
0x0
hdaudio:v1002AA01r00100700a01
cat: /sys/bus/hdaudio/devices/hdaudioC0D0/power: Is a directory
0x100700
cat: /sys/bus/hdaudio/devices/hdaudioC0D0/subsystem: Is a directory
0xaa0100
0x1
DRIVER=snd_hda_codec_hdmi
MODALIAS=hdaudio:v1002AA01r00100700a01

0x1002aa01
ATI
cat: /sys/bus/hdaudio/devices/hdaudioC0D0/widgets: Is a directory
0x1
ALC256
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/driver: Is a directory
0x0
hdaudio:v10EC0256r00100002a01
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/power: Is a directory
0x100002
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/subsystem: Is a directory
0x19e53207
0x1
DRIVER=snd_hda_codec_realtek
MODALIAS=hdaudio:v10EC0256r00100002a01

0x10ec0256
Realtek
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/widgets: Is a directory
`

micmute/trigger:
`none rfkill-any rfkill-none kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock ACAD-online BAT1-charging-or-full BAT1-charging BAT1-full BAT1-charging-blink-full-solid usb-gadget usb-host disk-activity disk-read disk-write ide-disk mtd nand-disk cpu cpu0 cpu1 cpu2 cpu3 cpu4 cpu5 cpu6 cpu7 cpu8 cpu9 cpu10 cpu11 cpu12 cpu13 cpu14 cpu15 panic audio-mute [audio-micmute] rfkill0 phy0rx phy0tx phy0assoc phy0radio 
`

aymanbagabas, 2019-09-03:
Hi @wrtiap,

I see from your dmesg log that you're using kernel 5.0.0-25-generic. Once you install 5.2.11 and update GRUB, you should see a new option in GRUB to boot with the newer one. If you have GRUB hidden, you could get it to show menu by holding ESC key while booting.

Let me know if it still doesn't work because I'm working on submitting a newer version to the kernel and a broken driver is not good. Thank you and I appreciate you doing this.

wrtiap, 2019-09-03:
Oh yes sorry I didn't mention. After I tested the newer kernel I immediately removed it back again and used the older one again. I'll re-upload all the logs again in 5.2.11 below

wrtiap, 2019-09-03:
sudo dmidecode dmseg:

`# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.10.1 present.
# SMBIOS implementations newer than version 3.2.0 are not
# fully supported by this version of dmidecode.
Table at 0x8C4EA000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: HUAWEI
	Version: 1.19
	Release Date: 01/11/2019
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 8192 kB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		Smart battery is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.19
	Firmware Revision: 1.19

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: HUAWEI
	Product Name: KPL-W0X
	Version: M1F
	Serial Number: EVVPM18513000721
	UUID: 94eb498a-0aa2-47ac-ab13-3d8009f7f5b9
	Wake-up Type: Power Switch
	SKU Number: C128
	Family: MateBook D

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: HUAWEI
	Product Name: KPL-W0X
	Version: M1F
	Serial Number: PM00F01856001667
	Asset Tag: NULL
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: NULL
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 23 bytes
Chassis Information
	Manufacturer: HUAWEI
	Type: Notebook
	Lock: Not Present
	Version: M1F
	Serial Number: EVVPM18513000721
	Asset Tag: NULL
	Boot-up State: Safe
	Power Supply State: Safe
[... 293 lines cut ...]
DRIVER=snd_hda_codec_hdmi
MODALIAS=hdaudio:v1002AA01r00100700a01

0x1002aa01
ATI
cat: /sys/bus/hdaudio/devices/hdaudioC0D0/widgets: Is a directory
0x1
ALC256
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/driver: Is a directory
0x0
hdaudio:v10EC0256r00100002a01
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/power: Is a directory
0x100002
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/subsystem: Is a directory
0x19e53207
0x1
DRIVER=snd_hda_codec_realtek
MODALIAS=hdaudio:v10EC0256r00100002a01

0x10ec0256
Realtek
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/widgets: Is a directory`

cat /sys/devices/platform/huawei-wmi/leds/platform::micmute/trigger:
`none rfkill-any rfkill-none kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock ACAD-online BAT1-charging-or-full BAT1-charging BAT1-full BAT1-charging-blink-full-solid usb-gadget usb-host disk-activity disk-read disk-write ide-disk mtd nand-disk cpu cpu0 cpu1 cpu2 cpu3 cpu4 cpu5 cpu6 cpu7 cpu8 cpu9 cpu10 cpu11 cpu12 cpu13 cpu14 cpu15 panic audio-mute [audio-micmute] rfkill0 phy0rx phy0tx phy0assoc phy0radio`

wrtiap, 2019-09-03:
Once again I am very happy to be able to help in any way, and I should be the one thanking you!
Also not sure at all if relevant, but since you were mentioning audio drivers, I think (can't for the life of me figure out for sure) that my front speakers are not outputting any sound (my laptop has speakers on the left/right next to the keyboard, and also left/right on the underside of laptop). Though this may also be the case for windows (also can't be 100% certain).
Wow didn't realize this was you [https://aymanbagabas.com/2018/07/23/archlinux-on-matebook-x-pro.html](url), should I follow all the things in that guide that I care about fixing? Though I didn't want to change anything on my system yet in case it affects your diagnosis of my micmute LED problem (also slightly different matebook version).

aymanbagabas, 2019-09-03:
I'd be happy to help with the speakers issue but once we get this out of way. Try following the guide since it wouldn't affect what we're doing here.

I see that the driver's LED has the correct trigger `audio-micmute` which connects to `snd-hda-codec-realtek` driver in kernel as you can see in the logs you provided `DRIVER=snd_hda_codec_realtek`. Now this could be a selinux or apparmor thing so I'd disable them all and install the driver so that the system inserts it once it boots up. First, make sure you're using 5.2.11 and then do the following.

Disable selinux and apparmor:
```
sudo systemctl disable apparmor
```
Change `SELINUX=enforcing` in `/etc/selinux/config` to `SELINUX=permissive`.

Install the updated driver:
```
git clone https://github.com/aymanbagabas/Huawei-WMI
cd Huawei-WMI
make
sudo mkdir /lib/modules/$(uname -r)/updates/
sudo cp huawei-wmi.ko /lib/modules/$(uname -r)/updates/
sudo depmod
sudo depmod -a
```

Now, reboot, choose 5.2.11 and see if the LED works as expected. Report back with a `dmesg > dmesg.txt`.

wrtiap, 2019-09-03:
The only file in /etc/selinux/ for me is semanage.conf, which doesn't seem to be the right thing.
I will try without disabling selinux for now.

wrtiap, 2019-09-03:
Nothing seems to have changed so far (didn't/couldn't do the selinux disabling step)

[dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3570972/dmesg.txt)

aymanbagabas, 2019-09-03:
> Nothing seems to have changed so far (didn't/couldn't do the selinux disabling step)

Try `sudo getenforce`

wrtiap, 2019-09-04:
The command is not found. Did I not even install Linux on my computer properly lol?
I'm assuming since selinux isn't installed, it counts as being "disabled"? Let me know if I should find a way to install it properly so I can actually disable it and test again!

aymanbagabas, 2019-09-06:
Nah I think recent Ubuntu versions don't come with SELinux, they use AppArmor instead. TBH, at this point, I can't think of anything that would cause such behavior but I can assure you it's not the driver. It has to do with the sound drivers because activating the LED manually works as you said but not using ALSA. Maybe @wasakakero can help?

aymanbagabas, 2019-09-07:
Let's try this. So basically, we're going to patch the sound driver for realtek `snd-hda-codec-realtek`. Then we're going to compile that from kernel tree based on your current running kernel which should be `5.2.11`. 

```
git clone git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git && cd linux # clone kernel
git checkout v5.2.11 # checkout v5.2.11 branch
git am 0001-Add-Huawei-KPL-W0X-mic-mute-quirk.txt # apply patch
cp /lib/modules/$(uname -r)/build/.config . # copy running kernel config
cp /lib/modules/$(uname -r)/build/Module.symvers . # copy running kernel config
make oldconfig && make prepare # prepare kernel, enter default values
make -C /lib/modules/$(uname -r)/build M=$PWD/sound/pci/hda modules # make patched modules
sudo mkdir /lib/modules/$(uname -r)/updates # make dir if doesn't exist
sudo cp sound/pci/hda/snd-hda-codec-realtek.ko /lib/modules/$(uname -r)/updates/ # copy compiled module to current running kernel moduels
```
Now, reboot and test the LED using a GUI/ALSA/Pulseaudio so muting the mic from a GUI should turn on the LED. Make sure you're running the latest driver from `master` branch.

Also could you provide `alsa-info.sh` from `alsa-utils` it would be helpful to give more info about the codecs your laptop comes with.

[0001-Add-Huawei-KPL-W0X-mic-mute-quirk.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3586104/0001-Add-Huawei-KPL-W0X-mic-mute-quirk.txt)

wrtiap, 2019-09-07:
Sorry if it's a stupid question, but I got stuck on line 3:
~/linux$ git am 0001-Add-Huawei-KPL-W0X-mic-mute-quirk.txt # apply patch
fatal: could not open '0001-Add-Huawei-KPL-W0X-mic-mute-quirk.txt' for reading: No such file or directory

aymanbagabas, 2019-09-07:
> Sorry if it's a stupid question, but I got stuck on line 3:
> ~/linux$ git am 0001-Add-Huawei-KPL-W0X-mic-mute-quirk.txt # apply patch
> fatal: could not open '0001-Add-Huawei-KPL-W0X-mic-mute-quirk.txt' for reading: No such file or directory

Download the attached file and have it in the same directory.

wrtiap, 2019-09-07:
oh my god sorry! That was really stupid.
And sorry I'm stuck again line 6:
~/linux$ make oldconfig && make prepare # prepare kernel, enter default values
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
/bin/sh: 1: flex: not found
make[1]: *** [scripts/Makefile.lib:184: scripts/kconfig/lexer.lex.c] Error 127
make: *** [Makefile:559: oldconfig] Error 2

aymanbagabas, 2019-09-07:
> oh my god sorry! That was really stupid.

Don't be

> And sorry I'm stuck again line 6:
> ~/linux$ make oldconfig && make prepare # prepare kernel, enter default values
> HOSTCC scripts/basic/fixdep
> HOSTCC scripts/kconfig/conf.o
> HOSTCC [...]

Just do `sudo apt-get install flex` and retry.

wrtiap, 2019-09-07:
Thanks again. Just to make a note, I had to install bison as well after that. But now I got another error for the same line. Something about openssl, but this time openssl is/was already installed:
~/linux$ make oldconfig && make prepare # prepare kernel, enter default values
scripts/kconfig/conf  --oldconfig Kconfig
#
# No change to .config
#
  HOSTCC  scripts/sign-file
scripts/sign-file.c:25:10: fatal error: openssl/opensslv.h: No such file or directory
 #include <openssl/opensslv.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.host:90: scripts/sign-file] Error 1
make: *** [Makefile:1087: scripts] Error 2

aymanbagabas, 2019-09-07:
> Thanks again. Just to make a note, I had to install bison as well after that. But now I got another error for the same line. Something about openssl, but this time openssl is/was already installed: [...]

`sudo apt-get install libssl-dev`

wrtiap, 2019-09-07:
[alsa-info.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3586881/alsa-info.txt)
I've done everything above and still no luck with the LED. By the way I am not exactly sure what is meant by "test the LED using a GUI/ALSA/Pulseaudio so muting the mic from a GUI...", I just tried muting the microphone via functionkey and also from settings->sound.

aymanbagabas, 2019-09-07:
> [alsa-info.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3586881/alsa-info.txt)

Thank you! If you look at line 665 and 984 you will see that LED is enabled/seen by ALSA. If #7 got it working, then comparing your `alsa-info.sh` and the one from #7 (http://alsa-project.org/db/?f=0f1f4274a6b8d9a594c7b4a388cc87712713aba9) I can see that BIOS version is different, 1.22 there, 1.19 here. Product version is different, and that log was taken before the fix so you wouldn't find `Mic Mute-LED Mode` there. Maybe updating BIOS would make a difference? Do you have any ideas?

> I've done everything above and still no luck with the LED. By the way I am not exactly sure what is meant by "test the LED using a GUI/ALSA/Pulseaudio so muting the mic from a GUI...", I just tried muting the microphone via functionkey and also from settings->sound.

Yes, that's what I meant. The function key doesn't turn on LED, ALSA does, so when you press that key, it mutes the mic and then ALSA should call the function from the driver to turn the LED on/off. Same function when you do it manually from `/sys/devices/platform/huawei-wmi`.

wrtiap, 2019-09-07:
[alsa-info.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3586936/alsa-info.txt)
I've updated the BIOS to 1.22 now, and tried again but no luck still (via GUI or function key)

wrtiap, 2019-09-07:
Assuming the driver works by writing "1" or "0" into huawei-wmi/leds/platform::micmute/brightness when triggered, is there a way to track how it does so or see if it made an attempt to do so?
Just so you know, I just tried "chmod a+rwX brightness" to see if that might help (didn't), in case that'll negatively affect something.

aymanbagabas, 2019-09-07:
> Assuming the driver works by writing "1" or "0" into huawei-wmi/leds/platform::micmute/brightness when triggered, is there a way to track how it does so or see if it made an attempt to do so?
> Just [...]

Look at [L250](https://github.com/aymanbagabas/Huawei-WMI/blob/master/huawei-wmi.c#L250) and  [L298](https://github.com/aymanbagabas/Huawei-WMI/blob/master/huawei-wmi.c#L298). First one is the function that gets called when setting `platform::micmute/brightness` ignore the if statement we want what's in the else statement. The second one is where we hook up the LED to `audio-micmute` trigger, this is called once to setup/hook the LED to ALSA. You can see that it refers to the function from L250 `priv->cdev.brightness_set_blocking = &huawei_wmi_micmute_led_set;`.

If you wanna know or track if you made an attempt to change the LED you can apply this patch, recompile, `rmmod`, and `insmod`. You can apply using `patch -p1 < patch.txt`. Check the output of `dmesg -w` maybe upload `dmesg > dmesg.txt`.
[patch.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3586960/patch.txt)

wrtiap, 2019-09-08:
Ok I found the issue (it was on my side sorry)!
I got it to work in 5.2.11 only: a little while ago I found out how to add an "echo cancelled" version of mic input by editing some system file, which then adds a 2nd input device which I was using. Going back to "internal microphone-HD-Audio Generic" as default input device allows the trigger to work I suppose. Works even with apparmor enabled again. (Does this mean the trigger only works for specificly the default mic input? Or maybe the echo-cancellation one is not a very official? Sorry I wasn't able to test with another input device at this time.)
This doesn't help though for kernel 5.0.0.
Thank you so very much for all your time and patience! I'm happy to close this as I know how to get it to work once the new kernel is installed. If you want me to test anything more please let me know as I'll be super happy to do so.

aymanbagabas, 2019-09-08:
All good. I don't know what is "echo cancelled" but the trigger is hooked to the Realtek sound driver so it should also work if you were using an external microphone connected through the microphone jack. If this "echo cancelled" mutes the actual mic, I don't see why it wouldn't work.

Anyway, thank you for your great work! I'll be closing this one.

closed 2019-09-08 by aymanbagabas