aymanbagabas/Huawei-WMI #23: Support for fingerprint reader

alexlance, 2019-10-27:
Feel free to close this issue immediately if it's not feasible or within the scope of this project etc...

But wow it would be nice if the fingerprint reader worked under linux.

I read somewhere that Huawei were going to release their own Linux on Matebooks*. I wonder if that means they will release the drivers for the fingerprint reader too. 

*https://www.techradar.com/news/huawei-matebook-laptops-now-come-with-linux

dimon222, 2019-10-27:
Sadly, they didn't include fingerprint driver in it (yeah its stupid).
Confirmation from some reddit user:
https://www.reddit.com/r/MatebookXPro/comments/d2zqdr/does_this_mean_a_possibilty_of_getting_the/ezxz6ci

alexlance, 2019-10-27:
Ah ok. That makes no sense.
Thanks for checking :+1:

aymanbagabas, 2019-10-27:
I'd love to see that working on my laptop. I heard they're gonna be using Deepin OS for their Linux distribution, I don't think they'll do their "own" distribution.

This issue is beyond the scope of this project since it lives in user-space (not kernel related). Let's move the discussion to [this](https://gitlab.freedesktop.org/libfprint/libfprint/issues/112). Note that with kernel 5.3, the system recognizes the fingerprint ACPI device, but the actual driver is missing and that's where [goodix_fp_dump](https://gitlab.collabora.com/ao2/goodix_fp_dump) and `libfprint` come in place.

closed 2019-10-28 by aymanbagabas