aymanbagabas/Huawei-WMI #24: No batter protection path

goldcoders, 2019-12-13:
i already did try to install it 

```
sudo dkms ldtarball --archive=huawei-wmi-3.2-source-only.dkms.tar.gz
sudo dkms autoinstall -m huawei-wmi/3.2
```
installation is fine but i dont have the path specified  on the readme

```
Battery protection can be accessed from either /sys/class/power_supply/BAT0/charge_control_{start,end}_threshold or /sys/devices/platform/huawei-wmi/charge_control_thresholds
```

both of this path i dont have it...

any thing that can help me please thanks
i see someone also facing issue like this

i paste some commands output of my system

```
╰ neofetch 
             /////////////                uriah@dev 
         /////////////////////            --------- 
      ///////*767////////////////         OS: Pop!_OS 19.10 x86_64 
    //////7676767676*//////////////       Host: WRT-WX9 M1090 
   /////76767//7676767//////////////      Kernel: 5.3.0-22-generic 
  /////767676///*76767///////////////     Uptime: 19 mins 
 ///////767676///76767.///7676*///////    Packages: 2389 (dpkg) 
/////////767676//76767///767676////////   Shell: zsh 5.7.1 
//////////76767676767////76767/////////   Resolution: 2160x1440 
///////////76767676//////7676//////////   DE: GNOME 3.34.1 
////////////,7676,///////767///////////   WM: GNOME Shell 
/////////////*7676///////76////////////   WM Theme: Pop 
///////////////7676////////////////////   Theme: Pop-dark [GTK2/3] 
 ///////////////7676///767////////////    Icons: Pop [GTK2/3] 
  //////////////////////'////////////     Terminal: st 
   //////.7676767676767676767,//////      Terminal Font: Hack 
    /////767676767676767676767/////       CPU: Intel i7-8565U (8) @ 4.600GHz 
      ///////////////////////////         GPU: Intel UHD Graphics 620 
         /////////////////////            Memory: 2324MiB / 7706MiB 
             ///////////// 
   uriah@dev  ~                                      ✔   03:25  14/12/19 

```

```
╰ lsmod | grep wmi 
snd_rawmidi            36864  1 snd_seq_midi 
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi 
snd                    86016  25 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi 
huawei_wmi             16384  0 
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek 
sparse_keymap          16384  1 huawei_wmi 
wmi_bmof               16384  0 
intel_wmi_thunderbolt    20480  0 
wmi                    32768  3 intel_wmi_thunderbolt,huawei_wmi,wmi_bmof 
   uriah@dev  ~                                      ✔   03:25  14/12/19 

```

```
╰ uname -r 
5.3.0-22-generic 
   uriah@dev  ~                                      ✔   03:25  14/12/19 

```

goldcoders, 2019-12-13:
i forgot to reboot , after i reboot the path is there , you should put that on documentation im sure others will ask about this again thanks

aymanbagabas, 2019-12-13:
Glad it worked. I should update the readme sometime soon. Thank you.

closed 2019-12-13 by aymanbagabas