aymanbagabas/Huawei-WMI #26: I cannot run Matebook applet

edoardomazzella, 2019-12-22:
Hi,
i downloaded and installed Huawei-WMI and all the stuff seems to work. However i do not see the matebook applet available.
In the README.md of matebook-applet repo the following sentences are written:
"Starting with version 1.2 the best way to get the matebook-applet working is to install Huawei-WMI driver by Ayman Bagabas."
"After installing the driver you may start using the matebook-applet right away."
Is there something else to do in order to install the matebook-applet?
Thank you.

nekr0z, 2019-12-22:
> Is there something else to do in order to install the matebook-applet?

Yes, you need to download/install `matebook-applet` itself. :)

That sentence you've quoted basically means there's no need for additional setup: you just install the driver, install matebook-applet, and there you go, no need to tweak up the system and get your hands dirty. I might need to rephrase that to make it more clear.

aymanbagabas, 2019-12-26:
Can you please try v3.4 and provide a dmesg log?

edoardomazzella, 2019-12-28:
Hi, currently I uninstalled linux on my pc, i'll try as soon as possible.

closed 2020-01-23 by aymanbagabas