aymanbagabas/Huawei-WMI #27: Micmute LED Doesn't Work on Fedora 31

validatedev, 2019-12-22:
**Describe the bug**
It was working on Ubuntu 19.10 with Huawei-WMI 3.2, but cannot reproduce the feature anymore on Huawei Matebook 13

**To Reproduce**
Steps to reproduce the behavior:
Install the Huawei-WMI 3.2 (dnf method or dkms)
Micmute LED doesn't work

**Expected behavior**
Micmute LED should work

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Kernel [e.g. 5.2.9-200.fc30.x86_64] `5.3.16-300.fc31.x86_64 `
 - Distro [e.g. Fedora 30] `Fedora 31
 - Loaded WMI modules 
`huawei_wmi             16384  0
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek
sparse_keymap          16384  1 huawei_wmi
intel_wmi_thunderbolt    20480  0
wmi_bmof               16384  0
wmi                    36864  3 intel_wmi_thunderbolt,huawei_wmi,wmi_bmof
`
 - Dmesg log [dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3992738/dmesg.txt)
 - ALSA log if applicable `http://alsa-project.org/db/?f=0b611158bf130292dce34d54cf76e65d22f0f3af`
 - `evtest` if it's keyboard related. Run `sudo evtest` then choose "Huawei WMI hotkeys", test all hotkeys and submit output.
Just tried the micmute button:
`No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Video Bus
/dev/input/event5:	HD Camera: HD Camera
/dev/input/event6:	ELAN962C:00 04F3:30D0 Touchpad
/dev/input/event7:	Huawei WMI hotkeys
/dev/input/event8:	HDA Intel PCH Headphone Mic
/dev/input/event9:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event10:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event11:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event12:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event13:	HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-13]: 7
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "Huawei WMI hotkeys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 171 (KEY_CONFIG)
    Event code 190 (KEY_F20)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
    Event code 238 (KEY_WLAN)
    Event code 240 (KEY_UNKNOWN)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1577035064.538708, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1577035064.538708, type 1 (EV_KEY), code 190 (KEY_F20), value 1
Event: time 1577035064.538708, -------------- SYN_REPORT ------------
Event: time 1577035064.538737, type 1 (EV_KEY), code 190 (KEY_F20), value 0
Event: time 1577035064.538737, -------------- SYN_REPORT ------------
Event: time 1577035065.334673, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1577035065.334673, type 1 (EV_KEY), code 190 (KEY_F20), value 1
Event: time 1577035065.334673, -------------- SYN_REPORT ------------
Event: time 1577035065.334707, type 1 (EV_KEY), code 190 (KEY_F20), value 0
Event: time 1577035065.334707, -------------- SYN_REPORT ------------`

 - `acpi_listen` if it's keyboard related. Run `sudo acpi_listen` and test all hotkeys. Submit output.
`acpi_listen: can't open socket /var/run/acpid.socket: No such file or directory`

aymanbagabas, 2019-12-22:
Please try `v3.3` and reboot. Thank you!

aymanbagabas, 2019-12-23:
I found the issue. I'm having the same thing. It's DKMS not updating the new module with the one from the kernel. This results in loading the *old* module from the kernel and you can verify that if you see your distro key running `modinfo huawei-wmi` after installing this module.

You could build it manually and move it to `/lib/modules/$(uname -r)/updates/`.

validatedev, 2019-12-23:
Build from source works! Thanks for your attention.

closed 2019-12-23 by validatedev