aymanbagabas/Huawei-WMI #28: Micmute button doesn't work at Ubuntu 18.04 on Huawei Matebook 13

validatedev, 2019-12-23:
**Describe the bug**
Cannot able to get work Micmute button on Ubuntu 18.04 on Huawei Matebook 13. Tried the latest WMI 3.3 with every installation method, but no luck.

**To Reproduce**
Press Micmute button

**Expected behavior**
Should work

**Desktop (please complete the following information):**
 - Kernel [e.g. 5.2.9-200.fc30.x86_64] `5.0.0-37-generic #40~18.04.1-Ubuntu`
 - Distro [e.g. Ubuntu 18.04] `
 - Loaded WMI modules `snd_rawmidi            36864  1 snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
huawei_wmi             20480  1
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek
sparse_keymap          16384  1 huawei_wmi
wmi_bmof               16384  0
intel_wmi_thunderbolt    20480  0
snd                    86016  25 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi
wmi                    28672  3 intel_wmi_thunderbolt,huawei_wmi,wmi_bmof
`
 - Dmesg log [dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3996497/dmesg.txt)

 - `evtest` if it's keyboard related. Run `sudo evtest` then choose "Huawei WMI hotkeys", test all hotkeys and submit output.
`No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Huawei WMI hotkeys
/dev/input/event5:	Video Bus
/dev/input/event6:	ELAN962C:00 04F3:30D0 Touchpad
/dev/input/event7:	HD Camera: HD Camera
/dev/input/event8:	HDA Intel PCH Headphone Mic
/dev/input/event9:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event10:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event11:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event12:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event13:	HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-13]: 4
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "Huawei WMI hotkeys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 171 (KEY_CONFIG)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
    Event code 238 (KEY_WLAN)
    Event code 240 (KEY_UNKNOWN)
    Event code 248 (KEY_MICMUTE)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1577140585.157223, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1577140585.157223, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
Event: time 1577140585.157223, -------------- SYN_REPORT ------------
Event: time 1577140585.157275, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
Event: time 1577140585.157275, -------------- SYN_REPORT ------------
Event: time 1577140586.235434, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1577140586.235434, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
Event: time 1577140586.235434, -------------- SYN_REPORT ------------
Event: time 1577140586.235480, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
Event: time 1577140586.235480, -------------- SYN_REPORT ------------
Event: time 1577140586.959544, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1577140586.959544, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
Event: time 1577140586.959544, -------------- SYN_REPORT ------------
Event: time 1577140586.959593, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
Event: time 1577140586.959593, -------------- SYN_REPORT ------------
`
 - `acpi_listen` if it's keyboard related. Run `sudo acpi_listen` and test all hotkeys. Submit output.
` ABBC0F5C-8EA1-11� 000000a0 00000000
button/micmute MICMUTE 00000080 00000000 K
 ABBC0F5C-8EA1-11� 000000a0 00000000
button/micmute MICMUTE 00000080 00000000 K
 ABBC0F5C-8EA1-11� 000000a0 00000000
button/micmute MICMUTE 00000080 00000000 K
 ABBC0F5C-8EA1-11� 000000a0 00000000
button/micmute MICMUTE 00000080 00000000 K
`

aymanbagabas, 2019-12-24:
Can you try v3.2 on that kernel?

validatedev, 2019-12-24:
> Can you please try https://github.com/aymanbagabas/Huawei-WMI/tree/fix-v3.3

Didn't fix.
evtest:
`Event: time 1577168617.347599, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1577168617.347599, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
Event: time 1577168617.347599, -------------- SYN_REPORT ------------
Event: time 1577168617.347649, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
Event: time 1577168617.347649, -------------- SYN_REPORT ------------
Event: time 1577168617.868907, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1577168617.868907, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
Event: time 1577168617.868907, -------------- SYN_REPORT ------------
Event: time 1577168617.868951, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
Event: time 1577168617.868951, -------------- SYN_REPORT ------------
`
acpi_listen
`validate@validatepc:~$ sudo acpi_listen
 ABBC0F5C-8EA1-11� 000000a0 00000000
button/micmute MICMUTE 00000080 00000000 K
 ABBC0F5C-8EA1-11� 000000a0 00000000
button/micmute MICMUTE 00000080 00000000 K
 ABBC0F5C-8EA1-11� 000000a0 00000000
button/micmute MICMUTE 00000080 00000000 K
 ABBC0F5C-8EA1-11� 000000a0 00000000
button/micmute MICMUTE 00000080 00000000 K
 ABBC0F5C-8EA1-11� 000000a0 00000000
button/micmute MICMUTE 00000080 00000000 K
`
dmesg:
[dmesg2.txt](https://github.com/aymanbagabas/Huawei-WMI/files/3997394/dmesg2.txt)

aymanbagabas, 2019-12-24:
> Didn't fix.
> evtest:
> `Event: time 1577168617.347599, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287 Event: time 1577168617.347599, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1 Event: time [...]

What's your `modinfo huawei-wmi`? Did you place your compiled `fix-v3.3` `.ko` file under `/lib/modules/$(uname -r)/updates/`? Reboot?

validatedev, 2019-12-24:
`
filename:       /lib/modules/5.0.0-37-generic/updates/huawei-wmi.ko
license:        GPL v2
description:    Huawei WMI laptop extras driver
author:         Ayman Bagabas <ayman.bagabas@gmail.com>
alias:          wmi:ABBC0F5B-8EA1-11D1-A000-C90629100000
srcversion:     527F74F63D3290F53B1A376
depends:        ledtrig-audio,wmi,sparse-keymap
retpoline:      Y
name:           huawei_wmi
vermagic:       5.0.0-37-generic SMP mod_unload 
parm:           battery_reset:Reset battery charge values to (0-0) before disabling it using (0-100) (bint)
parm:           report_brightness:Report brightness keys. (bint)
`
Yes I did all of the steps that you gave to me.

aymanbagabas, 2019-12-24:
Can you run `ls /sys/devices/platform/huawei-wmi`

validatedev, 2019-12-24:
> Can you run `ls /sys/devices/platform/huawei-wmi`

`
charge_control_thresholds  driver_override  input  modalias  subsystem
driver                     fn_lock_state    leds   power     uevent
`

aymanbagabas, 2019-12-24:
> `charge_control_thresholds driver_override input modalias subsystem driver fn_lock_state leds power uevent`

What about `ls /sys/devices/platform/huawei-wmi/leds`, also what happens on these two:
```
echo 0 | sudo tee /sys/devices/platform/huawei-wmi/leds/platform::micmute/brightness
echo 1 | sudo tee /sys/devices/platform/huawei-wmi/leds/platform::micmute/brightness
```

aymanbagabas, 2019-12-24:
Can you please test `fix-v3.3` again? I've updated the branch.

validatedev, 2019-12-25:
Well, `xdotool key XF86AudioMicMute` works. I tried the updated branch but nothing has changed.

aymanbagabas, 2019-12-25:
@validatedev I've updated `fix-v3.3` AGAIN! Sorry to put you through all this, I think I messed up and since v3.3 is now included in 5.5 we have to fix this ASAP. Can you please test the update and include a dmesg log. You can do `sudo rmmod huawei_wmi` then `make clean && make` and finally `sudo insmod huawei-wmi.ko` so you don't have to reboot.

validatedev, 2019-12-25:
Well, still doesn't work but I don't know the problem whether on the Ubuntu's side or driver's. Because `xdotool key XF86AudioMicMute` works flawlessly, including LED.

aymanbagabas, 2019-12-25:
You're probably missing the micmute key remap. Checkout the readme. And if you could upload a dmesg log that'd be so much help.

validatedev, 2019-12-25:
dmesg: 
[dmesg4.txt](https://github.com/aymanbagabas/Huawei-WMI/files/4000699/dmesg4.txt)

validatedev, 2019-12-25:
Good news! Your keymap fix has worked! And your last fix also fixed the micmute LED. Thanks for all your effort!

closed 2019-12-25 by aymanbagabas

aymanbagabas, 2019-12-26:
Can you test v3.4 and provide a dmesg log?

validatedev, 2019-12-27:
Right now I'm on the Ubuntu 20.04 with kernel 5.0.37-HWE-18.04 because of the bug between kernel 5.0 and 5.5, I used the one-time dkms install method and micmute LED doesn't work atm.
[dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/4004455/dmesg.txt)

validatedev, 2019-12-27:
@aymanbagabas could you look at there when you have time please?

aymanbagabas, 2019-12-27:
Please use v3.2 if you're using kernel 5.0. I've updated the readme. Thank
you

validatedev, 2019-12-27:
> Please use v3.2 if you're using kernel 5.0. I've updated the readme. Thank you
> […](#)

Yeah I can confirm that LED works without any issue with 3.2. Thank you too!