aymanbagabas/Huawei-WMI #30: Keyboard Backlight Preference is Resetting When Reboot or Shutdown

validatedev, 2019-12-25:
**Describe the bug**
I usually disable the keyboard backlight for battery consuming purposes. But after rebooting or shutdown-booting the PC, my keyboard backlight preference resets all time. Huawei Matebook 13 - Ubuntu 18.04.3 LTS.

**To Reproduce**
Steps to reproduce the behavior:
1. Disable the keyboard backlight
2. Reboot

**Expected behavior**
Keyboard backlight should retain its value

**Desktop (please complete the following information):**
 - Kernel [e.g. 5.2.9-200.fc30.x86_64] `5.0.0-23-generic #24~18.04.1-Ubuntu`
 - Distro [e.g. Fedora 30] `Ubuntu 18.04 LTS`
 - Loaded WMI modules 
`snd_rawmidi            36864  1 snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
huawei_wmi             16384  0
snd                    86016  25 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek
sparse_keymap          16384  1 huawei_wmi
wmi_bmof               16384  0
intel_wmi_thunderbolt    20480  0
wmi                    28672  3 intel_wmi_thunderbolt,huawei_wmi,wmi_bmof
`
 - Dmesg log 
[dmesg3.txt](https://github.com/aymanbagabas/Huawei-WMI/files/4000460/dmesg3.txt)

 - ALSA log if applicable 
 http://alsa-project.org/db/?f=d5ea51f9d8b6e1abccb45cd160026f5b8e1f2235

 - `evtest` if it's keyboard related. Run `sudo evtest` then choose "Huawei WMI hotkeys", test all hotkeys and submit output.

> `No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Huawei WMI hotkeys
/dev/input/event5:	Video Bus
/dev/input/event6:	HDA Intel PCH Headphone Mic
/dev/input/event7:	HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event8:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event9:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event10:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event11:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event12:	ELAN962C:00 04F3:30D0 Touchpad
/dev/input/event13:	HD Camera: HD Camera
Select the device event number [0-13]: 4
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "Huawei WMI hotkeys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 171 (KEY_CONFIG)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
    Event code 238 (KEY_WLAN)
    Event code 240 (KEY_UNKNOWN)
    Event code 248 (KEY_MICMUTE)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1577279080.126372, type 4 (EV_MSC), code 4 (MSC_SCAN), value 281
Event: time 1577279080.126372, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1577279080.126372, -------------- SYN_REPORT ------------
Event: time 1577279080.126603, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 0
Event: time 1577279080.126603, -------------- SYN_REPORT ------------
Event: time 1577279081.306404, type 4 (EV_MSC), code 4 (MSC_SCAN), value 282
Event: time 1577279081.306404, type 1 (EV_KEY), code 225 (KEY_BRIGHTNESSUP), value 1
Event: time 1577279081.306404, -------------- SYN_REPORT ------------
Event: time 1577279081.306480, type 1 (EV_KEY), code 225 (KEY_BRIGHTNESSUP), value 0
Event: time 1577279081.306480, -------------- SYN_REPORT ------------
Event: time 1577279091.370480, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1577279091.370480, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
Event: time 1577279091.370480, -------------- SYN_REPORT ------------
Event: time 1577279091.370911, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
Event: time 1577279091.370911, -------------- SYN_REPORT ------------
Event: time 1577279091.738040, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1577279091.738040, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 1
Event: time 1577279091.738040, -------------- SYN_REPORT ------------
Event: time 1577279091.738132, type 1 (EV_KEY), code 248 (KEY_MICMUTE), value 0
Event: time 1577279091.738132, -------------- SYN_REPORT ------------
Event: time 1577279094.542213, type 4 (EV_MSC), code 4 (MSC_SCAN), value 289
Event: time 1577279094.542213, type 1 (EV_KEY), code 238 (KEY_WLAN), value 1
Event: time 1577279094.542213, -------------- SYN_REPORT ------------
Event: time 1577279094.542308, type 1 (EV_KEY), code 238 (KEY_WLAN), value 0
Event: time 1577279094.542308, -------------- SYN_REPORT ------------
Event: time 1577279095.238368, type 4 (EV_MSC), code 4 (MSC_SCAN), value 289
Event: time 1577279095.238368, type 1 (EV_KEY), code 238 (KEY_WLAN), value 1
Event: time 1577279095.238368, -------------- SYN_REPORT ------------
Event: time 1577279095.238816, type 1 (EV_KEY), code 238 (KEY_WLAN), value 0
Event: time 1577279095.238816, -------------- SYN_REPORT ------------
Event: time 1577279096.242606, type 4 (EV_MSC), code 4 (MSC_SCAN), value 28a
Event: time 1577279096.242606, type 1 (EV_KEY), code 171 (KEY_CONFIG), value 1
Event: time 1577279096.242606, -------------- SYN_REPORT ------------
Event: time 1577279096.242850, type 1 (EV_KEY), code 171 (KEY_CONFIG), value 0
Event: time 1577279096.242850, -------------- SYN_REPORT ------------
^[[2~
`

validatedev, 2019-12-25:
When try to install fix-3.3 again because of the downgrading the kernel:
[err.txt](https://github.com/aymanbagabas/Huawei-WMI/files/4000481/err.txt)

There looks like some errors.

aymanbagabas, 2019-12-25:
This is because Ubuntu uses signed modules and you're compiling this module out-of-tree. Thus, you're getting SSL errors but has nothing to do with the build, it builds fine and generates `huawei-wmi.ko`.

closed 2019-12-25 by aymanbagabas

validatedev, 2019-12-25:
Interestingly, after upgrading the kernel 5.0.2x to 5.0.3x (they are HWE kernels) and apply your v3.3 fix, it recognizes the state.