aymanbagabas/Huawei-WMI #31: Huawei-WMI Driver not installed - But it is.

mircoianese, 2019-12-25:
Hello, I have a 2018 Matebook X Pro with Elementary OS installed. I've installed Huawei-WMI using dkms and matebook-applet (by adding the apt repository).

When I try to start matebook-applet from terminal I get:

`ERROR: 2019/12/25 14:46:14 main.go:91: Neither a supported version of Huawei-WMI driver, nor any of the required scripts are properly installed, see README.md#installation-and-setup for instructions`

But when I run `sudo dkms install huawei-wmi/3.3` I get:

`Module huawei-wmi/3.3 already installed on kernel 5.0.0-37-generic/x86_64`

What I am missing here?
Thanks a lot

nekr0z, 2019-12-25:
Try running `matebook-applet -vv` for more information. Also, after installing the driver did you make sure the driver loaded correctly (manually or during a reboot)?

aymanbagabas, 2019-12-25:
Can you please try the `fix-v3.3` branch and install it manually?

aymanbagabas, 2020-01-23:
Closing for being idle.

closed 2020-01-23 by aymanbagabas