aymanbagabas/Huawei-WMI #32: Charge_control thresholds not found

RuSern, 2019-12-29:
**Describe the bug**
Hi, I have installed Ubuntu 18.04 on my Matebook 13. I followed your instructions and installed the v3.2 driver. However, after installation I cannot find the "sys/class/power_supply/BAT0/charge_control_{start,end}_threshold or /sys/devices/platform/huawei-wmi/charge_control_thresholds" files for controlling the thresholds. I tried then to upgrade to Ubuntu 19.10 to use the v3.4 driver, but  I still cant find the files. I performed the installation using dkms tarball. Could you help me out?

**Desktop (please complete the following information):**
 - Kernel [5.3] 
 - Distro [Ubuntu 19.10] `
 - Loaded WMI modules `lsmod | grep wmi`
snd_rawmidi            36864  1 snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
mxm_wmi                16384  1 nouveau
wmi_bmof               16384  0
intel_wmi_thunderbolt    20480  0

Maybe I did not understand your installation instructions properly, could you help me out? Thanks.

RuSern, 2019-12-29:
I saw there was a previous issue similar that was similar to mine, and it was  solved by rebooting. However, I have also tried that but it doesnt seem to solve my problem

aymanbagabas, 2019-12-29:
There is a bug with DKMS > 2.6 where it doesn't update modules that already
exist in the kernel. Please try manual installation.

RuSern, 2019-12-29:
I tried to build it from source using your instructions:
git clone https://github.com/aymanbagabas/Huawei-WMI
cd Huawei-WMI
make
# To update use:
sudo cp huawei-wmi.ko /lib/modules/$(uname -r)/updates/
sudo depmod
# To install use:
sudo make install

When I ran sudo make install, this is the error message that returned:
make -C /lib/modules/5.3.0-24-generic/build/ M=/home/rusern/Huawei-WMI modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-24-generic'
  INSTALL /home/rusern/Huawei-WMI/huawei-wmi.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.3.0-24-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-24-generic'
depmod -a

aymanbagabas, 2019-12-29:
That's a SSL error, shouldn't affect anything. Did you reboot?

RuSern, 2019-12-29:
Oh sorry.
I rebooted my device, but I still dont see the files in the stated location.

[image]

[image]

nekr0z, 2019-12-29:
What's the output of `lsmod | grep wmi` now that you've installed the driver manually and rebooted?

RuSern, 2019-12-30:
Hi, this is what I got:
[image]

RuSern, 2019-12-30:
Hi, this is what I got:

lsmod | grep wmi
snd_rawmidi            36864  1 snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
mxm_wmi                16384  1 nouveau
snd                    86016  19 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi
intel_wmi_thunderbolt    20480  0
wmi_bmof               16384  0
wmi                    32768  4 intel_wmi_thunderbolt,wmi_bmof,mxm_wmi,nouveau

aymanbagabas, 2020-01-23:
> Hi, this is what I got:
>
> lsmod | grep wmi
>
> ```
> snd_rawmidi            36864  1 snd_seq_midi
> snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
> mxm_wmi                16384  1 nouveau [...]

The driver is clearly not loaded. Closing.

closed 2020-01-23 by aymanbagabas