aymanbagabas/Huawei-WMI #33: Mic mute led does not work on Matebook 14.

zypangpang, 2020-01-11:
I manually compiled and installed. The battery charge threshold works perfectly, but the mic mute led does not work. I don' t think this is a problem of your driver, because it works in Arch.  I suspect it is caused by not loading right drivers in kernel, since I use gentoo with self configured and compiled kernel. I thought it needed the LED driver, then I enabled all kernel LED drivers but with no luck.
The log shows some errors about micmute. Maybe you can tell what is wrong or missing. 

The output of hwinfo | grep -i led:
`FPU            : installed
failed to read /dev/mem
  detailed timings:
  H: Handlers=sysrq kbd leds event4 
  B: LED=7
  H: Handlers=sysrq kbd leds event12 
  B: LED=1f
  handlers = sysrq kbd leds event4
  handlers = sysrq kbd leds event12
wlp0s20f3: socket failed: Operation not permitted
  read_block0: open(/dev/nvme0n1) failed
  E: LED=1f
  E: SUBSYSTEM=leds
  E: SUBSYSTEM=leds
  E: SUBSYSTEM=leds
  E: SUBSYSTEM=leds
  E: SUBSYSTEM=leds
  P: /devices/pci0000:00/0000:00:14.3/leds/phy0-led
  E: DEVPATH=/devices/pci0000:00/0000:00:14.3/leds/phy0-led
  E: SUBSYSTEM=leds
  P: /devices/platform/huawei-wmi/leds/platform::micmute
  E: DEVPATH=/devices/platform/huawei-wmi/leds/platform::micmute
  E: SUBSYSTEM=leds
  E: LED=7
  E: SUBSYSTEM=leds
  E: SUBSYSTEM=leds
  E: SUBSYSTEM=leds
/devices/pci0000:00/0000:00:14.3/leds/phy0-led
/devices/platform/huawei-wmi/leds/platform::micmute
  prop read: rdCR.lZF+r4EgHp4 (failed)
  old prop read: rdCR.lZF+r4EgHp4 (failed)
  prop read: rdCR.n_7QNeEnh23 (failed)
  old prop read: rdCR.n_7QNeEnh23 (failed)
  prop read: rdCR.EMpH5pjcahD (failed)
  old prop read: rdCR.EMpH5pjcahD (failed)
  prop read: rdCR.f5u1ucRm+H9 (failed)
  old prop read: rdCR.f5u1ucRm+H9 (failed)
  prop read: rdCR.8uRK7LxiIA2 (failed)
  old prop read: rdCR.8uRK7LxiIA2 (failed)
  prop read: rdCR.9N+EecqykME (failed)
  old prop read: rdCR.9N+EecqykME (failed)
  prop read: rdCR.CxwsZFjVASF (failed)
  old prop read: rdCR.CxwsZFjVASF (failed)
  prop read: RE4e.2p1+wDoPVY7 (failed)
  old prop read: RE4e.2p1+wDoPVY7 (failed)
  prop read: I+Ba.oRxb6OTtdD3 (failed)
  old prop read: I+Ba.oRxb6OTtdD3 (failed)
  prop read: BUZT.8tHHrMSckV9 (failed)
  old prop read: BUZT.8tHHrMSckV9 (failed)
  prop read: 8otl.9DHlPhQgPW0 (failed)
  old prop read: 8otl.9DHlPhQgPW0 (failed)
  prop read: Dhtl.jFWd3H1pM2B (failed)
  old prop read: Dhtl.jFWd3H1pM2B (failed)
  prop read: WnlC.4IgHDuwHho2 (failed)
  old prop read: WnlC.4IgHDuwHho2 (failed)
  prop read: VCu0.RVN8AkDkdAE (failed)
  old prop read: VCu0.RVN8AkDkdAE (failed)
  prop read: W60f.DsovyAdnHdB (failed)
  old prop read: W60f.DsovyAdnHdB (failed)
  prop read: 6NW+.C5pKJ5dMBB0 (failed)
  old prop read: 6NW+.C5pKJ5dMBB0 (failed)
  prop read: nS1_.CXdX8HLeBw9 (failed)
  old prop read: nS1_.CXdX8HLeBw9 (failed)
  prop read: qLht.yOu3vAJiJzA (failed)
  old prop read: qLht.yOu3vAJiJzA (failed)
  prop read: CLZK.7Vp9NpAdGv4 (failed)
  old prop read: CLZK.7Vp9NpAdGv4 (failed)
  prop read: Rgik.H1+rdpLA6r2 (failed)
  old prop read: Rgik.H1+rdpLA6r2 (failed)
  prop read: 1GTX.NHcUWUZO_TD (failed)
  old prop read: 1GTX.NHcUWUZO_TD (failed)
  prop read: iL_V.qlP9KASjzIE (failed)
  old prop read: iL_V.qlP9KASjzIE (failed)
  prop read: 5Dex.nVIpzckPywA (failed)
  old prop read: 5Dex.nVIpzckPywA (failed)
  prop read: _Znp.RUSOmWP4ob1 (failed)
  old prop read: _Znp.RUSOmWP4ob1 (failed)
  prop read: MZfG.9j+6KsNGXP8 (failed)
  old prop read: MZfG.9j+6KsNGXP8 (failed)
  prop read: rdCR.WekvYnKwov9 (failed)
  old prop read: rdCR.WekvYnKwov9 (failed)
  prop read: wLCS.VtNQfCtlGlE (failed)
  old prop read: wLCS.VtNQfCtlGlE (failed)
  prop read: cS_q.SE1wIdpsiiC (failed)
  old prop read: cS_q.SE1wIdpsiiC (failed)
  prop read: 3eEv.SE1wIdpsiiC (failed)
  old prop read: 3eEv.SE1wIdpsiiC (failed)
  prop read: XpUz.SE1wIdpsiiC (failed)
  old prop read: XpUz.SE1wIdpsiiC (failed)
  prop read: __k1.SE1wIdpsiiC (failed)
  old prop read: __k1.SE1wIdpsiiC (failed)
  prop read: RA+5.SE1wIdpsiiC (failed)
  old prop read: RA+5.SE1wIdpsiiC (failed)
  prop read: uLFA.SE1wIdpsiiC (failed)
  old prop read: uLFA.SE1wIdpsiiC (failed)
  prop read: aoQf.SE1wIdpsiiC (failed)
  old prop read: aoQf.SE1wIdpsiiC (failed)
  prop read: 1_gj.SE1wIdpsiiC (failed)
  old prop read: 1_gj.SE1wIdpsiiC (failed)
  prop read: U9xn.SE1wIdpsiiC (failed)
  old prop read: U9xn.SE1wIdpsiiC (failed)
  prop read: jzeG.SE1wIdpsiiC (failed)
  old prop read: jzeG.SE1wIdpsiiC (failed)
  prop read: A9vK.SE1wIdpsiiC (failed)
  old prop read: A9vK.SE1wIdpsiiC (failed)
  prop read: dK9P.SE1wIdpsiiC (failed)
  old prop read: dK9P.SE1wIdpsiiC (failed)
  prop read: 4WPT.SE1wIdpsiiC (failed)
  old prop read: 4WPT.SE1wIdpsiiC (failed)
  prop read: YhfX.SE1wIdpsiiC (failed)
  old prop read: YhfX.SE1wIdpsiiC (failed)
  prop read: +svb.SE1wIdpsiiC (failed)
  old prop read: +svb.SE1wIdpsiiC (failed)
  prop read: xnLL.v5OBcNn2we7 (failed)
  old prop read: xnLL.v5OBcNn2we7 (failed)
  prop read: fHJ1.l3v0gVevrhC (failed)
  old prop read: fHJ1.l3v0gVevrhC (failed)
  prop read: FKGF.h6VbFfok0SF (failed)
  old prop read: FKGF.h6VbFfok0SF (failed)
  prop read: lfzD.CnuP7VbJhG1 (failed)
  old prop read: lfzD.CnuP7VbJhG1 (failed)
  prop read: csaW.Iqb1w4eiQc6 (failed)
  old prop read: csaW.Iqb1w4eiQc6 (failed)
  prop read: k4bc.2DFUsyrieMD (failed)
  old prop read: k4bc.2DFUsyrieMD (failed)
  prop read: DrDI.7RzpnbkIFo5 (failed)
  old prop read: DrDI.7RzpnbkIFo5 (failed)
  prop read: 32ra.NMc1t9M02C6 (failed)
  old prop read: 32ra.NMc1t9M02C6 (failed)
  prop read: ADDn.xvTGa+XkbXF (failed)
  old prop read: ADDn.xvTGa+XkbXF (failed)
  prop read: pBe4.xYNhIwdOaa6 (failed)
  old prop read: pBe4.xYNhIwdOaa6 (failed)
  prop read: c3zD.+49ps10DtUF (failed)
  old prop read: c3zD.+49ps10DtUF (failed)
  prop read: AH6Q.pBarWtRCs3F (failed)
  old prop read: AH6Q.pBarWtRCs3F (failed)
  prop read: rdCR.j8NaKXDZtZ6 (failed)
  old prop read: rdCR.j8NaKXDZtZ6 (failed)
  prop read: wkFv.j8NaKXDZtZ6 (failed)
  old prop read: wkFv.j8NaKXDZtZ6 (failed)
  prop read: +rIN.j8NaKXDZtZ6 (failed)
  old prop read: +rIN.j8NaKXDZtZ6 (failed)
  prop read: 4zLr.j8NaKXDZtZ6 (failed)
  old prop read: 4zLr.j8NaKXDZtZ6 (failed)
  prop read: 94PJ.j8NaKXDZtZ6 (failed)
  old prop read: 94PJ.j8NaKXDZtZ6 (failed)
  prop read: EBSn.j8NaKXDZtZ6 (failed)
  old prop read: EBSn.j8NaKXDZtZ6 (failed)
  prop read: JIVF.j8NaKXDZtZ6 (failed)
  old prop read: JIVF.j8NaKXDZtZ6 (failed)
  prop read: OPYj.j8NaKXDZtZ6 (failed)
  old prop read: OPYj.j8NaKXDZtZ6 (failed)
  prop read: wl2P.nfEXJulusj6 (failed)
  old prop read: wl2P.nfEXJulusj6 (failed)
  prop read: 63Vg.ndpeucax6V1 (failed)
  old prop read: 63Vg.ndpeucax6V1 (failed)
  prop read: ZsBS.GQNx7L4uPNA (failed)
  old prop read: ZsBS.GQNx7L4uPNA (failed)
  <4>[    3.088162] thermal thermal_zone3: failed to read out thermal zone (-61)
  <6>[    4.866978] iwlwifi 0000:00:14.3: Conflict between TLV & NVM regarding enabling LAR (TLV = enabled NVM =disabled)
  <6>[    5.094949] iwlwifi 0000:00:14.3: Conflict between TLV & NVM regarding enabling LAR (TLV = enabled NVM =disabled)
  <6>[    8.339789] iwlwifi 0000:00:14.3: Conflict between TLV & NVM regarding enabling LAR (TLV = enabled NVM =disabled)
  <3>[   11.447850] iwlwifi 0000:00:14.3: Unhandled alg: 0x707
  ledtrig_audio 16384 1 huawei_wmi, Live 0x0000000000000000
  BIOS Keyboard LED Status:
  Memory Range: 0x91227000-0x91227fff (rw,non-prefetchable,disabled)
  Memory Range: 0x000c0000-0x000dffff (rw,non-prefetchable,disabled)
  Detailed Timings #0:`

lsmod:
`hid_multitouch         28672  0
iwldvm                196608  0
i2c_designware_platform    16384  0
huawei_wmi             24576  0
i2c_designware_core    20480  1 i2c_designware_platform
ledtrig_audio          16384  1 huawei_wmi
wmi_bmof               16384  0
intel_wmi_thunderbolt    16384  0
x86_pkg_temp_thermal    20480  0
serio_raw              16384  0
iwlmvm                352256  0
btusb                  49152  0
btrtl                  16384  1 btusb
btbcm                  16384  1 btusb
btintel                20480  1 btusb
mousedev               24576  0
iwlwifi               335872  2 iwldvm,iwlmvm
bluetooth             425984  25 btrtl,btintel,btbcm,btusb
intel_lpss_pci         20480  0
intel_pch_thermal      16384  0
intel_lpss             16384  1 intel_lpss_pci
mfd_core               16384  1 intel_lpss
wmi                    24576  3 intel_wmi_thunderbolt,huawei_wmi,wmi_bmof
i2c_hid                28672  0
pinctrl_cannonlake     36864  1
pinctrl_intel          24576  1 pinctrl_cannonlake
efivarfs               16384  1`

Kernel:
`Linux gentoo-zypang 5.3.18-gentoo #4 SMP Sat Jan 11 10:47:13 CST 2020 x86_64 Intel(R) Core(TM) i5-8265U CPU @ 1.60GHz GenuineIntel GNU/Linux`

The lsmod of a working Arch:
[lsmod.txt](https://github.com/aymanbagabas/Huawei-WMI/files/4049117/lsmod.txt)

Kernel config about LED drivers:
[image]

Thanks in advance!

aymanbagabas, 2020-01-11:
The driver doesn't depend on any LED driver, hence the name `platform::micmute`, it uses ACPI WMI to control the LED and then uses `ledtrig_audio` to hook it up with ALSA. However, I will look more into this log. Meanwhile, could you please provide a dmesg log with the issue.

Do you have `/sys/devices/platform/huawei-wmi/leds/platform::micmute` in your system? Can you control the LED by writing to `/sys/devices/platform/huawei-wmi/leds/platform::micmute/brightness`?

aymanbagabas, 2020-01-11:
Aha I don't see any codecs in your Gento lsmod. Realtek sound codecs are used to enable the `ledtrig_audio`. Taken from your Arch lsmod `snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek`. Make sure you have enabled sound. The LED should work though but not through ALSA. You should be able to turn it on/off manually.

zypangpang, 2020-01-12:
Thanks! I got it worked today. I tweaked a little in my kernel config.  Previously I built the codecs in the kernel. Now I build it as a module. Then the led works. I also changed some other settings in kernel config, but I think building the codecs as modules solved the problem. :)

aymanbagabas, 2020-01-12:
Glad to hear that

closed 2020-01-12 by aymanbagabas