aymanbagabas/Huawei-WMI #34: usermod: group 'huawei-wmi' does not exist

Futureknows, 2020-06-12:
Fedora 32. Installed Huawei-WMI (huawei-wmi-3.4-1dkms.noarch.rpm).
Installed the batpro script. 
Reboot.
Trying to add user to group gives error: 
`$ sudo usermod -a -G huawei-wmi user
usermod: group 'huawei-wmi' does not exist

Running batpro status gives error:
`$ sudo batpro status`
`iopl: Operation not permitted
^Ciopl failed: You may need to run as root or give the process the CAP_SYS_RAWIO
capability. On non-x86 architectures, this operation probably isn't possible.
`

What am I missing?

aymanbagabas, 2020-06-12:
You don't need a new user, you can use `sudo` instead. I don't think this is related to the driver, please refer to [matebook-applet](https://github.com/nekr0z/matebook-applet) and open a new issue there since this is an issue with `batpro`.

closed 2020-06-12 by aymanbagabas

nekr0z, 2020-06-12:
> What am I missing?

You're missing a lot of things.

You're trying to use `batpro` on Linux Kernel >5.0, which is a bad idea. You're trying to add an user to work with `batpro` instead of properly using `sudo` as advised in the repository that holds `batpro`. You're confusing this repository that holds the driver with [Rouven's](https://github.com/qu1x/huawei-wmi) that holds the user/group/systemd things.