aymanbagabas/Huawei-WMI #35: [Question _ not a bug] What is the expected behavior ?

jeannich, 2020-07-07:
Hi,

_This is a question, not a bug report.  
Kindly redirect me if there is another channel for such questions._  

I have a MACH-WX9 (Matebook X Pro) and I am using Huawei-WMI 3.4 on Linux Mint 20.  
Huawei-WMI seems to work correctly but the behavior is a bit strange to me so I am wondering if my laptop is behaving correctly or not.  
I have setup those thresholds: min 40% and max 70%  

Several cases:  
- battery at 85% (just after I installed huawei-wmi and configured 40%-70%), laptop ON, plug in the charger:  
  Observed result: Laptop charges at arround 5W and stays forever at 85%  
- battery at 30%, laptop ON, plug in the charger:  
  Observed result: The laptop will charge at full speed (around 40W) until 69% then power charge drops to 5W and laptop battery stays at 69%  
- battery at 50%, laptop OFF, plug in the charger:  
  Observed result: I left the laptop in this charging condition for a whole night but in the morning the laptop was still at 50% (it may have increased to 51 or 52 but I am not sure)  

Are those observations matching the expected behavior ?  

How could we described the overall behavior ?  
I'll try this, correct me if I am wrong:  
- If charger is plugged in when battery is **within** or **above** configured charge thresholds (40-70 in my case), laptop will charge at a very low rate (around 5w) just to prevent battery level to decrease
- If charger is plugged in when battery is below configured charge thresholds, then the laptop will charge at full speed until max threshold is reached
- those 2 rules applies when laptop is both ON or OFF

aymanbagabas, 2020-07-07:
Your observation is correct. Mine have the same behavior. If the battery is below the thresholds, it charges until it reaches the high threshold. If the battery is within or above the thresholds, it stays that way and prevents the battery from dropping.

If I'm not mistaken, using the Windows 10 would give you the same behavior, meaning that's just how this feature works on a lower level.

jeannich, 2020-07-07:
Thanks @aymanbagabas, good to know that you have the same behavior and also to know that Windows is behaving the same.

Actually, is this behavior driven by the linux kernel/driver or is it decided by the battery itself?

If it is the battery that decides when and how to charge (based on the values sent by this Huawei-WMI driver of course), then it could be interesting to document observed behavior so that Huawei-WMI users knows what to expect.
It could also help to track potential unexpected battery decisions.

For instance, this afternoon, I stopped by laptop when it had 50% battery, and I plugged it.
2 hours later it was fully charged at 99% ... strange behavior as it does not follow the pattern I described above.

Maybe sometimes the battery decides by itself to fully charge and sometimes not?

Anyway, could be nice to document this and maybe add it to the README of this project (I can propose a PR if you are OK with this idea)

aymanbagabas, 2020-07-09:
> Actually, is this behavior driven by the linux kernel/driver or is it decided by the battery itself?

No, it's the battery itself.

> Anyway, could be nice to document this and maybe add it to the README of this project (I can propose a PR if you are OK with this idea)

I mean this project is not limited to the battery protection feature but it tries implements the whole Huawei ACPI interface. As far as the battery goes, that's just the default behavior of the battery which happens no matter what operating system you're running. I don't see the point of reaffirming the facts of this behavior, so thanks no need to. However, if you're up to contributing the actual driver, that would be a greater help for the community and this project.

closed 2020-07-09 by aymanbagabas