aymanbagabas/Huawei-WMI #36: Start and end thresholds getting reset

JeffR1992, 2020-07-11:
I'm current using Ubuntu 18.04.4 LTS, and noticed that the values that I set in `/sys/class/power_supply/BAT0`, in the `charge_control_start_threshold` and `charge_control_end_threshold` files, are getting reset to 0 after rebooting my Matebook X Pro a few times. Is there a way to prevent this?

nekr0z, 2020-07-11:
> getting reset to 0 after rebooting my Matebook X Pro a few times

This is expected, known behaviour of the chipset. Happens on Windows as well, Huawei PC Manager monitors that and re-instates the thresholds to what user had selected. Taking care of this is not the driver's job and can be easily handled in userspace. There's [a project](https://github.com/qu1x/huawei-wmi) for that.

JeffR1992, 2020-07-12:
Thanks, decided to use chmod 444.

nekr0z, 2020-07-13:
> Thanks, decided to use chmod 444.

That alone won't help against the issue you've described.

aymanbagabas, 2020-10-20:
Closing due to inactivity.

closed 2020-10-20 by aymanbagabas