aymanbagabas/Huawei-WMI #38: Add keyboard backlight timeout as an option

german-rios-gonzalez, 2020-08-12:
Hi, first of all thank you very much for this project, the effort and the great results, I came here because by default the keyboard backlight timeout it's about 15 seconds on my Matebook D14 2020, and currently I can't find a way to set the backlight always on, I don't have any knowledge about the kernel internals but I guess you can add this option on this project, this would be wonderful for people that work in dark conditions.

nekr0z, 2020-08-12:
Do you know for a fact that changing keyboard backlight timeout is at all possible on yout MateBook?

I don’t really know about D14, but at least for the MateBook 13 I own I know no way to tweak that on any operating system. I was actually under impression it wasn’t possible on any MateBooks.

(Originally published at: https://evgenykuznetsov.org/en/reactions/2020/matebook-keyb-backlight/)

german-rios-gonzalez, 2020-08-12:
Hi, @nekr0z I have the model NBL-WAQ9R Matebook D14 2020, the software that comes preinstalled with this laptop have the option to never turn off, I wiped out this operative system but this article that show what I'm talking about.

https://consumer.huawei.com/en/support/content/en-us00723328/#:~:text=Open%20PC%20Manager%2C%20click%20the,to%20turn%20on%20the%20backlight.

nekr0z, 2020-08-13:
Wow! I had no idea! @aymanbagabas, this is definitely worth looking into.

aymanbagabas, 2020-08-13:
Interesting! Could you please provide an ACPI dump? 
https://01.org/linux-acpi/utilities

german-rios-gonzalez, 2020-08-14:
Hi, @aymanbagabas thanks for the link, I followed the instructions and I attached the zipped result file. (For the record I use Fedora as operative system)
[acpidump.zip](https://github.com/aymanbagabas/Huawei-WMI/files/5073040/acpidump.zip)

aymanbagabas, 2020-08-14:
This looks promising, these 2 methods 'looks' like the ones to get/set the keyboard backlight (assuming `KBBL` stands for KeyBoard BackLight). They appear in the HWMI interface in SSDT5 under 0x0602 and 0x0702.

```
    Method (GLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100){})
        Name (CONT, Zero)
        Local0 = Arg0
        CreateByteField (BUFF, Zero, STAT)
        CreateByteField (BUFF, One, SURV)
        CreateByteField (BUFF, 0x02, KBBL)
        CONT = Zero
        Acquire (\_SB.PCI0.LPC0.EC0.ECMT, 0x03E8)
        \_SB.PCI0.LPC0.EC0.ECM3 (0xC4)
        While (((CONT != 0x0A) && ((\_SB.PCI0.LPC0.EC0.ECXC & One) == Zero)))
        {
            CONT += One
            Sleep (0x32)
        }

        KBBL = \_SB.PCI0.LPC0.EC0.ECXD
        Release (\_SB.PCI0.LPC0.EC0.ECMT)
        STAT = Zero
        Return (BUFF) /* \GLIV.BUFF */
    }

    Method (SLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100){})
        Local0 = Arg0
        CreateByteField (Arg0, 0x02, KBBV)
        CreateByteField (BUFF, Zero, STAT)
        Acquire (\_SB.PCI0.LPC0.EC0.ECMT, 0x03E8)
        If ((((KBBV == 0x04) || (KBBV == 0x08)) || (KBBV == 0x10)))
        {
            \_SB.PCI0.LPC0.EC0.ECM3 (0xC5)
            \_SB.PCI0.LPC0.EC0.ECD3 (KBBV)
            STAT = Zero
        }
        Else
        {
            STAT = One
        }

        Release (\_SB.PCI0.LPC0.EC0.ECMT)
        Return (BUFF) /* \SLIV.BUFF */
    }
```

Could you try these commands out and report the output
* `echo 0x000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg`
* `sudo cat /sys/kernel/debug/huawei-wmi/arg`
Cycle through the keyboard backlight levels and run the same commands.

german-rios-gonzalez, 2020-08-14:
Hi, @aymanbagabas I get these results for the 3 states of the keyboard backlight 

Keyboard backlight turn off:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x000602

[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x00,0x00,0x04,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

Keyboard backlight level 1:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x000602

[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

Keyboard backlight level 2:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x000602

[{0x00,0x00,0x00,0x00},{0x00,0x00,0x10,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

german-rios-gonzalez, 2020-08-14:
Hi, I'm sorry I was start it before see you change the last command, so this is the results

Keyboard backlight turn off:

echo 0x000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x000602

[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

Keyboard backlight level 1:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x000602

[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x00,0x00,0x08,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

Keyboard backlight level 2:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x000602

[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

aymanbagabas, 2020-08-14:
Hahaa, bingo! I gotta try this in my machine. You can see the changes here in the first and last cases (the second seems odd). Looking at the methods, the available values are 0x04, 0x08, and 0x10.

Could you try these with each value and observe the changes to the keyboard backlight
* `echo 0x[VALUE]0702 | sudo tee /sys/kernel/debug/huawei-wmi/arg`
* `sudo cat /sys/kernel/debug/huawei-wmi/call`

german-rios-gonzalez, 2020-08-14:
Hi, I can't see any difference maybe I do it wrong?

Keyboard backlight level 2 and 0x04:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x0x040702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x0x040702

[root@Huawei-NBL-WAQ9R ~]# sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

Keyboard backlight level 2 and 0x08:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x0x080702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x0x080702

[root@Huawei-NBL-WAQ9R ~]# sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

Keyboard backlight level 2 and 0x10:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x0x100702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x0x100702

[root@Huawei-NBL-WAQ9R ~]# sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

aymanbagabas, 2020-08-14:
Ok, there is a catch, you gotta run each set of commands twice to get a result, otherwise, you'll get 0x01.

german-rios-gonzalez, 2020-08-14:
Sorry for the delay, but I think I screwed again, because I can't see any change, for example this is the value that I need to echo 0x0x040702?? or is 0x040702?

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x0x040702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x0x040702
[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x0x040702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x0x040702
[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x0x080702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x0x080702
[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x0x080702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x0x080702
[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x [... cut ...] x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

aymanbagabas, 2020-08-14:
No, it's gonna be in this form 0x080702

german-rios-gonzalez, 2020-08-14:
My bad, here are the results now

Keyboard backlight level 2 and 0x04:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x040702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x040702

[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x040702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x040702
[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

Keyboard backlight level 2 and 0x08:

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x080702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x080702
[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

[german.rios.gonzalez@Huawei-NBL-WAQ9R ~]$ echo 0x080702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x080702
[root@Huawei-NBL-WAQ9R ~]# cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00 [... cut ...] x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]

aymanbagabas, 2020-08-14:
Did the keyboard backlight change?

german-rios-gonzalez, 2020-08-14:
No, it didn't change the intensity of the light or the time out.

german-rios-gonzalez, 2020-08-14:
Yeah, after the 0x04 the second time it turn off the backlight, the 0x08 the second time turn on the backlight in level 1, and the 0x10 the second time turn on the backlight on the level 2

aymanbagabas, 2020-08-14:
@nekr0z can you confirm this is working on your machine?

nekr0z, 2020-08-14:
> @nekr0z can you confirm this is working on your machine?

Yes and no. 0x04 followed by a call switches backight to full on my machine (no need to do the commands twice either, once is enough), while 0x08 and 0x10 do nothing. And neither of these commands do anything to the duration of the backlight.

aymanbagabas, 2020-08-14:
I guess that's still good news! It means we can control the keyboard backlight from the software. I'll look more into the timeout duration.

nekr0z, 2020-08-14:
As for the timeout duration, I don't believe this option was available before 2020, so it may be exclusive to the newer models. I definitely remember the corresponding woes on a forum dedicated to WRT-W19.

That being said, it's cool that keyboard backlight can be controlled from software — even on my model (and I guess on MBX as well?); it's even more cool that at least new models allow for backlight duration control; and it's absolutely great that you're on the right track to figuring these things out. ;)

Don't hesitate to ping me if anything needs testing on a WRT-W19.

aymanbagabas, 2020-10-20:
@german-rios-gonzalez can you provide a `sudo dmidecode` for your computer? I need the device's dmi id to match against it in the driver.

german-rios-gonzalez, 2020-10-20:
Hi @aymanbagabas off course

# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.11.1 present.
# SMBIOS implementations newer than version 3.2.0 are not
# fully supported by this version of dmidecode.
Table at 0xAE2DD000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: HUAWEI
	Version: 1.07
	Release Date: 06/02/2020
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 8 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		Smart battery is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.7
	Firmware Revision: 1.7

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: HUAWEI
	Product Name: NBLK-WAX9X
	Version: M1010
	Serial Number: M6TPM20429000448
	UUID: ed2f74d6-1c64-49e2-b46f-82706b52d9dc
	Wake-up Type: Power Switch
	SKU Number: C331
	Family: MateBook D

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: HUAWEI
	Product Name: NBLK-WAX9X-PCB
	Version: M1010
	Serial Number: PM03C0204R006015
	Asset Tag: NULL
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: NULL
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 23 bytes
Chassis Information
	Manufacturer: HUAWEI
	Type: Notebook
	Lock: Not Present
	Version: M1010
	Serial Number: M6TPM20429000448
	Asset Tag: NULL
	Boot-up State: Safe
	Power Supply State: Safe
[... 275 lines cut ...]
Portable Battery
	Location: Rear cover inside
	Manufacturer: Desay
	Manufacture Date: 3/4/2020
	Serial Number: 1
	Name: HB4692Z9ECW-41
	Chemistry: Lithium Ion
	Design Capacity: 0 mWh
	Design Voltage: 15280 mV
	SBDS Version: Not Specified
	Maximum Error: 1%
	OEM-specific Information: 0x00000048

Handle 0x0017, DMI type 43, 31 bytes
TPM Device
	Vendor ID: AMD
	Specification Version: 2.0
	Firmware Revision: 3.34
	Description: TPM Device
	Characteristics:
		TPM Device characteristics not supported
	OEM-specific Information: 0x00000000

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table

aymanbagabas, 2020-10-20:
Please try https://github.com/aymanbagabas/Huawei-WMI/tree/feat/kbdlight and report the behavior. FYI, I haven't tested it on my machine (I don't have it on me at the moment) so you might encounter kernel panics 🤞 

@nekr0z I'd love it if you can test this too. You'd need to load the driver with `handle_kbdlight=1` or use the sysfs interface because your device handles this somewhere else, _assuming the BIOS_, the same goes to my device MACH-WX9. The sysfs interface takes `0`, `1`, or `2` for off, level 1, level 2 respectively.

nekr0z, 2020-10-20:
Ooh, interesting, it works! Well, somewhat. ;-)

I installed the patched driver onto my machine and rebooted. Nothing new.

Then, I `rmmod`ed the driver and `insmod`ed it with `handle_kbdlight=1`. `kbdlight` interface appeared under `/sys/devices/platform/huawei-wmi/`. So far so good.

Now, my WRT-W19 has 3 modes of keyboard lighting: "off" (self-explanatory), "half" (low brighness for about 15 seconds, then turning off), and "full" (bright about 15 seconds, then turning off). The modes are controlled by a dedicated key (combined with F3) and switch in cycle, with one caveat: if the keyboard is in "half" or "full" mode _and_ the lighting has timed out, pushing the key doesn't switch to the next mode, it only turns the lighting on.

Well, here are the results of my playing with `kbdlight` interface:

* reads (all the results reproduce reliably, no one-offs):
  * keyboard in "off":
    ```
    $ cat /sys/devices/platform/huawei-wmi/kbdlight
    cat: /sys/devices/platform/huawei-wmi/kbdlight: No such device
    ```
  * keyboard in "half":
    ```
    $ cat /sys/devices/platform/huawei-wmi/kbdlight
    3
    ```
  * keyboard in "full":
    ```
    $ cat /sys/devices/platform/huawei-wmi/kbdlight
    4
    ```
* writes (all the results reproduce reliably, behavior indepentent of what the keyboard lighting is initially set to):
  * `$ echo 0 | sudo tee /sys/devices/platform/huawei-wmi/kbdlight`
    ```
    0
    ```
    sets keyboard to "full"
  * `$ echo 1 | sudo tee /sys/devices/platform/huawei-wmi/kbdlight`
    ```
    1
    tee: /sys/devices/platform/huawei-wmi/kbdlight: No such device
    ```
    does not change the keyboard lighting setting
  * `$ echo 2 | sudo tee /sys/devices/platform/huawei-wmi/kbdlight`
    ```
    2
    tee: /sys/devices/platform/huawei-wmi/kbdlight: No such device
    ```
    does not change the keyboard lighting setting
  * `$ echo 3 | sudo tee /sys/devices/platform/huawei-wmi/kbdlight`
    ```
    3
    tee: /sys/devices/platform/huawei-wmi/kbdlight: Invalid argument
    ```
    does not change the keyboard lighting setting, also true for all the other values I've tried

german-rios-gonzalez, 2020-10-21:
Hi @aymanbagabas, I don't know if I do it all ok, but the keyboard can cycle through the 3 levels of backlight, from off, to medium, to maximum, and the timeout it's off 💃, and there isn't kernel panics signals yet, I'm going to work all day and be right back if anything went wrong

aymanbagabas, 2020-11-15:
I finally was able to run this on my machine (MACH-WX9) and I see a couple of differences.

```
    Method (GLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100){})
        Local0 = Arg0
        CreateByteField (BUFF, Zero, STAT)
        CreateByteField (BUFF, One, AMLL)
        CreateByteField (BUFF, 0x02, KBLL)
        AMLL = 0xFF
        KBLL = \_SB.PCI0.LPCB.EC0.RRAM (0x0C4F)
        STAT = Zero
        Return (BUFF) /* \GLIV.BUFF */
    }

    Method (SLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100){})
        Local0 = Arg0
        CreateByteField (Arg0, 0x02, SOCP)
        CreateByteField (BUFF, Zero, STAT)
        \_SB.PCI0.LPCB.EC0.WRAM (0x0C4F, SOCP)
        STAT = Zero
        Return (BUFF) /* \SLIV.BUFF */
    }
```

The return buf always returns `buf[1] = 0xff` and `buf[2]` has one of the following values `0x01` off, `0x02` level 1, and `0x04` level 2 which is different from what you have. It seems like older their older models have more quirks and edge cases. I will see what I can do.

aymanbagabas, 2020-11-16:
@nekr0z could you please provide a `dmidecode` output? As this project keeps growing, it's becoming hard to keep track of which device is which. So I'm thinking of building a database of Huawei devices information in the wiki section of this repository.

`dmidecode` is used to identify platform devices in the kernel and the driver uses those to apply quirks to different devices. From what I've noticed, new devices report kbdlight button presses and they're handled through WMI as opposed to old ones which are handled in the firmware.

aymanbagabas, 2020-11-16:
@german-rios-gonzalez your device doesn't seem to have `\SKBT` and `\GKBT` introduced in https://github.com/aymanbagabas/Huawei-WMI/pull/44

aymanbagabas, 2020-11-16:
@nekr0z @german-rios-gonzalez I've updated https://github.com/aymanbagabas/Huawei-WMI/tree/feat/kbdlight to fix some issues.

@nekr0z I've noticed that whenever I change `kbdlight` on my machine, the change does not take effect immediately like if it was on level 2 and I set it to off, it takes some time for the change to apply. This might be the case for you as well. However, I think this doesn't matter much because our devices already handle kbdlight just fine, though, it would be nice to control this from the software.

german-rios-gonzalez, 2020-11-16:
Hi, @aymanbagabas so far it's working without issues, I didn't notice any drawbacks, the time out it's off and the keyboard backlight it's always on

sermart1234, 2020-11-16:
**Offset 0x18 - KBBH
Offset 0x19 - KBBL**

**0x32 = 50 = 5 secons**
[image]

**0x012C = 300 = 30 seconds**
[image]

nekr0z, 2020-11-16:
> @nekr0z could you please provide a `dmidecode` output?

[dmidecode.txt](https://github.com/aymanbagabas/Huawei-WMI/files/5545248/dmidecode.txt)

nekr0z, 2020-11-16:
Hmm. I've upgraded my BIOS to 1.28, and I can no longer get `kbdlight` interface to appear...

sermart1234, 2020-11-16:
@nekr0z, acpidump, please `acpixtract -a acpidump.out`
On Magicbook Pro 2020 (BIOS 1.03) doesn't work `kbdlight`.
You can try get timeout:
```
echo 0x00001206 | sudo tee /sys/kernel/debug/huawei-wmi/arg
sudo cat /sys/kernel/debug/huawei-wmi/call
```
Set timeout:
```
echo 0x[value in seconds]1106 | sudo tee /sys/kernel/debug/huawei-wmi/arg
sudo cat /sys/kernel/debug/huawei-wmi/call

echo 0x00001106 | sudo tee /sys/kernel/debug/huawei-wmi/arg //disable timeout
```

aymanbagabas, 2020-11-16:
The keyboard backlight would need more quirks...

Here are the two functions `\SLIV` and `\GLIV` responsible for set/get the keyboard backlight callable at `0x00000702` and  `0x00000602`.

<details>
<summary>Magicbook Pro 2020 (HLYL-WXX9)</summary>

    Method (GLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100) {})
        Name (CONT, Zero)
        Local0 = Arg0
        CreateByteField (BUFF, Zero, STAT)
        CreateByteField (BUFF, One, SURV)
        CreateByteField (BUFF, 0x02, KBBL)
        CONT = Zero
        Acquire (\_SB.PCI0.LPC0.EC0.ECMT, 0x03E8)
        \_SB.PCI0.LPC0.EC0.ECM3 (0xC4)
        While (((CONT != 0x0A) && ((\_SB.PCI0.LPC0.EC0.ECXC & One) == Zero)))
        {
            CONT += One
            Sleep (0x32)
        }

        If ((\_SB.PCI0.LPC0.EC0.ECXD == 0x04))
        {
            KBBL = Zero
            STAT = Zero
        }
        ElseIf ((\_SB.PCI0.LPC0.EC0.ECXD == 0x08))
        {
            KBBL = 0x32
            STAT = Zero
        }
        ElseIf ((\_SB.PCI0.LPC0.EC0.ECXD == 0x10))
        {
            KBBL = 0x64
            STAT = Zero
        }
        Else
        {
            STAT = One
        }

        Release (\_SB.PCI0.LPC0.EC0.ECMT)
        Return (BUFF) /* \GLIV.BUFF */
    }

    Method (SLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100) {})
        Local0 = Arg0
        CreateByteField (Arg0, 0x02, KBBV)
        CreateByteField (BUFF, Zero, STAT)
        Acquire (\_SB.PCI0.LPC0.EC0.ECMT, 0x03E8)
        If ((KBBV == Zero))
        {
            \_SB.PCI0.LPC0.EC0.ECM3 (0xC5)
            \_SB.PCI0.LPC0.EC0.ECD3 (0x04)
            STAT = Zero
        }
        ElseIf ((KBBV == 0x02))
        {
            \_SB.PCI0.LPC0.EC0.ECM3 (0xC5)
            \_SB.PCI0.LPC0.EC0.ECD3 (0x08)
            STAT = Zero
        }
        ElseIf ((KBBV == 0x04))
        {
            \_SB.PCI0.LPC0.EC0.ECM3 (0xC5)
            \_SB.PCI0.LPC0.EC0.ECD3 (0x10)
            STAT = Zero
        }
        Else
        {
            STAT = One
        }

        Release (\_SB.PCI0.LPC0.EC0.ECMT)
        Return (BUFF) /* \SLIV.BUFF */
    }

</details>

<details>
<summary>Matebook 13 2019 (WRT-WX9)</summary>

    Method (GLIV, 1, Serialized)
    {
        Name (BUFF, Buffer (0x0100){})
        CreateByteField (BUFF, Zero, STAT)
        CreateByteField (BUFF, One, ENLV)
        CreateByteField (BUFF, 0x02, KBLV)
        ENLV = \_SB.PCI0.LPCB.H_EC.ALSD.SWAP (\_SB.PCI0.LPCB.H_EC.CADA)
        Local0 = \_SB.PCI0.LPCB.H_EC.RDER (0x0131)
        Switch (Local0)
        {
            Case (0x04)
            {
                KBLV = Zero
            }
            Case (0x08)
            {
                KBLV = 0x32
            }
            Case (0x10)
            {
                KBLV = 0x64
            }
            Default
            {
                KBLV = Zero
            }

        }

        STAT = Zero
        Return (BUFF) /* \GLIV.BUFF */
    }

    Method (SLIV, 1, Serialized)
    {
        Name (BUFF, Buffer (0x0100){})
        CreateByteField (Arg0, 0x02, SKBL)
        CreateByteField (BUFF, Zero, STAT)
        STAT = Zero
        Switch (ToInteger (SKBL))
        {
            Case (Zero)
            {
                \_SB.PCI0.LPCB.H_EC.WTER (0x0131, 0x04)
            }
            Case (0x02)
            {
                \_SB.PCI0.LPCB.H_EC.WTER (0x0131, 0x08)
            }
            Case (0x04)
            {
                \_SB.PCI0.LPCB.H_EC.WTER (0x0131, 0x10)
            }
            Default
            {
                STAT = One
            }

        }

        Return (BUFF) /* \SLIV.BUFF */
    }

</details>

<details>
<summary>Matebook D14 2020 (NBL-WAQ9R)</summary>
    Method (GLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100) {})
        Name (CONT, Zero)
        Local0 = Arg0
        CreateByteField (BUFF, Zero, STAT)
        CreateByteField (BUFF, One, SURV)
        CreateByteField (BUFF, 0x02, KBBL)
        CONT = Zero
        Acquire (\_SB.PCI0.LPC0.EC0.ECMT, 0x03E8)
        \_SB.PCI0.LPC0.EC0.ECM3 (0xC4)
        While (((CONT != 0x0A) && ((\_SB.PCI0.LPC0.EC0.ECXC & One) == Zero)))
        {
            CONT += One
            Sleep (0x32)
        }

        KBBL = \_SB.PCI0.LPC0.EC0.ECXD
        Release (\_SB.PCI0.LPC0.EC0.ECMT)
        STAT = Zero
        Return (BUFF) /* \GLIV.BUFF */
    }

    Method (SLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100) {})
        Local0 = Arg0
        CreateByteField (Arg0, 0x02, KBBV)
        CreateByteField (BUFF, Zero, STAT)
        Acquire (\_SB.PCI0.LPC0.EC0.ECMT, 0x03E8)
        If ((((KBBV == 0x04) || (KBBV == 0x08)) || (KBBV == 0x10)))
        {
            \_SB.PCI0.LPC0.EC0.ECM3 (0xC5)
            \_SB.PCI0.LPC0.EC0.ECD3 (KBBV)
            STAT = Zero
        }
        Else
        {
            STAT = One
        }

        Release (\_SB.PCI0.LPC0.EC0.ECMT)
        Return (BUFF) /* \SLIV.BUFF */
    }

</details>

<details>
<summary>Matebook X Pro 2018 (MACH-WX9)</summary>

    Method (GLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100){})
        Local0 = Arg0
        CreateByteField (BUFF, Zero, STAT)
        CreateByteField (BUFF, One, AMLL)
        CreateByteField (BUFF, 0x02, KBLL)
        AMLL = 0xFF
        KBLL = \_SB.PCI0.LPCB.EC0.RRAM (0x0C4F)
        STAT = Zero
        Return (BUFF) /* \GLIV.BUFF */
    }

    Method (SLIV, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100){})
        Local0 = Arg0
        CreateByteField (Arg0, 0x02, SOCP)
        CreateByteField (BUFF, Zero, STAT)
        \_SB.PCI0.LPCB.EC0.WRAM (0x0C4F, SOCP)
        STAT = Zero
        Return (BUFF) /* \SLIV.BUFF */
    }

</details>

sermart1234, 2020-11-16:
@aymanbagabas
> Here are the two functions `\SLIV` and `\GLIV` responsible for set/get the keyboard backlight callable at `0x00000702` and `0x00000602`.
On Magicbook Pro 2020 `\SLIV` does not control the backlight

aymanbagabas, 2020-11-16:
> On Magicbook Pro 2020 `\SLIV` does not control the backlight

How do you control the backlight? On the MACH-WX9, the keyboard backlight button works out of the box. However, using these two functions, I am able to get/set the keyboard backlight, though setting it takes some time to apply which is weird.

nekr0z, 2020-11-17:
> @nekr0z, acpidump, please `acpixtract -a acpidump.out`

You want me to publish all the 39 resulting `.dat`s?

> You can try get timeout:

On WRT-WX9 this always returns a bunch of `0x00`s with a single `0x01`in the first position of the second array. The set method you've mentioned does nothing.

n3vu0r, 2020-11-17:
No idea if my MateBook D14 AMD 2018 (KPL-W0X) can control timeouts but here are [dmidecode.txt] and [acpidump.txt].

[dmidecode.txt]: https://github.com/aymanbagabas/Huawei-WMI/files/5551922/dmidecode.txt
[acpidump.txt]: https://github.com/aymanbagabas/Huawei-WMI/files/5551921/acpidump.txt

sermart1234, 2020-11-17:
> You want me to publish all the 39 resulting `.dat`s?

Тo. I copied the wrong command. Sorry/

sermart1234, 2020-11-17:
@n3vu0r, your device doesn't have `\SKBT` and `\GKBT`. You can try upgrade BIOS.

sermart1234, 2020-11-17:
> How do you control the backlight? On the MACH-WX9, the keyboard backlight button works out of the box. However, using these two functions, I am able to get/set the keyboard backlight, though setting it takes some time to apply which is weird.

On HLYL-WXX9 the keyboard backlight button works also. 
`GLIV` always returns `STAT = 1`.
`SLIV` just doesn't work.  
`echo 0x00040702 | tee /sys/kernel/debug/huawei-wmi/arg`, `echo 0x00020702 | tee /sys/kernel/debug/huawei-wmi/arg`, `echo 0x00040702 | tee /sys/kernel/debug/huawei-wmi/arg` doesn't change the `GLIV` return

sermart1234, 2020-11-17:
@aymanbagabas, I think, on your device you can read/write timeout value in EC memory.

I can read level backlight on my laptop. `od -Ax -t x1 /sys/kernel/debug/ec/ec0/io`
Turn off:
0000f0 fc 07 00 00 00 00 00 00 03 03 00 00 **0**0 00 00 00
Level 1:
0000f0 fc 07 00 00 00 00 00 00 03 03 00 00 **4**0 00 00 00
Level 2:
0000f0 fc 07 00 00 00 00 00 00 03 03 00 00 **8**0 00 00 00
Also FN led:
0000f0 fc 07 00 00 00 00 00 00 03 03 00 00 0**2** 00 00 00

nekr0z, 2020-11-17:
> > You want me to publish all the 39 resulting `.dat`s?
> 
> No. I copied the wrong command. Sorry. `acpidump > acpidump.out`

[acpidump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/5553486/acpidump.txt)

sermart1234, 2020-11-17:
@aymanbagabas. Also

`GVER` return BIOS version 1.03
<details>
  <summary>GVER | 0x00000101</summary>
[{0x00,0x00,0x00,0x00},{0x00,0x01,0x00,0x10,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]
</details>
<details>
  <summary>GTSI | 0x00000102</summary>
[{0x00,0x00,0x00,0x00},{0x00,0xa1,0x49,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]
</details>

`GTMP` return 0x26 or 0x27

<details>
  <summary>GTMP | 0x00000202</summary>
[{0x00,0x00,0x00,0x00},{0x00,0x00,0x26,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]
</details>

GPSI always return 5. Because PSIT = 0x05

<details>
  <summary>GPSI | 0x00000402</summary>
[{0x00,0x00,0x00,0x00},{0x00,0x05,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]
</details>

aymanbagabas, 2020-11-17:
> On HLYL-WXX9 the keyboard backlight button works also.
> `GLIV` always returns `STAT = 1`.
> `SLIV` just doesn't work.
> `echo 0x00040702 | tee /sys/kernel/debug/huawei-wmi/arg`, `echo 0x00020702 | tee [...]

You're getting `0x01` because you have to run each call twice, look at the implementation of your device's wmi method. This behavior is the same on other devices. Notice `WMIC` in this code.

```
            Method (WMAA, 3, Serialized)
            {
                Acquire (MUTW, 0xFFFF)
                WMIC++
                Name (WMRP, Package (0x02)
                {
                    Buffer (0x04) {}, 
                    Buffer (0x0100) {}
                })
                Local0 = Arg2
                CreateByteField (Local0, Zero, MFID)
                CreateByteField (Local0, One, SFID)
                Local1 = DerefOf (WMRP [One])
                CreateByteField (Local1, Zero, STAT)
                STAT = One
                If ((WMIC == 0x02))
                {
                    WMIC = Zero
                    Switch (MFID)
                    {
                        Case (One)
                        {
                            Switch (SFID)
                            {
                                Case (On
...
```

aymanbagabas, 2020-11-17:
> @aymanbagabas. Also
> 
> `GVER` return BIOS version 1.03
> 
> GVER | 0x00000101
> GTSI | 0x00000102
> `GTMP` return 0x26 or 0x27
> 
> GTMP | 0x00000202
> `GPSI` always return 5. Because PSIT = 0x05
> 
> GPSI | 0x00000402

I'm not sure if those are 'usefull' or not. The only thing I can think of is to check available features based on `GVER`.

sermart1234, 2020-11-17:
@aymanbagabas
0x00000F04 STUB
0x00000E04 GTUB

The command sequence returns connected power adapter. [{0x00,0x00,0x00,0x00},{0x00,0x0**1**,0x00,
```
echo 0x00010F04 | tee /sys/kernel/debug/huawei-wmi/arg
cat /sys/kernel/debug/huawei-wmi/call
echo 0x00000E04 | tee /sys/kernel/debug/huawei-wmi/arg
cat /sys/kernel/debug/huawei-wmi/call
```

sermart1234, 2020-11-19:
@aymanbagabas
[Honor Magicbook **2019**.zip](https://github.com/aymanbagabas/Huawei-WMI/files/5568211/Honor.Magicbook.2019.zip)

aymanbagabas, 2020-11-19:
@sermart1234 Could you test `\SLIV` and `\GLIV` more? It wouldn't make sense for these functions to behave differently on other devices.

sermart1234, 2020-11-19:
@aymanbagabas
> You're getting `0x01` because you have to run each call twice, look at the implementation of your device's wmi method. This behavior is the same on other devices. Notice `WMIC` in this code.

oh. I've already tried calling everything twice. Could you give an example of a sequence of commands?

aymanbagabas, 2020-11-19:
@sermart1234

```
echo 0x00000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
sudo cat /sys/kernel/debug/huawei-wmi/call
sudo cat /sys/kernel/debug/huawei-wmi/call
echo 0x00020702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
sudo cat /sys/kernel/debug/huawei-wmi/call
sudo cat /sys/kernel/debug/huawei-wmi/call
echo 0x00000702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
sudo cat /sys/kernel/debug/huawei-wmi/call
sudo cat /sys/kernel/debug/huawei-wmi/call
```

sermart1234, 2020-11-20:
@aymanbagabas, oh, I always did
<details>
  <summary>But...</summary>
[root@LAPTOP-LP7S54VC ~]# echo 0x00000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg<br>
0x00000602<br>
[root@LAPTOP-LP7S54VC ~]# sudo cat /sys/kernel/debug/huawei-wmi/call<br>
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]<br>[root@LAPTOP-LP7S54VC ~]# sudo cat /sys/kernel/debug/huawei-wmi/call<br>
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}]
</details>

aymanbagabas, 2020-11-25:
@sermart1234 could you try this in the same order make sure the backlight is on

```
echo 0x00000702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
sudo cat /sys/kernel/debug/huawei-wmi/call
sudo cat /sys/kernel/debug/huawei-wmi/call
echo 0x00000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
sudo cat /sys/kernel/debug/huawei-wmi/call
sudo cat /sys/kernel/debug/huawei-wmi/call
```

keltkelt, 2020-11-27:
Hello,

ive bought a Matebook 13 2020 recently and got the exact same problem with 15 second keyboard backlight timing.
Can anyone please tell me how to disable the timing on a windows machine? 

I only registered to ask this on this Forum. :-)

sermart1234, 2020-11-28:
@aymanbagabas dosn't work
<details>
  <summary>output</summary>
[root@LAPTOP-LP7S54VC ~]# sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}][root@LAPTOP-LP7S54VC ~]# sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}][root@LAPTOP-LP7S54VC ~]# echo 0x00000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
0x00000602
[root@LAPTOP-LP7S54VC ~]# sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}][root@LAPTOP-LP7S54VC ~]# sudo cat /sys/kernel/debug/huawei-wmi/call
[{0x00,0x00,0x00,0x00},{0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00}][root@LAPTOP-LP7S54VC ~]#[root@LAPTOP-LP7S54VC ~]# 
</details>

sermart1234, 2020-11-28:
@aymanbagabas also Honor Magicbook **2019**
[Honor Magicbook **2019** BIOS 1.12.zip](https://github.com/aymanbagabas/Huawei-WMI/files/5610684/Honor.Magicbook.2019.BIOS.1.12.zip)

aymanbagabas, 2020-11-30:
> ```
> echo 0x00000702 | sudo tee /sys/kernel/debug/huawei-wmi/arg
> sudo cat /sys/kernel/debug/huawei-wmi/call
> sudo cat /sys/kernel/debug/huawei-wmi/call
> echo 0x00000602 | sudo tee /sys/kernel/debug/huawei-wmi/arg
> sudo cat /sys/kernel/debug/huawei-wmi/call
> sudo cat /sys/kernel/debug/huawei-wmi/call
> ```

@sermart1234 Could you test this on a Honor Magicbook 2019?

aymanbagabas, 2020-11-30:
> Hello,
> 
> ive bought a Matebook 13 2020 recently and got the exact same problem with 15 second keyboard backlight timing.
> Can anyone please tell me how to disable the timing on a windows machine?
> 
> I only registered to ask this on this Forum. :-)

@keltkelt I need to peek through your laptop's ACPI tables. Could you please provide a `acpidump` and a `dmidecode` outputs?
```
sudo acpidump > acpidump.out
dmidecode > dmidecode.txt
```

[28 more replies]