aymanbagabas/Huawei-WMI #40: Huawei Matebook D15 from 2020

krysopath, 2020-09-06:
**battery thresholds and external microphone**

I recently purchased Matebook D15 and am really glad you pioneered support. Most features do work right out of the box. I am missing control over battery thresholds, and an issue with my external microphone audio jack.

**To Reproduce battery**
Steps to reproduce the behavior:
1. run `5.4.0-42-generic` in ubuntu 20.04
2. Try to set battery charging thresholds in `/sys/class/power_supply`
3. device does not expose that interface via kernel sysfs

**Expected behavior**

I want to set charging thresholds via sysfs interface to have less wear on the battery.

**To Reproduce audio**
Steps to reproduce the behavior:
1. Buy above machine and run `5.4.0-42-generic` in ubuntu
2. Plug a external headset
3. pulseaudio will continue to record with internal microphone, as no external input is recognized

**Expected behavior**
I want to use external input source when plugging in external headset via that jack.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop:**
`Linux porta 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux` with ubuntu focal 20.04

```
$ lsmod |grep wmi
snd_rawmidi            36864  1 snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
wmi_bmof               16384  0
snd                    90112  27 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi
huawei_wmi             16384  0
ledtrig_audio          16384  3 snd_hda_codec_generic,huawei_wmi,snd_hda_codec_realtek
sparse_keymap          16384  1 huawei_wmi
wmi                    32768  2 huawei_wmi,wmi_bmof

```

[dmesg.log](https://github.com/aymanbagabas/Huawei-WMI/files/5179515/dmesg.log)

alsa-info: http://alsa-project.org/db/?f=6e2af66bedb483f0f613cae40382c90a4d590efa

**Additional context**
I can help and will dedicate time if this behavior can be fixed via a contribution. Give me directions for more specific debug output.

aymanbagabas, 2020-09-06:
Could you please provide an acpi dump? Refer to https://01.org/linux-acpi/utilities for more information.

krysopath, 2020-09-07:
Installed `acpica-tools` via apt.

```
$ sudo acpidump > acpi-tables.out
```

gzipped to satisfy github contraints.

[acpi-tables.out.gz](https://github.com/aymanbagabas/Huawei-WMI/files/5183406/acpi-tables.out.gz)

krysopath, 2020-09-07:
I also updated  [lspci.txt](https://github.com/aymanbagabas/Huawei-WMI/files/5183882/lspci.txt) by running it with uid 0. Now includes `Capabilities` for each device.

krysopath, 2020-09-22:
Can I help by further remote command execution or via learning how acpi actually does work?

aymanbagabas, 2020-11-17:
Can you upload the output of `dmidecode > dmidecode.txt`

krysopath, 2020-11-17:
[dmidecode.txt.gz](https://github.com/aymanbagabas/Huawei-WMI/files/5554674/dmidecode.txt.gz)

aymanbagabas, 2021-09-01:
Could you please try building the module from the source and report what files get exposed under `/sys/devices/platform/huawei-wmi/`

sermart1234, 2021-11-27:
I try test on Huawei Matebook D15 with 3700U

sermart1234, 2022-01-31:
@krysopath please test this https://github.com/aymanbagabas/Huawei-WMI/issues/55#issuecomment-1025651920

sermart1234, 2022-02-03:
@krysopath if you set "0 0" or "0 100" then it turn on smart charge
https://github.com/aymanbagabas/Huawei-WMI/issues/55#issuecomment-1028953673

krysopath, 2022-03-06:
@sermart1234  @aymanbagabas 

Still not working is pulseaudios recording device change, when a microphone jack is plugged into the laptop, but

was able to verify smart charge device node working. Thank you all. <3

sermart1234, 2022-05-11:
@krysopath test please
https://github.com/aymanbagabas/Huawei-WMI/pull/64

closed 2022-06-03 by sermart1234