aymanbagabas/Huawei-WMI #46: Help with understanding the battery thresholds feature

ldan93, 2021-01-01:
Hello,

I'm trying to implement the battery thresholds feature under MacOS (hackintosh) on the Matebook X Pro 2018 (MACH-WX9). I'm trying to do this only by using ACPI calls.

From reading your code and the DSDT of this machine, I understand that the \GBTT and \SBTT ACPI methods are responsible for getting and setting min/max battery thresholds : 

[image]

I've used this line of the SBTT method to try to set the threshold to 60/75 (3C/4B) : 
`\_SB.PCI0.LPCB.EC0.ECXT (0xC7, One, 0x3C, 0x4B, Zero, Zero)`

From both MacOS and Linux (via acpi_call), this ACPI call doesn't seem to be effective : the battery is still charging between 60 and 75%. So I guess this isn't enough / there is something I don't really understand.

Do you have any idea what would be the ACPI call(s) equivalent to the following command ? 
echo "60 75" | sudo tee /sys/devices/platform/huawei-wmi/charge_control_thresholds

Thank you very much !

aymanbagabas, 2021-01-04:
Hi @ldan93,

The line you used to set the threshold _should_ theoretically be effective. You can verify that by looking at EC memory, I use RWEverything in Windows and `modprobe ec_sys && watch -n1 hexdump -Cv /sys/kernel/debug/ec/ec0/io` in Linux. Also if you want to debug ACPI, I'd recommend using `acpiexec` to run a sandboxed ACPI environment where you can test these functions.

Now these two functions expect one argument in little-endian, the first two bytes are reserved for HWMI (Huawei WMI) interface and the other two are the arguments that these functions take and operate on. So if you want to set the thresholds to 60/70 you would call `\SBTT 0x4b3c0000`. If you were using WMI, the first two bytes would be `0x1003` because that's the HWMI identifier that maps to `\SBTT` and the WMI would call `\SBTT 0x4b3c1003` in that case so these first two bytes really don't matter if you're using ACPI directly instead of WMI.

I found this cool project https://github.com/the-darkvoid/macOS-IOElectrify that implements ACPI WMI for MacOS I thought it would be very helpful to the Hackintosh community. It would be awesome to fork/extend that to support more devices. Look at https://github.com/torvalds/linux/blob/master/drivers/platform/x86/wmi.c if you're more interested in this topic.

ldan93, 2021-01-04:
Thank you very much for all your useful explanations and advice !
Calling \SBTT with the syntax you explained works under MacOS. I will have a thorough look at the projects you mentioned.

closed 2021-01-04 by ldan93