aymanbagabas/Huawei-WMI #47: Some of the function hotkeys not work in TTYs

pur2evil, 2021-01-21:
For the record, all of the hotkeys, they do work perfectly in Desktop Environments.
However, some function hotkeys like Volume Up/Down, Mute… they just don't work in TTYs.
I'm not sure if this is a bug. Or the driver is for Desktop only?
Help needed.

aymanbagabas, 2021-01-21:
Most of the hotkeys work through firing ACPI events, a desktop environment like GNOME or KDE implements their own ACPI handler. You could try [acpid](https://wiki.archlinux.org/index.php/acpid) if you want to have your own handler that _should_ work outside of a desktop environment.

pur2evil, 2021-01-22:
Yea, acpid. That was my Plan B.
I will try it when I got time.
Thanks man. You are awesome.

closed 2021-09-01 by aymanbagabas