aymanbagabas/Huawei-WMI #49: Huawei Matebook D16 micmute and battery protection

EnginAcikgoz, 2021-05-13:
Hi, microphone led doesn't turn on and off automatically.
And battery protection feature doesn't work.
If you wish, I can give extra information about the system.

aymanbagabas, 2021-05-13:
Please follow the bug report template and provide more information https://github.com/aymanbagabas/Huawei-WMI/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

EnginAcikgoz, 2021-05-13:
**acpidump,  dmidecode and dmesg

 outputs**
[acpidump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/6472713/acpidump.txt)
[dmidecode.txt](https://github.com/aymanbagabas/Huawei-WMI/files/6472822/dmidecode.txt)
[dmesg-out.txt](https://github.com/aymanbagabas/Huawei-WMI/files/6472788/dmesg-out.txt)

**uname -a**
Linux unknown 5.12.1-2-MANJARO #1 SMP PREEMPT Sun May 2 13:56:40 UTC 2021 x86_64 GNU/Linux

**lsmod | grep wmi**
wmi_bmof               16384  0
huawei_wmi             24576  0
sparse_keymap          16384  1 huawei_wmi
ledtrig_audio          16384  2 snd_hda_codec_generic,huawei_wmi
wmi                    36864  2 huawei_wmi,wmi_bmof

**sudo evtest**
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Lid Switch
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Video Bus
/dev/input/event5:	Huawei WMI hotkeys
/dev/input/event6:	GXTP7863:00 27C6:01E0 Mouse
/dev/input/event7:	PC Speaker
/dev/input/event8:	HD-Audio Generic HDMI/DP,pcm=3
/dev/input/event9:	HD-Audio Generic HDMI/DP,pcm=7
/dev/input/event10:	HD-Audio Generic HDMI/DP,pcm=8
/dev/input/event11:	HD-Audio Generic Headphone
/dev/input/event12:	GXTP7863:00 27C6:01E0 Touchpad
/dev/input/event13:	HD Camera: HD Camera
/dev/input/event14:	Logitech Wireless Mouse
/dev/input/event15:	Soundcore Liberty 2 Pro (AVRCP)
Select the device event number [0-15]: 5
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "Huawei WMI hotkeys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 171 (KEY_CONFIG)
    Event code 190 (KEY_F20)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
    Event code 238 (KEY_WLAN)
    Event code 240 (KEY_UNKNOWN)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)

Thank you.

EnginAcikgoz, 2021-05-20:
Hello again, i still waiting. Do you need any information?

EnginAcikgoz, 2021-05-23:
[image]
i tried run a script/program with fn+f7 keys and shortcut looks as "AudioMicMute" and shortcut doesn't working. Shortcut does only open and close microphone. Other commands does not working.

closed 2021-09-01 by aymanbagabas