aymanbagabas/Huawei-WMI #52: Huawei D16 battery threshold start point does not work

m1khal3v, 2021-09-29:
**Describe the bug**
The start point is set at 70% and the stop point is at 90%. The battery charges up to 89% and holds this value 

**Expected behavior**
When the stop point is reached, the charging should be switched off until the start point is reached. 

Kernel: 5.14.8-051408-generic
Distro: Ubuntu 20.04.3 LTS
Loaded WMI modules:
```
snd_rawmidi            40960  1 snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
huawei_wmi             20480  0
ledtrig_audio          16384  2 snd_hda_codec_generic,huawei_wmi
sparse_keymap          16384  1 huawei_wmi
wmi_bmof               16384  0
snd                    94208  22 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm,snd_rawmidi
wmi                    32768  2 huawei_wmi,wmi_bmof

```
[acpidump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7252788/acpidump.txt) | [dmidecode.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7252793/dmidecode.txt) | [dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7252818/dmesg.txt)

m1khal3v, 2021-09-29:
Now I tried to install version 3.4 using dkms autoinstall, but it didn't change anything

m1khal3v, 2021-09-29:
Also I updated the BIOS to the latest version and it didn't help. I attach dumps after manipulations 
[acpidump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7253723/acpidump.txt) | [dmidecode.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7253728/dmidecode.txt) | [dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7253726/dmesg.txt)

aymanbagabas, 2021-09-30:
This is the expected behavior according to Huawei's website https://consumer.huawei.com/en/support/content/en-us00727970/

closed 2021-09-30 by aymanbagabas

m1khal3v, 2021-09-30:
> This is the expected behavior according to Huawei's website https://consumer.huawei.com/en/support/content/en-us00727970/

Oh, thanks for info. It's amazing