aymanbagabas/Huawei-WMI #53: Huawei MateBook D16 microphone led does not work

m1khal3v, 2021-09-29:
**Describe the bug**
LED on the fn key does not light up when microphone is turned off

**Expected behavior**
When the microphone is turned off, the LED on the fn key should light up

Kernel: 5.14.8-051408-generic
Distro: Ubuntu 20.04.3 LTS
Loaded WMI modules:
```
snd_rawmidi            40960  1 snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
huawei_wmi             20480  0
ledtrig_audio          16384  2 snd_hda_codec_generic,huawei_wmi
sparse_keymap          16384  1 huawei_wmi
wmi_bmof               16384  0
snd                    94208  22 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm,snd_rawmidi
wmi                    32768  2 huawei_wmi,wmi_bmof

```
[acpidump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7252788/acpidump.txt) | [dmidecode.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7252793/dmidecode.txt) | [dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7252818/dmesg.txt)

m1khal3v, 2021-09-29:
Now I tried to install version 3.4 using dkms autoinstall, but it didn't change anything

m1khal3v, 2021-09-29:
Also I updated the BIOS to the latest version and it didn't help. I attach dumps after manipulations 
[acpidump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7253723/acpidump.txt) | [dmidecode.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7253728/dmidecode.txt) | [dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7253726/dmesg.txt)

EnginAcikgoz, 2021-09-29:
I have the same problem on the same device.
Also the charge limit doesn't work either.

m1khal3v, 2021-09-30:
@EnginAcikgoz hello, look battery issue here https://github.com/aymanbagabas/Huawei-WMI/issues/52

aymanbagabas, 2021-09-30:
Does the led light up when you do
```sh
echo 1 | sudo tee /sys/devices/platform/huawei-wmi/leds/platform::micmute/brightness
```

EnginAcikgoz, 2021-09-30:
> Does the led light up when you do
> 
> ```shell
> echo 1 | sudo tee /sys/devices/platform/huawei-wmi/leds/platform::micmute/brightness
> ```

yes, the code is working and led light up.

aymanbagabas, 2021-09-30:
@EnginAcikgoz this means there is nothing wrong with `huawei-wmi`. What desktop environment are you using? When I use GNOME or KDE, this simply works fine because these DEs know how to talk to the micmute led interface. If you're not using a DE that recognizes XF86 keys, then you won't get the led to work when you press the key.

The way this works is, when you press the key, the driver fires an event with the code [`0x287`](https://github.com/aymanbagabas/Huawei-WMI/blob/master/huawei-wmi.c#L81) which then gets translated to [`F20`](https://github.com/systemd/systemd/blob/main/hwdb.d/60-keyboard.hwdb#L716) in SystemD. This is because `XF86AudioMicMute` keycode is beyond what X.Org supports (or something between these lines). The DE keyboard event handler detects the event and mutes the microphone which then triggers the realtek sound [hda](https://elixir.bootlin.com/linux/v5.14.9/source/sound/pci/hda/patch_realtek.c#L8740) to set the led using the `huawei-wmi` micmute led function based on Huawei's sound PCI vendor id [`0x19e5`](https://elixir.bootlin.com/linux/v5.14.9/source/sound/pci/hda/patch_realtek.c#L8740) which you can verify when you run `alsa-info.sh`

m1khal3v, 2021-09-30:
> echo 1 | sudo tee /sys/devices/platform/huawei-wmi/leds/platform::micmute/brightness
LED lights up

> @EnginAcikgoz this means there is nothing wrong with `huawei-wmi`. What desktop environment are you using? When I use GNOME or KDE, this simply works fine because these DEs know how to talk to the [...]
I use Gnome 3.36.9 and xorg-server 2:1.20.11-1ubuntu1~20.04.2. Could you tell us how to find clues why this does not work on our hardware?

aymanbagabas, 2021-09-30:
@m1khal3v could you upload the output of `alsa-info.sh`? Also what happens if you mute the mic from GNOME sound settings?

m1khal3v, 2021-09-30:
@aymanbagabas http://alsa-project.org/db/?f=bdffc32fb0826fbe78792752b0fade8b4c4a43f7

aymanbagabas, 2021-09-30:
k now I see the problem.

It seems like new Huawei devices have a different subsystem id compared to the old ones. [Linux-Hardware.org](http://linux-hardware.org/?view=search&subvendorid=1e83&typeid=sound#list) shows devices with this subsystem vendor id `HVY-WXX9`, `KLVC-WXX9`, `KLVL-WXX9`, `WRTB-WXX9`, `MACHC-WAX9`, and `WRTB-WXX9`.

```
!!PCI Soundcards installed in the system
!!--------------------------------------

03:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Device [1002:1637]
	Subsystem: Device [1e83:3e30]
03:00.6 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) HD Audio Controller [1022:15e3]
	Subsystem: Device [1e83:3e30]
```

m1khal3v, 2021-10-01:
@aymanbagabas try to hook fn+f7
```
sudo evtest /dev/input/event5                                                                                                                                                                   11:40:43 
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "Huawei WMI hotkeys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 171 (KEY_CONFIG)
    Event code 190 (KEY_F20)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
    Event code 238 (KEY_WLAN)
    Event code 240 (KEY_UNKNOWN)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1633077645.416658, type 4 (EV_MSC), code 4 (MSC_SCAN), value 287
Event: time 1633077645.416658, type 1 (EV_KEY), code 190 (KEY_F20), value 1
Event: time 1633077645.416658, -------------- SYN_REPORT ------------
Event: time 1633077645.416682, type 1 (EV_KEY), code 190 (KEY_F20), value 0
Event: time 1633077645.416682, -------------- SYN_REPORT ------------
```

aymanbagabas, 2021-10-11:
also try
```
sudo depmod -a
sudo modprobe -f snd-hda-codec-realtek
```

aymanbagabas, 2021-10-11:
does `modinfo snd-hda-codec-realtek` returns kernel 5.14?

enginacikgoz, 2021-10-12:
These are a bit complicated for me. I will try it as soon as possible though. Thank you. =)

enginacikgoz, 2021-10-13:
Sir, `zsh: no such file or directory: /boot/config-5.13.19-2-MANJARO` what i should do?
ls /boot
`amd-ucode.img  grub                                initramfs-5.13-x86_64.img  memtest86+
efi            initramfs-5.13-x86_64-fallback.img  linux513-x86_64.kver       vmlinuz-5.13-x86_64`

enginacikgoz, 2021-10-13:
I can't build it for 5.13.19-2-MANJARO version. please forgive my ignorance

aymanbagabas, 2021-11-04:
@m1khal3v could you try the steps above and report back?

enginacikgoz, 2021-11-06:
```git clone https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git
cd linux
git checkout v5.13 # or whatever matches your kernel
wget https://github.com/aymanbagabas/Huawei-WMI/files/7326045/0001-ALSA-hda-realtek-New-Huawei-vendor-ID.patch.txt
git am < 0001-ALSA-hda-realtek-New-Huawei-vendor-ID.patch.txt
zcat /proc/config.gz > .config
/usr/lib/modules/5.15.0-1-MANJARO/build/Module.symvers .
make oldconfig # accept all default values if any (press enter)
make scripts prepare modules_prepare
make -C . M=sound/pci/hda
cp sound/pci/hda/snd-hda-codec-realtek.ko /lib/modules/5.15.0-1-MANJARO/updates/
sudo depmod -a
```
modinfo snd-hda-codec-realtek
```
filename:       /lib/modules/5.15.0-1-MANJARO/updates/snd-hda-codec-realtek.ko
description:    Realtek HD-audio codec
license:        GPL
srcversion:     0EA10D76B45226A7E2E36D4
alias:          hdaudio:v10EC1220r*a01*
alias:          hdaudio:v10EC1168r*a01*
alias:          hdaudio:v10EC0B00r*a01*
alias:          hdaudio:v10EC0900r*a01*
alias:          hdaudio:v10EC0899r*a01*
alias:          hdaudio:v10EC0897r*a01*
alias:          hdaudio:v10EC0892r*a01*
alias:          hdaudio:v10EC0889r*a01*
alias:          hdaudio:v10EC0888r*a01*
alias:          hdaudio:v10EC0888r00100101a01*
alias:          hdaudio:v10EC0887r*a01*
alias:          hdaudio:v10EC0885r*a01*
alias:          hdaudio:v10EC0885r00100103a01*
alias:          hdaudio:v10EC0885r00100101a01*
alias:          hdaudio:v10EC0883r*a01*
alias:          hdaudio:v10EC0882r*a01*
alias:          hdaudio:v10EC0880r*a01*
alias:          hdaudio:v10EC0867r*a01*
alias:          hdaudio:v10EC0711r*a01*
alias:          hdaudio:v10EC0703r*a01*
alias:          hdaudio:v10EC0701r*a01*
alias:          hdaudio:v10EC0700r*a01*
alias:          hdaudio:v10EC0680r*a01*
alias:          hdaudio:v10EC0671r*a01*
alias:          hdaudio:v10EC0670r*a01*
alias:          hdaudio:v10EC0668r*a01*
alias:          hdaudio:v10EC0667r*a01*
alias:          hdaudio:v10EC0665r*a01*
alias:          hdaudio:v10EC0663r*a01*
alias:          hdaudio:v10EC0662r00100300a01*
alias:          hdaudio:v10EC0662r00100101a01*
alias:          hdaudio:v10EC0662r00100002a01*
alias:          hdaudio:v10EC0862r*a01*
alias:          hdaudio:v10EC0861r*a01*
alias:          hdaudio:v10EC0660r*a01*
alias:          hdaudio:v10EC0861r00100340a01*
alias:          hdaudio:v10EC0623r*a01*
alias:          hdaudio:v10EC0300r*a01*
alias:          hdaudio:v10EC0299r*a01*
alias:          hdaudio:v10EC0298r*a01*
alias:          hdaudio:v10EC0295r*a01*
alias:          hdaudio:v10EC0294r*a01*
alias:          hdaudio:v10EC0293r*a01*
alias:          hdaudio:v10EC0292r*a01*
alias:          hdaudio:v10EC0290r*a01*
alias:          hdaudio:v10EC0289r*a01*
alias:          hdaudio:v10EC0288r*a01*
alias:          hdaudio:v10EC0287r*a01*
alias:          hdaudio:v10EC0286r*a01*
alias:          hdaudio:v10EC0285r*a01*
alias:          hdaudio:v10EC0284r*a01*
alias:          hdaudio:v10EC0283r*a01*
alias:          hdaudio:v10EC0282r*a01*
alias:          hdaudio:v10EC0280r*a01*
alias:          hdaudio:v10EC0276r*a01*
alias:          hdaudio:v10EC0275r*a01*
alias:          hdaudio:v10EC0274r*a01*
alias:          hdaudio:v10EC0272r*a01*
alias:          hdaudio:v10EC0270r*a01*
alias:          hdaudio:v10EC0269r*a01*
alias:          hdaudio:v10EC0268r*a01*
alias:          hdaudio:v10EC0267r*a01*
alias:          hdaudio:v10EC0262r*a01*
alias:          hdaudio:v10EC0260r*a01*
alias:          hdaudio:v10EC0257r*a01*
alias:          hdaudio:v10EC0256r*a01*
alias:          hdaudio:v10EC0255r*a01*
alias:          hdaudio:v10EC0245r*a01*
alias:          hdaudio:v10EC0236r*a01*
alias:          hdaudio:v10EC0235r*a01*
alias:          hdaudio:v10EC0234r*a01*
alias:          hdaudio:v10EC0233r*a01*
alias:          hdaudio:v10EC0231r*a01*
alias:          hdaudio:v10EC0230r*a01*
alias:          hdaudio:v10EC0225r*a01*
alias:          hdaudio:v10EC0222r*a01*
alias:          hdaudio:v10EC0221r*a01*
alias:          hdaudio:v10EC0215r*a01*
depends:        snd-hda-codec,snd-hda-core,snd-hda-codec-generic,snd
retpoline:      Y
name:           snd_hda_codec_realtek
vermagic:       5.15.0-MANJARO+ SMP preempt mod_unload 
```

And led still not working..

m1khal3v, 2021-11-08:
> @m1khal3v could you try the steps above and report back?

Sorry for late answer. I get an error at the step _make oldconfig_:
```
make oldconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
/bin/sh: 1: flex: not found
make[1]: *** [scripts/Makefile.host:9: scripts/kconfig/lexer.lex.c] Error 127
make: *** [Makefile:625: oldconfig] Error 2
```

aymanbagabas, 2021-11-08:
> > @m1khal3v could you try the steps above and report back?
> 
> Sorry for late answer. I get an error at the step _make oldconfig_:

Try to install `flex`, for Ubuntu/Debian `sudo apt-get install flex`

m1khal3v, 2021-11-08:
> > > @m1khal3v could you try the steps above and report back?
> > 
> > 
> > Sorry for late answer. I get an error at the step _make oldconfig_:
> 
> Try to install `flex`, for Ubuntu/Debian `sudo apt-get install flex`

Now i have an error on step _make scripts prepare modules_prepare_:
```
scripts/sign-file.c:25:10: fatal error: openssl/opensslv.h: Not found
   25 | #include <openssl/opensslv.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.host:95: scripts/sign-file] Error 1
make: *** [Makefile:1196: scripts] Error 2

```

aymanbagabas, 2021-11-08:
> Now i have an error on step _make scripts prepare modules_prepare_:

You probably need to install `openssl` dev package. In Ubuntu/Debian, the package would be `libssl-dev`.

m1khal3v, 2021-11-08:
> > Now i have an error on step _make scripts prepare modules_prepare_:
> 
> You probably need to install `openssl` dev package. In Ubuntu/Debian, the package would be `libssl-dev`.
```
modinfo snd-hda-codec-realtek
filename:       /lib/modules/5.14.15-051415-generic/updates/snd-hda-codec-realtek.ko
...
```

The LED does not light up when i press on the keyboard or from the settings :(

m1khal3v, 2021-11-09:
@aymanbagabas Today I noticed that the sound in the right earpiece has disappeared, how can I roll back the changes?

aymanbagabas, 2021-11-09:
[0001-ALSA-hda-realtek-New-Huawei-vendor-ID.patch.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7507422/0001-ALSA-hda-realtek-New-Huawei-vendor-ID.patch.txt)

@m1khal3v @enginacikgoz I've changed the name of the entry in the patch, perhaps this makes a difference? Could you try this instead? Just make sure you undo the previous one in the linux tree `git reset HEAD^ && git checkout sound`

enginacikgoz, 2021-11-09:
> [0001-ALSA-hda-realtek-New-Huawei-vendor-ID.patch.txt](https://github.com/aymanbagabas/Huawei-WMI/files/7507422/0001-ALSA-hda-realtek-New-Huawei-vendor-ID.patch.txt)
>
> @m1khal3v @enginacikgoz I've [...]

Still not working. But i got extra information.

just to be sure i did the steps here from scratch and just changed the patch file
```
git clone https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git
cd linux
git checkout v5.15 # or whatever matches your kernel
wget https://github.com/aymanbagabas/Huawei-WMI/files/7507422/0001-ALSA-hda-realtek-New-Huawei-vendor-ID.patch.txt 
git am < 0001-ALSA-hda-realtek-New-Huawei-vendor-ID.patch.txt
zcat /proc/config.gz > .config
cp /usr/lib/modules/5.15.0-1-MANJARO/build/Module.symvers .
make oldconfig # accept all default values if any (press enter)
make scripts prepare modules_prepare
make -C . M=sound/pci/hda
cp sound/pci/hda/snd-hda-codec-realtek.ko /lib/modules/5.15.0-1-MANJARO/updates/
sudo depmod -a
```

Still not working but `make -C . M=sound/pci/hda` gives me this output
```
  CC [M]  sound/pci/hda/hda_bind.o
  CC [M]  sound/pci/hda/hda_codec.o
  CC [M]  sound/pci/hda/hda_jack.o
  CC [M]  sound/pci/hda/hda_auto_parser.o
  CC [M]  sound/pci/hda/hda_sysfs.o
  CC [M]  sound/pci/hda/hda_controller.o
  CC [M]  sound/pci/hda/hda_proc.o
  CC [M]  sound/pci/hda/hda_hwdep.o
  CC [M]  sound/pci/hda/hda_beep.o
  LD [M]  sound/pci/hda/snd-hda-codec.o
  CC [M]  sound/pci/hda/hda_generic.o
  LD [M]  sound/pci/hda/snd-hda-codec-generic.o
  CC [M]  sound/pci/hda/patch_realtek.o
  LD [M]  sound/pci/hda/snd-hda-codec-realtek.o
  CC [M]  sound/pci/hda/patch_cmedia.o
  LD [M]  sound/pci/hda/snd-hda-codec-cmedia.o
  CC [M]  sound/pci/hda/patch_analog.o
  LD [M]  sound/pci/hda/snd-hda-codec-analog.o
  CC [M]  sound/pci/hda/patch_sigmatel.o
  LD [M]  sound/pci/hda/snd-hda-codec-idt.o
  CC [M]  sound/pci/hda/patch_si3054.o
  LD [M]  sound/pci/hda/snd-hda-codec-si3054.o
  CC [M]  sound/pci/hda/patch_cirrus.o
  LD [M]  sound/pci/hda/snd-hda-codec-cirrus.o
  CC [M]  sound/pci/hda/patch_ca0110.o
  LD [M]  sound/pci/hda/snd-hda-codec-ca0110.o
  CC [M]  sound/pci/hda/patch_ca0132.o
  LD [M]  sound/pci/hda/snd-hda-codec-ca0132.o
  CC [M]  sound/pci/hda/patch_conexant.o
  LD [M]  sound/pci/hda/snd-hda-codec-conexant.o
  CC [M]  sound/pci/hda/patch_via.o
  LD [M]  sound/pci/hda/snd-hda-codec-via.o
  CC [M]  sound/pci/hda/patch_hdmi.o
  CC [M]  sound/pci/hda/hda_eld.o
  LD [M]  sound/pci/hda/snd-hda-codec-hdmi.o
  CC [M]  sound/pci/hda/hda_intel.o
  LD [M]  sound/pci/hda/snd-hda-intel.o
  MODPOST sound/pci/hda/Module.symvers
  CC [M]  sound/pci/hda/snd-hda-codec-analog.mod.o
  LD [M]  sound/pci/hda/snd-hda-codec-analog.ko
  BTF [M] sound/pci/hda/snd-hda-codec-analog.ko
Skipping BTF generation for sound/pci/hda/snd-hda-codec-analog.ko due to unavailability of vmlinux
  CC [M]  sound/pci/hda/snd-hda-codec-ca0110.mod.o
  LD [M]  sound/pci/hda/snd-hda-codec-ca0110.ko
  BTF [M] sound/pci/hda/snd-hda-codec-ca0110.ko
Skipping BTF generation for sound/pci/hda/snd-hda-codec-ca0110.ko due to unavailability of vmlinux
[... 237 lines cut ...]
alias:          hdaudio:v10EC0268r*a01*
alias:          hdaudio:v10EC0267r*a01*
alias:          hdaudio:v10EC0262r*a01*
alias:          hdaudio:v10EC0260r*a01*
alias:          hdaudio:v10EC0257r*a01*
alias:          hdaudio:v10EC0256r*a01*
alias:          hdaudio:v10EC0255r*a01*
alias:          hdaudio:v10EC0245r*a01*
alias:          hdaudio:v10EC0236r*a01*
alias:          hdaudio:v10EC0235r*a01*
alias:          hdaudio:v10EC0234r*a01*
alias:          hdaudio:v10EC0233r*a01*
alias:          hdaudio:v10EC0231r*a01*
alias:          hdaudio:v10EC0230r*a01*
alias:          hdaudio:v10EC0225r*a01*
alias:          hdaudio:v10EC0222r*a01*
alias:          hdaudio:v10EC0221r*a01*
alias:          hdaudio:v10EC0215r*a01*
depends:        snd-hda-codec,snd-hda-core,snd-hda-codec-generic,snd
retpoline:      Y
name:           snd_hda_codec_realtek
vermagic:       5.15.0-MANJARO+ SMP preempt mod_unload 
```

I hope this information was helpful. Thanks for be interested.

aymanbagabas, 2021-11-09:
What's the output of `cat /sys/bus/hdaudio/devices/hdaudio*/*`?

aymanbagabas, 2021-11-09:
@enginacikgoz could you also provide your `sudo alsa-info.sh` output?

enginacikgoz, 2021-11-09:
cat /sys/bus/hdaudio/devices/hdaudio*/*
```
0x1
R6xx HDMI
cat: /sys/bus/hdaudio/devices/hdaudioC0D0/driver: Is a directory
0x0
hdaudio:v1002AA01r00100700a01
cat: /sys/bus/hdaudio/devices/hdaudioC0D0/power: Is a directory
0x100700
cat: /sys/bus/hdaudio/devices/hdaudioC0D0/subsystem: Is a directory
0xaa0100
0x1
DRIVER=snd_hda_codec_hdmi
MODALIAS=hdaudio:v1002AA01r00100700a01

0x1002aa01
ATI
cat: /sys/bus/hdaudio/devices/hdaudioC0D0/widgets: Is a directory
0x1
Generic
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/driver: Is a directory
0x0
hdaudio:v10EC0256r00100002a01
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/power: Is a directory
0x100002
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/subsystem: Is a directory
0x1e833243
0x1
DRIVER=snd_hda_codec_generic
MODALIAS=hdaudio:v10EC0256r00100002a01

0x10ec0256
Realtek
cat: /sys/bus/hdaudio/devices/hdaudioC1D0/widgets: Is a directory
```

And here is `sudo alsa-info.sh`
[alsa-info](https://pastebin.com/xkTK6FV7)

aymanbagabas, 2021-11-09:
k, i guess it's clear what the issue now haha. the card uses the `snd_hda_codec_generic` driver and not the `realtek` one hehe. so we're basically patching the wrong driver lol

enginacikgoz, 2021-11-09:
> k, i guess it's clear what the issue now haha. the card uses the `snd_hda_codec_generic` driver and not the `realtek` one hehe. so we're basically patching the wrong driver lol

Sorry for bothering you so much. =')

enginacikgoz, 2021-11-10:
Additional info:
When I run the `xdotool key XF86AudioMicMute` command, the microphone can be turned on and off, but the led does not work

enginacikgoz, 2021-11-10:
Additional info:
When i change `/sys/devices/platform/huawei-wmi/leds/platform::micmute/trigger` parameter like "kbd-capslock,BAT0-charging-or-full,audio-mute,hci0-power" nothing was happend. Led does not syncing with this triggers. But `echo 1 > brightness ` works flawlessly

aymanbagabas, 2021-11-15:
> k, i guess it's clear what the issue now haha. the card uses the `snd_hda_codec_generic` driver and not the `realtek` one hehe. so we're basically patching the wrong driver lol

Okay, I take that back, `10EC0256` is a realtek driver so `snd_hda_codec_realtek` should be loaded when you boot the system. `/sys/bus/hdaudio/devices/hdaudioC1D0` shows that the loaded driver is in fact `snd_hda_codec_generic` which is realy confusing.

sermart1234, 2021-11-27:
Om my Honor Magicbook Pro 2020 i have popup notification, but led don't  turn on.

sermart1234, 2021-11-27:
```
sermart@sermart-HLYL-WXX9:~/$ xinput
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ ELAN2204:00 04F3:30F5 Mouse               id=10   [slave  pointer  (2)]
⎜   ↳ ELAN2204:00 04F3:30F5 Touchpad            id=11   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Power Button                              id=8    [slave  keyboard (3)]
    ↳ ov9734_azurewave_camera: ov9734           id=9    [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=13   [slave  keyboard (3)]
    ↳ Huawei WMI hotkeys                        id=12   [slave  keyboard (3)]
sermart@sermart-HLYL-WXX9:~/$ xinput test 12
key press   198 
key release 198 
key press   198 
key release 198
```
198 is micmute key

canack, 2021-11-28:
i have got same outputs with sertmant1234.
`key press   198`
`key release 198`
`echo 1 > brightness` works but triggers doesn't work.

sermart1234, 2021-11-29:
On Huawei Matebook D15 AMD micmute led works
```
[nixos@nixos:~]$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ ELAN2204:00 04F3:30F5 Mouse             	id=12	[slave  pointer  (2)]
⎜   ↳ ELAN2204:00 04F3:30F5 Touchpad          	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ ov9734_azurewave_camera: ov9734         	id=11	[slave  keyboard (3)]
    ↳ Huawei WMI hotkeys                      	id=14	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=15	[slave  keyboard (3)]

[nixos@nixos:~]$ xinput test 14
key press   198 
key release 198 
key press   198 
key release 198 
```

`cat trigger` print 
`none kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock ACAD-online BAT1-charging-or-full BAT1-charging BAT1-full BAT1-charging-blink-full-solid rfkill-any rfkill-none audio-mute [audio-micmute] rfkill0 phy0rx phy0tx phy0assoc phy0radio rfkill1`

sermart1234, 2021-11-29:
@aymanbagabas how i can fix audio-micmute trigger?

aymanbagabas, 2021-11-29:
> @aymanbagabas how i can fix audio-micmute trigger?

when you press the micmute key, do you get `Unknown key pressed, code: 0x...` in `dmesg`? 

the wmi interface fires a key event when you press a function key. then this gets processed using [`huawei_wmi_process_key`](https://github.com/aymanbagabas/Huawei-WMI/blob/master/huawei-wmi.c#L691) which maps keycodes to linux sparse keymap defined in https://github.com/aymanbagabas/Huawei-WMI/blob/master/huawei-wmi.c#L75

the micmute key works by mapping `0x287` which is usually what you get from the interface, not `198`, to `KEY_MICMUTE` which then Systemd `60-keyboard.hwdb` keyboard mapping maps to `F20` because of some issue with the micmute keycode being not compatible with window systems. then the desktop environment should handle this key event to mute the mic which then triggers the led trigger `audio-micmute` and turn the led on.

sermart1234, 2021-12-01:
@aymanbagabas 
```
[   39.044958] atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
[   39.044972] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
[   39.160598] atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
[   39.160606] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
[   44.739427] atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
[   44.739442] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.

```

sermart1234, 2021-12-05:
@aymanbagabas `huawei_wmi_process_key` work. I delete `{ KE_KEY,    0x287,          { KEY_MICMUTE } }` and micmute notification is not displayed. But...
```
//{ KE_KEY,    0x287,          { KEY_MICMUTE } },
{ KE_KEY,    0x287,          { KEY_CONFIG } },
```
don't work. 
it spams in dmesg whenever I press the Huawei hotkeys
```
[   44.739427] atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
[   44.739442] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
```

aymanbagabas, 2021-12-06:
@sermart1234 good observation. Huawei-WMI driver creates a new keyboard for the hotkeys which are defined in that table, `0x287` is the micmute keycode extracted from the WMI method. `atkbd` is the original driver for the keyboard which doesn't handle Huawei hotkeys and instead emit `0xf8` for almost all hotkeys.

sermart1234, 2021-12-06:
@aymanbagabas how do triggers for LEDs work? how can emulate a trigger? Does it depend on the audio driver? should I configure systemd, udev? Or will you just have to write a script for processing the micmute button?

aymanbagabas, 2021-12-06:
@sermart1234 AFAICT each trigger is implemented differently in the LED subsystem. if `echo 1 > brightness` works, then the triggers _should_ work too unless something wrong with the trigger. as you can see in the source code [here](https://github.com/aymanbagabas/Huawei-WMI/blob/master/huawei-wmi.c#L304-L317), we only set the default trigger which can be changed from the LED sysfs interface

sermart1234, 2021-12-08:
@aymanbagabas Should the Realtek driver send events to turn off the LED? or DE?

sermart1234, 2022-02-20:
Copy 99-Huawei.hwdb to /etc/udev/hwdb.d/ then update the hwdb tables
dmesg:
`[  210.011813] SELinux:  Context u:object_r:system_data_file:s0 is not valid (left unmapped).`

aymanbagabas, 2022-03-11:
Can someone try this patch with the above instructions? Thank you!

[0001-PATCH-ALSA-hda-realtek-New-Huawei-vendor-ID.txt](https://github.com/aymanbagabas/Huawei-WMI/files/8235925/0001-PATCH-ALSA-hda-realtek-New-Huawei-vendor-ID.txt)

canack, 2022-03-11:
> Can someone try this patch with the above instructions? Thank you!
> 
> [0001-PATCH-ALSA-hda-realtek-New-Huawei-vendor-ID.txt](https://github.com/aymanbagabas/Huawei-WMI/files/8235925/0001-PATCH-ALSA-hda-realtek-New-Huawei-vendor-ID.txt)

unfortunately it is not working :( I guess matebook D16 doesn't use realtek driver.
If you wish, I can give you information by running additional commands on the computer I use.

m1khal3v, 2022-06-17:
Hello everybody. Any news here? May i help with something?

aymanbagabas, 2022-08-04:
Closing this one as it is related to the AMD sound driver. https://bugzilla.kernel.org/show_bug.cgi?id=215119#c1

closed 2022-08-04 by aymanbagabas

ochi12, 2024-04-10:
Huawei D15 2020 internal microphone array not detected