aymanbagabas/Huawei-WMI #55: Charge thresholds doesn't work anymore after BIOS update on HUAWEI MateBook D 15 AMD

mpi3d, 2021-11-22:
Hi, I just updated my BIOS from version 1.11 to version 1.17 on my HUAWEI MateBook D 15 AMD. After that, setting charge thresholds on Linux as stopped working. Everything else like FN lock and mic led is still working. Also, I'm still able to change those thresholds on Windows.

Steps to reproduce the behavior:
1. Download the latest BIOS from HUAWEI Support: https://consumer.huawei.com/en/support/laptops/matebook-d-15-amd/
2. Upgrade your BIOS using Windows on your SSD
3. Start using Linux to set charge thresholds.

I just want to have charging thresholds working on Linux as it was working before.

 - acpidump: [acpidump.zip](https://github.com/aymanbagabas/Huawei-WMI/files/7583524/acpidump.zip)
 - dmidecode: [dmidecode.zip](https://github.com/aymanbagabas/Huawei-WMI/files/7583530/dmidecode.zip)
 - Kernel: `5.14.10-1-MANJARO`
 - Distro: Manjaro
 - Loaded WMI modules:
``` sh
$ lsmod | grep wmi
huawei_wmi             24576  0
ledtrig_audio          16384  3 snd_ctl_led,snd_hda_codec_generic,huawei_wmi
sparse_keymap          16384  1 huawei_wmi
wmi_bmof               16384  0
wmi                    36864  2 huawei_wmi,wmi_bmof
```

I can get Windows charge thresholds:
``` sh
$ cat /sys/devices/platform/huawei-wmi/charge_control_thresholds
40 70
```

I can write them even if nothing happen (The battery continue charging above those thresholds):
``` sh
$ cat /sys/devices/platform/huawei-wmi/charge_control_thresholds
95 100
$ echo "70 90" | sudo tee /sys/devices/platform/huawei-wmi/charge_control_thresholds
70 90
$ cat /sys/devices/platform/huawei-wmi/charge_control_thresholds
70 90
```

Special case:
``` sh
$ echo "0 100" | sudo tee /sys/devices/platform/huawei-wmi/charge_control_thresholds
0 100
$ cat /sys/devices/platform/huawei-wmi/charge_control_thresholds
40 70
```

larax2do, 2021-11-25:
Hello, I have the matebook D14 dual boot Fedora and Windows. I just noticed that the windows battery charge control does not work either. Yesterday I got the BIOS  update :c

sermart1234, 2021-11-29:
@mpi3d @larax2do Charge thresholds work on Huawei MateBook D 15 with 3700U. BIOS 1.17. 
In version 1.17, a new feature has been added - smart charging. It overrides battery protection. You have to turn it off so that the battery protection works as before. Later I will try to do it on Linux

mpi3d, 2021-11-29:
@sermart1234 Indeed, it does work after turning off smart charging. But after unplugging for more than 1 minute the charger, I'm unable to change charge thresholds. I need to reboot into Windows to be able to change them again.

LucasPintoRicardo, 2021-12-16:
Hello everyone, it doesn't work as expected on my Matebook D 14 too (R5 3500U with dual-boot Fedora/Windows), even with smart charging turned off, and I have the same problem on Windows. I think that this issue doesn't come from Huawei-WMI but from the BIOS itself (version 1.17), because in my case, it seems that only the end threshold works. 

I don't know if you can reproduce this: 

1. If I have my start-end thresholds at 40%-70% and my laptop is at 60%, my battery will start charging when plugged in (until it hits 70%, see 3.)... But if I put my end threshold at 60% **exactly**, then it will stop charging. 
2. If I put it below 60%, it will charge (ignoring the threshold).
3. And if I put it at 65% it will charge until it hits those 65%. 

So to mitigate the issue, I have to play with the end threshold, putting it at my exact current battery capacity when in range (it could maybe get automated with a script). The only thing that doesn't work with this DIY solution is that if the end threshold is below your current capacity, your battery won't discharge and reach it, but it will fully charge instead, so you have to unplug your laptop until you are in range (in my example between 40% and 70%, see 2.).

dalareo, 2022-01-27:
I am using Honor Magicbook 14 and battery thresholds don't work anymore on Linux after applying las BIOS update either. Any news?

sermart1234, 2022-01-31:
@dalareo if you have Windows: PCManager - Settings - Battery and turn off Smart charging.
On Linux you can try check the battery protection mode
```
sudo -i
echo 0x00001603 | sudo tee /sys/kernel/debug/huawei-wmi/arg
echo 0x00001603 | sudo tee /sys/kernel/debug/huawei-wmi/arg //strictly twice
```
[{0x00,0x00,0x00,0x00},{0x00,0x**01**,0x48,0x28,0x46,0x00... - 70%
[{0x00,0x00,0x00,0x00},{0x00,0x**02**,0x48,0x46,0x5a,0x00... - 90%
[{0x00,0x00,0x00,0x00},{0x00,0x**03**,0x48,0x5f,0x64,0x00... - 100%
[{0x00,0x00,0x00,0x00},{0x00,0x**04**,0x48,0x28,0x46,0x00... - smart charge

You can try set threshold
```
sudo -i
echo 0x462848011503 | sudo tee /sys/kernel/debug/huawei-wmi/arg // 70%
//echo 0x5a4648021503 | sudo tee /sys/kernel/debug/huawei-wmi/arg // 90%
//echo 0x645f48031503 | sudo tee /sys/kernel/debug/huawei-wmi/arg // 100%
//echo 0x462848041503 | sudo tee /sys/kernel/debug/huawei-wmi/arg // smart charge
echo 0x00001603 | sudo tee /sys/kernel/debug/huawei-wmi/arg
echo 0x00001603 | sudo tee /sys/kernel/debug/huawei-wmi/arg //strictly twice
```
Please, test this. Is it safe.
PCManager cannot read the values and always overwrites them, maybe.

dalareo, 2022-02-01:
Hi @sermart1234 I get "permissions denied" for that file. Even with root user.

sermart1234, 2022-02-02:
@dalareo sorry. I copied the wrong commands. Please repeat

sermart1234, 2022-02-03:
@aymanbagabas acpidump and dmidecode [honor_magicbook_pro_2020_4600h_bios_1_08.zip](https://github.com/aymanbagabas/Huawei-WMI/files/7994127/honor_magicbook_pro_2020_4600h_bios_1_08.zip)
New methods:
```
0x00000C02 GBAI
0x00001203 SBAC
0x00001303 GBAC
0x00001403 GAIT
0x00001503 SBCM #battery charge mode
0x00001603 GBCM #battery charge mode
```
`SBCM` can set thresholds as a `SBTT`. Both write in  `\_SB.PCI0.LPC0.EC0.BCCL` and  `\_SB.PCI0.LPC0.EC0.BCCH`
Set `echo 0x462848011503 | sudo tee /sys/kernel/debug/huawei-wmi/arg`
0x28 = 40
0x46 = 70
0x48 = 72 - unknow
0x[stop_threshold][start_threshold]48[mode]1503
mode 1 - home
mode 2 - office
mode 3 - travel
mode 4 - smart charge - disable user-settable thresholds
All modes except 4 do not affect thresholds
<details>
  <summary>SBTT</summary>
  
    Method (SBTT, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100) {})
        Local0 = Arg0
        CreateByteField (Arg0, 0x02, BCLB)
        CreateByteField (Arg0, 0x03, BCHB)
        CreateByteField (BUFF, Zero, STAT)
        STAT = Zero
        If (((BCLB == Zero) && (BCHB == 0x64)))
        {
            \_SB.PCI0.LPC0.EC0.BCCL = Zero
            \_SB.PCI0.LPC0.EC0.BCCH = Zero
        }
        Else
        {
            \_SB.PCI0.LPC0.EC0.BCCL = BCLB /* \SBTT.BCLB */
            \_SB.PCI0.LPC0.EC0.BCCH = BCHB /* \SBTT.BCHB */
        }

        Return (BUFF) /* \SBTT.BUFF */
    }

    Method (GBTT, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100) {})
        Local0 = Arg0
        CreateByteField (BUFF, Zero, STAT)
        CreateByteField (BUFF, One, GLCP)
        CreateByteField (BUFF, 0x02, GHCP)
        STAT = Zero
        GLCP = \_SB.PCI0.LPC0.EC0.BCCL
        GHCP = \_SB.PCI0.LPC0.EC0.BCCH
        Return (BUFF) /* \GBTT.BUFF */
    }
</details>
<details>
  <summary>SBCM</summary>

    Method (SBCM, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100) {})
        Local0 = Arg0
        CreateByteField (Arg0, 0x02, CMOD)
        CreateByteField (Arg0, 0x03, DTTS)
        CreateByteField (Arg0, 0x04, BCLB)
        CreateByteField (Arg0, 0x05, BCHB)
        CreateByteField (BUFF, Zero, STAT)
        STAT = Zero
        \_SB.PCI0.LPC0.EC0.CHAM = CMOD /* \SBCM.CMOD */
        \_SB.PCI0.LPC0.EC0.DTTS = DTTS /* \SBCM.DTTS */
        \_SB.PCI0.LPC0.EC0.BCCL = BCLB /* \SBCM.BCLB */
        \_SB.PCI0.LPC0.EC0.BCCH = BCHB /* \SBCM.BCHB */
        Return (BUFF) /* \SBCM.BUFF */
    }

    Method (GBCM, 1, NotSerialized)
    {
        Name (BUFF, Buffer (0x0100) {})
        Local0 = Arg0
        CreateByteField (BUFF, One, CMOD)
        CreateByteField (BUFF, 0x02, DTTS)
        CreateByteField (BUFF, 0x03, BCLB)
        CreateByteField (BUFF, 0x04, BCHB)
        CreateByteField (BUFF, Zero, STAT)
        CMOD = \_SB.PCI0.LPC0.EC0.CHAM
        DTTS = \_SB.PCI0.LPC0.EC0.DTTS
        BCLB = \_SB.PCI0.LPC0.EC0.BCCL
        BCHB = \_SB.PCI0.LPC0.EC0.BCCH
        STAT = Zero
        Return (BUFF) /* \GBCM.BUFF */
    }
</details>

SBTT is legacy?

mpi3d, 2022-02-05:
Sounds like setting thresholds using this new method also works on HUAWEI MateBook D 15 AMD.
It would be great if this could be implemented without using this "debug" method.
Anyway, thanks for your work @sermart1234

sermart1234, 2022-02-09:
@mpi3d When I find out what the unknown parameter is responsible for, I will implement the interface. In addition, we need to deal with the old interface

mpi3d, 2022-02-10:
Ok thanks

sermart1234, 2022-05-11:
@mpi3d please test 
https://github.com/aymanbagabas/Huawei-WMI/pull/64

mpi3d, 2022-05-11:
#64 has solved the issue. Closing.
Thanks @sermart1234.

closed 2022-05-11 by mpi3d