aymanbagabas/Huawei-WMI #65: Battery Protection not working

Rolo1973, 2022-08-21:
Hi, 

on my Huawei matebook D AMD (Arch KDE) Huawei WMI does not work for battery protection. I have a problem with all the distro I tried those years (Ubuntu, Arch, Manjaro, Debian based and so on) that on ma laptop that is a Huawei Matebook D AMD SoC even if Huawei-WMI driver is included in the kernel (so I can use FN key and OSD for volume ...) the battery protection doesn't work even in theory system say it stop charging at 70% , it charge 100%. 
if I boot to windows (with HuaweiPCManager that set battery to 70%) and reboot to linux battery stay at 70% but after 2 or 3 reboot to linux it loose the setting and goes to fully charged 100%, because of that I install Huawei-WMI to sets group write privileges and reinstates battery charge-thresholds. I have 2 services : huawei-wmi-reinstate and huawei-wmi-privilege but both are unloaded and inactive on systemd, how to get those 2 service loaded at startup and active? Trying to activate I have this error: $ sudo systemctl enable --now huawei-wmi-privilege.service Unit /usr/lib/systemd/system/huawei-wmi-privilege.service is added as a depende ncy to a non-existent unit sys-devices-platform-huawei\x2dwmi.device
Command  sudo systemctl enable --now huawei-wmi-reinstate.service   doesn,t provide errors but is not active and still unloaded
Can you check?  Battery protection is the only thing not working on my huawei Matebook D 2018 version AMD
Thanks!!!

egormanga, 2022-08-21:
Actually, I've misdirected you; that issue goes to [qu1x/huawei-wmi](https://github.com/qu1x/huawei-wmi). Sorry for that.

Rolo1973, 2022-08-21:
Ahh don't worry Done
Thanks

closed 2022-08-21 by Rolo1973

aymanbagabas, 2022-08-21:
I think this is related to the new Smart Battery protection mode introduced in the new models.
CC/ @sermart1234

egormanga, 2022-08-22:
Moved to here: https://github.com/qu1x/huawei-wmi/issues/16
(which now also states it's an 2018 model)

ahmetlii, 2023-03-11:
This issue seems like a rabbit hole, as the driver ignores values despite being properly set even after ruling out any other possibilities.