aymanbagabas/Huawei-WMI #67: Microphone mute button does not work

smotim, 2022-09-20:
**Describe the bug**
Starting with kernel version 5.19.8-200, the microphone mute button stopped working for me (honor magicbook pro 2020). Adjacent f-buttons work, such as sound control.

**To Reproduce**
Steps to reproduce the behavior:
1. install a system like mine
2. boot with kernel 5.19.8 or 5.19.9
3. Press the f7 button (with Fn off)

**Expected behavior**
Gnome pop-up window with a strikethrough microphone icon, the appearance of the same icon in the tray. The microphone will stop working

**Screenshots**

**(please complete the following information):**
 - acpidump `sudo acpidump > acpidump.out` I didn't find the right rpm package. If necessary, tell me, I will do
 - dmidecode 
[dmidecode.txt](https://github.com/aymanbagabas/Huawei-WMI/files/9606259/dmidecode.txt)

 - Kernel `5.19.9-200.fc36.x86_64 (same bug in 5.18)`
 - Distro Fedora 36
 - Loaded WMI modules ```
wmi_bmof               16384  0
wmi                    32768  1 wmi_bmof

 - Dmesg log 
[dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/9606122/dmesg.txt)
 - ALSA log [alsa-info.txt](https://github.com/aymanbagabas/Huawei-WMI/files/9606145/alsa-info.txt)
 - `evtest` if it's keyboard related. Run `sudo evtest` then choose "Huawei WMI hotkeys", test all hotkeys and submit output.```
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Lid Switch
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	2.4G Mouse
/dev/input/event5:	2.4G Mouse
/dev/input/event6:	Video Bus
/dev/input/event7:	ELAN2204:00 04F3:30F5 Mouse
/dev/input/event8:	ELAN2204:00 04F3:30F5 Touchpad
/dev/input/event9:	PC Speaker
/dev/input/event10:	ov9734_azurewave_camera: ov9734
/dev/input/event11:	HD-Audio Generic HDMI/DP,pcm=3
/dev/input/event12:	HD-Audio Generic HDMI/DP,pcm=7
/dev/input/event13:	HD-Audio Generic Headphone
Select the device event number [0-13]: 

What should I choose here?

 - `acpi_listen` if it's keyboard related. Run `sudo acpi_listen` and test all hotkeys. Submit output. I didn't find the right rpm package. If necessary, tell me, I will do

**Additional context**
In the kernel version, where everything works fine, I output more lines when loaded wmi modules failed. Maybe that's the point?

smotim, 2022-09-20:
```
huawei_wmi             20480  0
ledtrig_audio          16384  2 snd_hda_codec_generic,huawei_wmi
sparse_keymap          16384  1 huawei_wmi
wmi_bmof               16384  0
wmi                    32768  2 huawei_wmi,wmi_bmof
```
An addition to the Additional context. Here is the output in Linux fedora 5.19.7-200

smotim, 2022-10-05:
I was able to fix it. Downloaded rpm from Releases. In Gnome Software, you can see that the application is already installed. Then I deleted it, and then installed it again via the downloaded rpm. Now the microphone mute button is working again.

smotim, 2022-10-06:
I just noticed that the fix only works until the next kernel update. Then I'll have to repeat it. Is there anything that can be done to make the effect last forever?

smotim, 2022-10-09:
`sudo modprobe huawei_wmi`
It seems it can help if you disable Secure Boot before these

sermart1234, 2023-04-13:
> I didn't find the right rpm package

acpica-tools