aymanbagabas/Huawei-WMI #68: Performance mode on Huawei Matebook 16s

vikanezrimaya, 2022-12-28:
**Is your feature request related to a problem? Please describe.**
On Windows, Huawei PC Manager allows to enable "Performance mode" when the laptop is plugged in and sufficiently charged. On Linux, similar functionality is not exposed in an obvious way (perhaps it already exists, but in a different form out-of-scope for this driver? Please advise if so).

**Describe the solution you'd like**
Add a toggle, not unlike the battery protection interface, that reproduces the actions that Huawei PC manager does upon enabling "Performance mode".

**Describe alternatives you've considered**
 - On supported processors, `power-profiles-daemon` allows to enable "performance" or "power-saver" mode by interacting with the `intel_pstate` driver. However, no matter how much load I throw at the laptop, I can't seem to make the fans spin quite the same as on Windows when running a game.

**Additional context**
Perhaps joining forces with the developers of `power-profiles-daemon` might be a wise move here?

vikanezrimaya, 2022-12-28:
Hm. Okay, funny story, I'm not sure if that thing actually does anything but spin the fans more. Here are some Geekbench benchmarks:
 - [Windows: Performance mode vs Normal mode](https://browser.geekbench.com/v5/cpu/compare/19550182?baseline=19550234)
 - [Windows in Performance mode vs Linux](https://browser.geekbench.com/v5/cpu/compare/19550182?baseline=19550066)
 - [Windows in Balanced mode vs Linux](https://browser.geekbench.com/v5/cpu/compare/19550234?baseline=19550066)

They are showing pretty similar results, but it looks like at least in CPU-related matters Linux wins every time (though in Windows, Performance mode resulted in 4% gain in benchmark score).

vikanezrimaya, 2022-12-28:
Oh. Wait a second, I think this was already added in #56. Will close after I figure out how to install a new version.

vikanezrimaya, 2022-12-28:
Compiling from `feat/kbdlight` and loading the resulting module (and setting `/sys/devices/platform/huawei-wmi/power_unlock` to `1` of course!) resolved my issue. Closing.

closed 2022-12-28 by vikanezrimaya

sermart1234, 2022-12-28:
@vikanezrimaya can you send dmidecode and acpidump?