aymanbagabas/Huawei-WMI #69: Ignore Ambient Light Sensoring in Huawei MateBook X Pro 2022 laptops

dark-beep-boop, 2023-02-21:
**The problem**
I have an Arch Linux installation in a Huawei MateBook X Pro 2022 laptop and there's an annoying message that keeps spamming in the journal:

```
...
ene 08 17:49:16 logos kernel: input input12: Unknown key pressed, code: 0x02c1
ene 08 17:49:16 logos kernel: input input12: Unknown key pressed, code: 0x02c1
ene 08 17:49:17 logos kernel: input input12: Unknown key pressed, code: 0x02c1
...
```

[This Archlinux forum thread](https://bbs.archlinux.org/viewtopic.php?pid=2085797#p2085797) describes the issue in more detail, and, according to [litew's answer](https://bbs.archlinux.org/viewtopic.php?pid=2083523#p2083523) to the thread, it's related to the ambient light sensor, which is ignored in other laptops.

The main problem is that it floods the journal and can make it challenging to use journalctl normally.

**The workaround**
One workaround could be to add { KE_IGNORE, 0x2c1, { KEY_RESERVED } }, to huawei-wmi.c.

aymanbagabas, 2023-02-21:
This looks like it should be handled by https://github.com/systemd/systemd/blob/main/hwdb.d/60-keyboard.hwdb
See the Huawei section in the file

dark-beep-boop, 2023-02-21:
I've tried adding a custom rule to hwdb to capture the keycode targetting any device:
 ```
*
 KEYBOARD_KEY_2c1=reserved
```
I can verify that the rule has been added using system-hwdb query with any existent modalias:
```
~❯ sudo systemd-hwdb query dmi:bvnHUAWEI:bvr1.10:bd05/23/2022:br1.10:efr1.10:svnHUAWEI:pnMRGF-XX:pvrM1010:rvnHUAWEI:rnMRGF-XX-PCB:rvrM1010:cvnHUAWEI:ct10:cvrM1010:skuC100:
KEYBOARD_KEY_2c1=reserved
```
But the message keeps showing in the journal, so I don't think the problem is with hwdb.

aymanbagabas, 2023-02-21:
@dark-beep-boop thanks again for reporting this issue. This is a bug indeed and should be added here and upstream. Would you mind creating a pull request with the fix?

aymanbagabas, 2023-02-22:
Fixed in https://github.com/aymanbagabas/Huawei-WMI/pull/70

closed 2023-02-22 by aymanbagabas