aymanbagabas/Huawei-WMI #71: keyboard backlight timeout option "never"

www159, 2023-03-16:
Hi, I have install branch feat/kbdlight and it works fine on my laptop.
backlight timeout can be easily set but I wonder if I can set the backlight never turning off?

There is the Huawei  F&Q show[image]

and I have reviewed the related issues. Is there something like `timeout-toggle` to implenment the function?

sermart1234, 2023-04-13:
`echo 0 | sudo tee /sys/devices/platform/huawei-wmi/kbdlight_timeout`

Alexey336, 2023-04-26:
> `echo 0 | sudo tee /sys/devices/platform/huawei-wmi/kbdlight_timeout`

For me there is no such file. Do I need to build newer driver from source? I have this feature on windows

sermart1234, 2023-04-27:
@Alexey336 build branch https://github.com/aymanbagabas/Huawei-WMI/tree/feat/kbdlight

Alexey336, 2023-04-29:
> @Alexey336 build branch https://github.com/aymanbagabas/Huawei-WMI/tree/feat/kbdlight

Still have no such file

closed 2023-04-30 by www159

sermart1234, 2023-05-02:
Please, send acpidump and dmidecode output