aymanbagabas/Huawei-WMI #73: Battery tresholds, kbd backlight, power unlock not working after BIOS update

DimseBoms, 2023-05-02:
### Description
Battery charge tresholds, keyboard backlight timeout and power unlock does not work correctly after updating the BIOS on my Huawei Matebook D14 (2020) AMD to version 1.19. The battery treshold seems locked to 100%, keyboard backlight timeout seems to always be set to 5 minutes and power unlock does not seem to change the power draw. FN key and FN lock seems to work though.

I have no issues that I know of in the reinstate scripts and the smart charge feature is turned off.

### Steps to reproduce the behavior:
1. Go to /etc/default/huawei-wmi/*
2. Try to change the values for battery tresholds/kbd backlight/power unlock
3. Do not see the expected changes in behaviour

### Expected behaviour
Expect battery behaviour, kbd backlight and power limits to represent the options that are set in /etc/default/huawei-wmi/*

### Additional information
 - acpidump: [acpidump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11373462/acpidump.txt)
 - dmidecode: [dmidecode.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11373533/dmidecode.txt)
 - Kernel: `Linux dmitriys-huawei-matebook-d14 5.19.0-41-generic #42~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 18 17:40:00 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`
 - Distro: `Ubuntu LTS 22.04`
 - Loaded WMI modules: [lsmod.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11373571/lsmod.txt)
 - Dmesg log: [dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11373589/dmesg.txt)
 - ALSA log: [alsa.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11373617/alsa.txt)

### Comment
Tell me if there is anything else I should supply or if theres anything I should test or do.

viktory683, 2023-06-10:
I join it

**Description**

Battery threshold not working at all
Fn key works fine as expected.

**Additional information:**
 - acpidump: [acpidump.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11713044/acpidump.txt)
 - dmidecode: [dmidecode.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11713047/dmidecode.txt)
 - Kernel: `Linux sky 6.3.6-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Mon, 05 Jun 2023 15:12:42 +0000 x86_64 GNU/Linux`
 - Distro `Arch Linux`
 - Loaded WMI modules: [lsmod.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11713049/lsmod.txt)
 - Dmesg log: [dmesg.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11713050/dmesg.txt)
 - ALSA log: [alsa.txt](https://github.com/aymanbagabas/Huawei-WMI/files/11713051/alsa.txt)

mixin82, 2023-10-21:
I have Matebook 16 and after I updated to BIOS 2.29 the battery threshold wouldn't have any effect, battery was charging all the time. There are several people reporting that this has changed since version 2.28 (I had an older version before).
Anyway, by following several links I found this comment which it seems to work for me: https://github.com/aymanbagabas/Huawei-WMI/issues/55#issuecomment-1025651920.
I didn't have Smart charge set, but I think HUAWEI simply changed the way to set the threshold with newer BIOS.
Hi @sermart1234, thanks a lot for your contribution! I can see your [PR](https://github.com/aymanbagabas/Huawei-WMI/pull/64) been merged into `feat/kbdlight` but it's not in `master`. Also latest kernel doesn't look in sync with this repo. My simple question is this: how do we get your change into the official kernel? Thanks a lot again, great job!