aymanbagabas/Huawei-WMI #85: dkms install fails on PopOS

leftcase, 2024-04-04:
**Describe the bug**
dkms install of kernel module fails on latest version of PopOS 

**To Reproduce**

- Steps to reproduce the behavior:
- Download tarball for huawei-wmi 3.4
- Run sudo dkms ldtarball --archive=Downloads/huawei-wmi-3.4-source-only.dkms.tar.gz
- Run sudo dkms install huawei-wmi/3.4

**Result**

```
Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=6.8.0-76060800daily20240311-generic -C /lib/modules/6.8.0-76060800daily20240311-generic/build M=/var/lib/dkms/huawei-wmi/3.4/build...(bad exit status: 2)
ERROR (dkms apport): binary package for huawei-wmi: 3.4 not found
Error! Bad return status for module build on kernel: 6.8.0-76060800daily20240311-generic (x86_64)
Consult /var/lib/dkms/huawei-wmi/3.4/build/make.log for more information.

```
**make.log details**

```
/var/lib/dkms/huawei-wmi/3.4/build/huawei-wmi.c:488:24: note: (near initialization for ‘huawei_wmi_battery_hook.add_battery’)
/var/lib/dkms/huawei-wmi/3.4/build/huawei-wmi.c:489:27: error: initialization of ‘int (*)(struct power_supply *, struct acpi_battery_hook *)’ from incompatible pointer type ‘int (*)(struct power_supply *)’ [-Werror=incompatible-pointer-types]
  489 |         .remove_battery = huawei_wmi_battery_remove,
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/huawei-wmi/3.4/build/huawei-wmi.c:489:27: note: (near initialization for ‘huawei_wmi_battery_hook.remove_battery’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/huawei-wmi/3.4/build/huawei-wmi.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.8.0-76060800daily20240311-generic/Makefile:1927: /var/lib/dkms/huawei-wmi/3.4/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.0-76060800daily20240311-generic'
```

**uname -a**
`Linux pop-os 6.8.0-76060800daily20240311-generic #202403110203~1711393930~22.04~331756a SMP PREEMPT_DYNAMIC Mon M x86_64 x86_64 x86_64 GNU/Linux`

aymanbagabas, 2024-04-04:
Hi @leftcase, the 3.4 release is outdated. Could you try building from master instead? Fwiw, I'm planning on a new release soon.

leftcase, 2024-04-06:
Hi @aymanbagabas, I built from master and this works. Thank you.
I seem to be missing the huawei-wmi group. When I've used huawei-wmi and matebook-applet in the past; in order to run matebook-applet as a standard user, the standard user must be part of the huawei-wmi group, and the huawei-wmi group does not exist on my current install.

aymanbagabas, 2024-04-06:
> Hi @aymanbagabas, I built from master and this works. Thank you.
> 
> 
> 
> I seem to be missing the huawei-wmi group. When I've used huawei-wmi and matebook-applet in the past; in order to run matebook-applet as a standard user, the standard user must be part of the huawei-wmi group, and the huawei-wmi group does not exist on my current install. I appreciate this quite possibly relates to another project. 
> 
> 

@leftcase The group and permissions are part of another project https://github.com/qu1x/huawei-wmi

closed 2024-04-06 by aymanbagabas