aymanbagabas/Huawei-WMI #87: Secound microphone

Equinoxu, 2024-04-25:
Is it possible to make secound microphone work?
Pavucontrol doesn't even see it
I was searching for this in the web but I didn't managed to fix this problem and I don't know where I should look for help with such a problem

Matebook D16 2024

egormanga, 2025-02-12:
If you refer to the dual-microphone array, it is handled by ALSA and/or SOF which is not related to WMI at all.

Make sure you've got all relevant `*-firmware` packages installed _(e.g. `alsa-firmware`, `sof-firmware` – those are for Arch Linux)_. Also note that no PipeWire/PulseAudio/ALSA tools will show you the ‘second’ microphone as a separate device or even as a stereo-configured channel, as it is used only internally by the sound card's DSP to implement noise & echo cancellation.

---

In the case you mean the headphone jack microphone input, then try placing this to `/etc/modprobe.d/snd_hda_intel.conf`:
```
options snd_hda_intel model=dell-headset-multi
```
then rebooting. Might need to try various different models (relevant to the headset mic) from [here](https://www.kernel.org/doc/html/latest/sound/hd-audio/models.html) instead, as that's just the one worked for my Honor MagicBook 14 2021.

_Note: might also want to check out `huawei-mbx-stereo` for better speaker support depending on your device (in case you need it!), **but that's risky** as mentioned, see more [here](https://patchwork.kernel.org/patch/11086601/) and [here](https://openbeagle.org/jdneal/linux/-/blob/v5.3.9/Documentation/sound/alsa-configuration.rst)._

egormanga, 2025-10-18:
For the record, I have my _stereo_ mic working natively (OOTB) on Honor MagicBook Art 14 (MRA-XXX).

Scratch testing reveals that the third (ANC) microphone located next to the UDB Type-A port also produces sound, but I have no way of determining if it's being used additively or subtractively (i.e. for ANC) on Linux.