aymanbagabas/Huawei-WMI #88: Obsolete ledtrig_audio_get feature.

dontknowhy, 2024-07-16:
**Describe the bug**
A clear and concise description of what the bug is.
This module uses a feature which is deleted from mainline kernel ~~since Linux 6.10~~ .
So I can't compile `huawei-wmi` for Linux kernel 6.10
Feature: `ledtrig_audio_get`

**To Reproduce**
```
$ env LANG=C sudo /boot/build-kernel.sh install
  INSTALL /boot
run-parts: executing /etc/kernel/postinst.d/dkms 6.10.0-Luo520-xanmod1 /boot/vmlinuz-6.10.0-Luo520-xanmod1
dkms: running auto installation service for kernel 6.10.0-Luo520-xanmod1.
Sign command: /lib/modules/6.10.0-Luo520-xanmod1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
Building module(s)...(bad exit status: 2)
Failed command:
make -j16 KERNELRELEASE=6.10.0-Luo520-xanmod1 -C /lib/modules/6.10.0-Luo520-xanmod1/build M=/var/lib/dkms/huawei-wmi/3.4/build CC=clang LD=ld.lld
Error! Bad return status for module build on kernel: 6.10.0-Luo520-xanmod1 (x86_64)
Consult /var/lib/dkms/huawei-wmi/3.4/build/make.log for more information.
dkms autoinstall on 6.10.0-Luo520-xanmod1/x86_64 failed for huawei-wmi(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.10.0-Luo520-xanmod1 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
make[1]: *** [arch/x86/Makefile:302: install] Error 11
make: *** [Makefile:240: __sub-make] Error 2
```
**Screenshots**
~~If applicable, add screenshots to help explain your problem.~~
Thanks to the amazing GFW.I can't upload any image.
```
$ cat /var/lib/dkms/huawei-wmi/3.4/build/make.log
DKMS make.log for huawei-wmi-3.4 for kernel 6.10.0-Luo520-xanmod1 (x86_64)
Tue Jul 16 13:42:28 CST 2024
make[2]: warning: -j16 forced in submake: resetting jobserver mode.
  CC [M]  /var/lib/dkms/huawei-wmi/3.4/build/huawei-wmi.o
/var/lib/dkms/huawei-wmi/3.4/build/huawei-wmi.c:400:28: error: call to undeclared function 'ledtrig_audio_get'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  400 |         huawei->cdev.brightness = ledtrig_audio_get(LED_AUDIO_MICMUTE);
      |                                   ^
1 error generated.
make[3]: *** [scripts/Makefile.build:244: /var/lib/dkms/huawei-wmi/3.4/build/huawei-wmi.o] Error 1
make[2]: *** [Makefile:2087: /var/lib/dkms/huawei-wmi/3.4/build] Error 2
```

**(please complete the following information):**
**acpidump, dmidecode, dmesg and ALSA log will upload later.**
 - Kernel `Linux sb-debian 6.9.9-Luo520-xanmod1 #2 SMP PREEMPT_DYNAMIC Sat Jul 13 13:56:03 CST 2024 x86_64 GNU/Linux`
 - Distro `Debian GNU/Linux trixie trixie/sid x86_64`
 - Loaded WMI modules:
```huawei_wmi             36864  0
ledtrig_audio          12288  2 snd_ctl_led,huawei_wmi
sparse_keymap          12288  1 huawei_wmi
wmi_bmof               12288  0
```

dontknowhy, 2024-07-16:
Pastebin because the upload feature is gone.
[dmesg but base64 encoded](https://paste.debian.net/1323405)
[alsa info](https://paste.debian.net/1323406)

egormanga, 2025-02-12:
This should be closed now as resolved.

closed 2025-02-12 by aymanbagabas