aymanbagabas/Huawei-WMI #98: Cannot enter supend-to-RAM state on Huawei Matebook X Pro 2024

amankhoza, 2025-01-26:
For some reason my machine does not enter deeper levels of suspend such as suspend-to-RAM instead it remains at s2idle. From looking at logs it appears to be an ACPI issue so I'm hoping whether one of the contributors of this repo can help or provide guidance, I'm willing to help test patches etc.

Here's an example of dmesg logs mentioning ACPI errors:
```
[    0.227988] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.227993] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228115] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228119] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228122] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228125] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228128] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228131] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228134] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228138] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228141] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.228145] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20240322/psobject-220)
[    0.236906] ACPI Error: No handler for Region [ECW1] ((____ptrval____)) [EmbeddedControl] (20240322/evregion-131)
[    0.236912] ACPI Error: Region EmbeddedControl (ID=3) has no handler (20240322/exfldio-261)
[    0.236933] ACPI Error: Aborting method \_SB.PC00.LPCB.HWEC.ECRD due to previous error (AE_NOT_EXIST) (20240322/psparse-529)
[    0.236938] ACPI Error: Aborting method \_SB.PC00.LPCB.HWEC.BAT0._STA due to previous error (AE_NOT_EXIST) (20240322/psparse-529)
[    0.236942] ACPI Error: Method execution failed \_SB.PC00.LPCB.HWEC.BAT0._STA due to previous error (AE_NOT_EXIST) (20240322/uteval-68)
[    2.215870] ACPI Error: Aborting method \_SB.IETM.CHRG.PPSS due to previous error (AE_NOT_FOUND) (20240322/psparse-529)
```

egormanga, 2025-02-12:
Please provide your ACPI tables dump (from `acpidump -b`) so we can investigate the issue (which, to be honest, might not belong here but that's not my call to decide).

aymanbagabas, 2025-02-12:
Hey @amankhoza, I don't think this is related to huawei-wmi. Please file a bug report to upstream.

Thank you @egormanga for bringing this to my attention, your work is always appreciated 🙂

closed 2025-02-12 by aymanbagabas

amankhoza, 2025-02-12:
Thanks both, do you know where I should file this bug report?

egormanga, 2025-02-12:
Probably, LKML or the Kernel Bugzilla.